procount/pinn #324: Feature to change default root password in recovery.

iudeen, 2019-10-18:
Is there a feature to change the default recovery username/password from `root` and `raspberry`?

lurch, 2019-10-18:
AFAIK this is where it's currently set https://github.com/procount/pinn/blob/master/buildroot/.config#L268
(but to change it I think you'd need to build your own version of PINN?)

RaspberryPiFan, 2019-10-27:
I think it's not possible without rebuilding because PINN runs from RAM and the system is loaded from a compressed file, so it's read only.

RaspberryPiFan, 2019-11-03:
I think it could be added by adding a config option like passwd=<your password>. This option can be read by PINN and it would run passwd <your password> in the background or directly modify /etc/passwd

lurch, 2019-11-03:
It _could_ be done, but then the password would be readable in plain-text via the cmdline.txt file, which kinda negates the point of changing it? :wink:

RaspberryPiFan, 2019-11-03:
Yes, it is not secure, but in some cases useful, for example when you have a non-english keyboard with no english letters

RaspberryPiFan, 2019-11-03:
But I also don't think it will be done in PINN

RaspberryPiFan, 2019-11-03:
Or a more crazy, but secure idea: A password change option that unpacks the image, changes the password, repacks the image, save it as second file, makes a third image that deletes the old one and replaces it with the new one and add the third image as next boot script, then this image adds a config.txt option to the new image that deletes the third image

lurch, 2019-11-04:
IIRC NOOBS (and PINN) run directly from RAM, and so the 'first image' is free to directly delete / replace the existing `recovery.rfs` :wink:

MishuMMx, 2019-11-13:
Is there a chance to specify the ssh public key and disable the ssh password authentication?
( Praying for this )

procount, 2020-04-02:
@MishuMMx - This is a separate issue, so I created #362

closed 2020-04-02 by procount