procount/pinn #327: Ubuntu 19.10

jaylfc, 2019-10-29:
Hey, great work! Could you please add the new Ubuntu release. Thanks in advance!

procount, 2019-10-30:
It's already on my list, but as it uses Uboot, it will need a little massaging first. 
Very busy at the moment, so it may be a few weeks yet.

Zaf9670, 2020-02-23:
Just following back up on this. Anything in particular that would be helpful to push this along?

I tried copying an existing image from SD card over but didn't have much luck. That may be due to the JSON files but I'm not sure. I really just want PINN to manage the USB boot until RPi Foundation get the EEPROM update out.

procount, 2020-02-24:
Creating some png slideshow files would be a great help

procount, 2020-02-27:
> I tried copying an existing image from SD card over but didn't have much luck. That may be due to the JSON files but I'm not sure.

Because it uses UBoot, it requires certain files to be located on the first partition of the disk, which is not compatible with multibooting different OSes off different partitions. Therefore UBoot needs to be removed.

Zaf9670, 2020-04-20:
How's the progress going for this OS being added? And if one were to assist making PNG slideshows would that just need to come in the form of a pull request?

procount, 2021-01-13:
superceded by Ubuntu 20.04 and 20.10

closed 2021-01-13 by procount