procount/pinn #328: silentinstall over web

eroom1966, 2019-11-03:
Regarding https://github.com/procount/pinn/issues/196#

Can I ask for an example of how this is done ?
I have tried to reproduce, but it will not perform the silentinstall - I am clearly missing something
... help !!

Here is a dropbox link to my modified version of pinn-lite
https://www.dropbox.com/sh/acmskpcarlcr73r/AADA7Q0EJJb_Il9pj6q8Quq1a?dl=0

This links to my os_list.json here
http://pilotaware.lode.co.uk/downloads/OGN/net/os_list.json

and my recovery.cmdline is

runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline loglevel=2 sdhci.debug_quirks2=4 alt_image_source=http://pilotaware.lode.co.uk/downloads/OGN/net/os_list.json select="OGNR" silentinstall

But it does not autoinstall, it goes interactive - clearly doing something wrong, but not sure what

Thx
Lee

procount, 2019-11-03:
Could you post the output of /tmp/debug?

eroom1966, 2019-11-03:
Aahh, I think I will need guidance on that
How do I go about capturing that 
Thx

procount, 2019-11-03:
https://github.com/procount/pinn/wiki/Troubleshooting

I just tried your settings and it never finished waiting for the network. Maybe I broke something again. It's not a feature I often check.

eroom1966, 2019-11-03:
If you need me to rest anything, very happy to give it a go

procount, 2019-11-03:
I think I've found it. 
It will need  a new release of PINN though.

eroom1966, 2019-11-03:
Hey thats great news, you’re a star
Let me know when you have a release and I will give it a try

eroom1966, 2019-11-03:
I just noticed you have a repo for the ciseco XRF product
What a really strange coincidence, an early version of the PilotAware product was using the ARF - a 500mw version of the XRF

Thx
Lee

procount, 2019-11-03:
Yeah, I bought the Ciseco for the RPI and Arduino hoping to use them, so I downloaded all the stuff I could. When they disappeared, I read someone was looking for the documentation etc so I put it up.

BTW, I managed to silently install your PilotAware OS, but it won;t work on the Pi3B+ - Just 4 red flashes which means it can't find the firmware for the 3B+. 
You should probably change the name of the OS from Raspbian to PilotAware to avoid confusion with Raspbian.

eroom1966, 2019-11-04:
> BTW, I managed to silently install your PilotAware OS, but it won;t work on the Pi3B+ - Just 4 red flashes which means it can't find the firmware for the 3B+.

Great news, can you let me have a pointer to the release with the fix
Thx
Lee

procount, 2019-11-04:
Will do.
I need to package it up.

procount, 2019-11-04:
https://sourceforge.net/projects/pinn/files/testing/pinn-331.zip/download
As this is beta software, please email me for the password to unzip it.

eroom1966, 2019-11-04:
email request set to your sourceforge address

eroom1966, 2019-11-05:
great job!
seems to do the trick

closed 2019-11-05 by eroom1966