procount/pinn #329: Blank Screen with LibreELEC after touch.

MysticWhiteDragon, 2019-11-13:
I want to use LibreELEC as the KODI distro in my CarPi project.  I can't get it to work by itself, but it will work if I use PINN and copy the overlays and config.txt to the LibreELEC partition that PINN makes.  It all works fine except that when I touch the screen, the screen goes blank and I can't get it to display again without restarting.

It works fine in PINN with touch.  Not sure why it turns off the screen in LibreELEC.

lurch, 2019-11-13:
Sounds like this is a problem with LibreELEC, rather than a problem with PINN? :man_shrugging:

procount, 2019-11-15:
The screen turns off in PINN as well, if you touch the correct part of the screen, for the correct duration, and the wind is blowing from the East 😄 

As I mentioned on the forum, once it has booted into LibreElec, you should try 
```
rmmod pimhyp4
moprobe goodix
```

procount, 2019-11-22:
Sorry, I had assumed you meant the Pimoroni Hypertouch screen, but I now realise that you didn't mention which touch screen you were using!
In any case, forget my rmmode/modprobe suggestion - it's not a module.

closed 2020-04-02 by procount