procount/pinn #330: Add OpenMediaVault as new OS

EctoOne, 2019-11-16:
Hi, I just want to ask if it would be possible to add OpenMediaVault as new OS?

IT probably doesn't make much sense to put a NAS Distro on a multiboot card, so it is more of a personal request because PINN lets me hide my Pis and simply reinstall any OS if needed. Without me crawling behind my desk or TV.

In any case, it is much appreciated what you do.

procount, 2019-11-18:
I will take a look.

EctoOne, 2019-11-22:
Just some quick infos about the Distro. I finally had some time to test the OMV_4_Raspberry_Pi_2_3_3Plus_4.img.xz image.

The initial setup takes quite some time to download stuff. I didn't noticed that before because I also think that an ethernet connection is required for the setup. But I'm not sure about that because I was too impatient when I only used Wlan before.

procount, 2019-12-12:
According to their sourceforge page,

"The pre-install images have been deprecated in favor of using this guide - https://forum.openmediavault.org/index.php/Thread/28789-Installing-OMV5-on-Raspberry-PI-s-Armbian-Supported-SBC-s/"

The previous images have been taken down.

So the idea is to install Raspbian-Lite and then install OMV over the top.

> The initial setup takes quite some time to download stuff. 

Yes, After installation, they recommend just leaving it for up to an hour for it to download and upgrade everything. It will auto-reboot without warning during this time, so you may have to select it again in PINN if you've set PINN up not to boot to the last OS

EctoOne, 2019-12-13:
Well, perfect timing I guess. I will take a look at it this weekend and see how it goes. Thanks for looking into it.

Small question though. Since I have not used PINNs Backup system before, I assume that I can make a backup after I've gone through the whole installation process. But how do backups actually handled, are they basically just like any other OS (including all necessary files) and can be installed on any new SD or is there something different? Because if they are just (normal) installation files, I would probably just put it directly into the OS folder of the SD card if possible.

lurch, 2019-12-13:
Which model of Raspberry Pi are you using? The Pi4 is only supported by Raspbian Buster.

procount, 2019-12-13:
>  I copied just the necessary files onto an USB stick and it says "The system is not supported"

Probably not enough!

To support backup you must have a compatible partition_setup.sh file and indicate it in os.json (?) Checkout the original files for Raspbianlite and use them for reference.

procount, 2019-12-13:
I thought I had a json file to allow installing of Stretch versions. Maybe I should resurrect it and add OMV as a flavour.....?

EctoOne, 2019-12-13:
I'm using a Pi 3.

I copied the following files over: Raspbian.png, boot.tar.xz, os.json, partition_setup.sh, partitions.json, and root.tar.xz

Taken from this solution: https://raspberrypi.stackexchange.com/questions/77682/can-i-install-a-pre-downloaded-image-of-raspbian-stretch-with-noobs-lite

And I got the files here: https://downloads.raspberrypi.org/raspbian_lite/archive/2019-04-09-22:48/

So is it because I changed the name/description in the os.json or use OpenMediaVault as folder name? Or can't I create a backup because the image is already on the USB stick?

procount, 2019-12-13:
Those are old files from before backup was supported. 
As a guess, copy partution_setup.sh from latest Raspbian Lite. Check is.json has a supports backup line in it.

EctoOne, 2019-12-13:
I just looked at the backup files from the buster installation and I've not seen anything about backups in the os.json. The partition_setup.sh has the line #supports_backup in PINN at the top.

I also looked at the os.json and the partition_setup.sh from retropie2 on sourceforge and it is the same. So I assume adding the backup line to the old file might work. Otherwise I will try to replace the old file with the new one.

EctoOne, 2019-12-13:
OK just did a quick test and copied the new partition_setup.sh but it still says unsupported. :(

EctoOne, 2019-12-13:
Checked some more files and I found this line: "supports_backup":   true,

In the os_list.json from the UbuntuMate image. Is that the line I need to add to my os.json?

And I guess I couldn't find the line before, because it is handled with the overrides.json. That had at least those entries for all Distros. And what's the difference between: "supports_backup":   true, and: "supports_backup":  update,?

EctoOne, 2019-12-13:
So I added the update line to os.json and now I get: An error occurred backing up. Perhaps there is no disk space?

Although the USB stick should have enough space. I even deleted the old update. So maybe fiddling around with all these files screwed something up. I will test if it works with a clean installation and hope that the overrides.json will work on the offline image.

procount, 2019-12-13:
Sorry I'm out this evening. I'll try and address over the weekend.

EctoOne, 2019-12-13:
Sure no problem, at least I had some time messing around with this. Although I'm out of ideas now, using the unaltered stretch image didn't work because I can't even install it. It has a lower priority then the online buster version and I'm not going to unplug the pi just for that.

procount, 2020-04-02:
Sorry, for late response, but I've added an option to install Stretch to PINN.
You just need to add: `alt_image_source=http://raw.githubusercontent.com/procount/pinn-os/master/os/os_stretch.json` to recovery.cmdline.
If you want to do this within PINN, proceed as follows:
1. Click on the `More` menu button twice to reach the Maintenance menu.
2. With PINN highlighted, click on the `Edit config` button
3. Click on the recovery.cmdline tab

You can use this editor to modify the content of recovery.cmdline and add the requried option in. Be aware that this is all one single line.
OR, you can click the `Edit Options` button, click on the `source` tab then enter the URL in the third box labelled `alt_image_source=`
Click OK to get back to the main menu.
After modifying this, you need to reboot, or maybe try the `Reload Repos` button in the Archival menu.
You should now have 3 more Raspbian Stretch distros to install.

closed 2020-11-25 by EctoOne