procount/pinn #331: test

procount, 2019-11-19:
test issue

procount, 2019-11-19:
@lurch  - Could you open a new test issue on this repo for me please?
I haven't been getting some notifications because they have been bouncing back from my email forwarder, so I just want to test it has now been fixed.

closed 2019-11-19 by procount

lurch, 2019-11-19:
> @lurch - Could you open a new test issue on this repo for me please?

#332 and #333 :slightly_smiling_face:

procount, 2019-11-19:
Thanks. I'm still not getting them on my email domain forwarder, although it doesn't seem to have unverified the email address (yet). At least they are going via google now. Just added my proper email directly, so maybe that will work 🤷‍♂

lurch, 2019-11-20:
ProTip for ProCount: Having multiple independent email addresses (e.g. work and/or personal and/or ISP) is really handy for diagnosing things if/when they go wrong :wink: 
I noticed myself a few years ago that Google seemed to tighten up their rules around forwarded emails...

ProTip2: You can also see GH notifications by clicking on the bell icon in the top-right (only works when you're logged into GH web-interface).

procount, 2019-11-20:
Yeah, Protip2 is how I discovered I was missing email notifications.
I'm still not getting notifications on my main account though :( But I don;t think I would get notifications of my own issues anyway, hence why i needed some help.

lurch, 2019-11-20:
I assume you've checked your spam folder? :wink: 

Let me know if/when you want me to create any additional test issues.

procount, 2019-11-20:
Yes, but Github support had already provided me with the bounced email headers.
I got your last message above, so maybe the whitelisting is finally working!

procount, 2019-11-20:
OK. I think that's enough testing. I seem to be getting plenty of notifications again.
Thanks.