procount/pinn #335: SD Card corruption?

FlyingPersian, 2019-11-22:
Hello,

I'm running the latest PINN with OSMC and RetroPie. I was changing a setting in Kodi a few times and the RPI kept crashing. After changing a setting in /home/osmc/.kodi/userdata/guisettings.xml, my RPI stopped booting into both OSes. PINN loads fine, even the menu etc. work, but when I choose to boot any OS, the screen just goes blue (as in loosing the connection to the RPI). This also happens also every time when I try to get into the recovery mode of PINN. I managed to get into it and get both the debug and dmesg. 
I ran the file system check for all three OSes and indeed there was a dirty bit or something. It was fixed and now when I run the check it's all good. Still, it does not boot into any OS except one random time when it booted into RetroPie. I did a lot of setting up n stuff on Kodi, so it'd be a shame if all was gone. Is there a way to properly check what the issue is (maybe the logs?) or at least backup the OSes? I did run the backup feature in PINN with OSMC, but it isn't working. The USB Stick is recognized (ext4 with /os), but it says it didn't work, perhaps cuz the stick is full, but it isn't (115GB free, 600MB needed for the backup according to the menu).
Weirdly, the SD card activity LED is still blinking. I think it just means that it's trying to read off the SD card, but it's failing since there is no video output anymore?

lurch, 2019-11-23:
I wonder if it might be a power-supply issue? :man_shrugging:   Can you try a different one?

FlyingPersian, 2019-11-23:
> I wonder if it might be a power-supply issue?  Can you try a different one?

Doesn't seem so! Tried a different one, exact same result. My best guess is that it's a corrupted SD card. I ran the file system check in PINN and indeed all three OSes (PINN, RetroPie, OSMC) had a dirty bit which was apparently fixed, but it still seems to be broken.

Bobbbay, 2019-11-24:
The first warning I see is: libpng warning: Unknown iTXt compression type or method
Then I see: "sh -c "fsck.fat -y /dev/mmcblk0p1"" 
 "CP437: Invalid argument
fsck.fat 3.0.26 (2014-03-07)
/dev/mmcblk0p1: 249 files, 26456/32183 clusters
" 
And the errors just go off from there...

FlyingPersian, 2019-11-24:
> At the end of the second file, it says: mounted filesystem with ordered data mode (or something similar) multiple times. For multiple booted systems?

I'm running both RetroPie and OSMC via PINN. But only one OS is running at a time. 

> File 1 says that retropie booted successfully? Line 79

I think that means that the filesystem check was okay. The partition is called RetroPie.

procount, 2019-12-17:
@FlyingPersian - Did you resolve this?
I couldn't see anything obviously wrong in your logs (you can ignore the libpng warning - one of the png files I used contains an unsupported chunk, that's all).
I can only suggest getting a new SD card and re-installing to see if that fixes it. Maybe check your SD card first with h2testw or similar.

FlyingPersian, 2019-12-18:
I simply re-installed everything on a new SD. Not sure what the issue was. Thanks!

closed 2019-12-18 by FlyingPersian

Bobbbay, 2019-12-18:
I personally don't believe this issue should be closed. I'm very glad everything worked out for you, but this might not work for everybody. Imagine having hundreds of projects on your Debian and a handful of saved game states on your RetroPie. Reinstalling on a new SD might not help. 

Also, if this is a bug, I think we should try to find it as soon as we can. If that bug influenced how your card worked, it might influence more major problems, too. 

Of course, this is just my opinion. 

Happy flashing, Bobbbay

lurch, 2019-12-18:
@Bobbbay If the problem was indeed a faulty / corrupted SD card (which "it does not boot into any OS except one random time when it booted into RetroPie" would seem to point to), I don't think there's much that PINN can be expected to do about that? :confused: :man_shrugging:

Bobbbay, 2019-12-19:
Good point @lurch. 

The thing is, if it was the SD card's problem, how come it ran normally after being re-formatted? @FlyingPersian said that they were playing around with some settings in Kodi. I'm guessing that means that the problem originates from the Kodi settings... Maybe some sort of flag? **My guess is that you ran out of storage space on the card**, but to be honest, I have no idea at all.

I just hate leaving a problem unsolved... Well, whatever it was, I guess it's over now. But I just can't stop wondering what caused this 😡😤😠...

I guess since re-formatting is **a** solution, closed this thread shall remain.

If anyone spies anything off, please do tell. Thanks!

Happy formatting, Bobbbay

FlyingPersian, 2019-12-19:
Well I'm not really sure I can be of any help since I reinstalled everything a.k.a. copied over the config dir of Kodi over to my laptop and then back to the new install. I don't even remember which SD card it was tbh (I have like 10 of them), so I can't even check the SD card. I might have thrown it out

Bobbbay, 2019-12-19:
Alright, closed it shall remain, until somebody else has the same problem. 

Happy merges, Bobbbay

lurch, 2019-12-19:
> until somebody else has the same problem.

Such pessimism... :wink: :laughing: