procount/pinn #336: unattended install broken for me

Imperas, 2019-11-23:
@procount 

Hi, after fixing the issue for me previously (#328 Thanks), it is now unfortunately broken - arrgghhh

When I investigated why the unattended installs were no longer working, by attaching a monitor, I see that the following popup appears

A New Version of PINN is available
Do you want to download this version
Yes Ignore No Show_Details

So unfortunately, all my unattended installs are broken, as they wait for a user to select something from the selection :-(

Is there a way I can disable this before all my remote installations get locked up, this is very urgent for me, my automatic installs happen periodically, meaning everything will get locked up

Thx
Lee

Imperas, 2019-11-23:
Update
it looks like this timed out after some time, and continued to install - not sure how long, is there a command line option to expedite this ?
Thx
Lee

closed 2019-11-23 by Imperas

procount, 2019-11-23:
IIRC, you can use no_update to prevent the dialog showing.