procount/pinn #339: Update doesn't work

RaspberryPiFan, 2019-11-25:
I'm using V3.3c and even with manual update check I can't install V3.3.2, it says "No update available".

procount, 2019-11-25:
Ah yes. 
It seems I've forgotten my own numbering scheme and 3.3.2 is not greater than 3.3c!

procount, 2019-11-26:
Fixed with a little hack to pretend the new version is 3.3d

RaspberryPiFan, 2020-02-11:
Again with 3.3.3 and 3.3.4?

procount, 2020-02-11:
Hmm. The thing about hacks is it is easy to forgot what you did. :wink:
I suggest anyone with this issue edits BUILD-DATA and changes the line mentioning "PINN version" to 
`PINN Version: v3.3.3`.
These are only minor changes, so missing out on these changes shouldn't be a problem.
It should fix itself when I bring out v3.4.

closed 2021-01-13 by procount