procount/pinn #341: pinn_init.sh (wrong docs?)

matthuisman, 2019-12-01:
"If a script called /pinn_init.sh exists in PINN's recovery partition, it will be executed before PINN is started. This script will be mounted as /tmp/media/pinn_init.sh and must complete before PINN starts. It can be useful to initialise third party display drivers before the main PINN splash screen is shown."

But here:
https://github.com/procount/pinn/blob/master/buildroot/package/recovery/init#L93

looks like it's actually /tmp/mnt/pinn_init.sh (not /tmp/media/pinn_init.sh)

matthuisman, 2019-12-01:
@procount 

Also, any clever way in the pinn_init.sh to detect it's a Pi4?
I pretty much want to just ignore it if it's not a Pi4..

procount, 2019-12-02:
Good catch. I will update it the documentation.

Why not check if /proc/device-tree/model contains "Pi4" ?

closed 2019-12-02 by matthuisman