procount/pinn #343: unattended install using wifi

eroom1966, 2019-12-07:
Following from #328 I am trying to perform the same install over WiFi rather than Wired
I create the following in the /recovery directory
wpa_supplicant.conf

           country=GB
            ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
            update_config=1
            network={
                scan_ssid=1
                ssid="myssid"
                psk="mypsk"
            }

But it fails to install, so I presume the WiFi has failed to connect
The wifi connects perfectly with some config during normal operation

any clues ?
Do I need to add more command line parameters, or stall the process until the WiFi link is available ?

thx
Lee

procount, 2019-12-07:
It shouldn't make any difference if it is wifi or wired.
Please check your wifi works with PINN before doing attempting a silentinstall.

Imperas, 2019-12-10:
> It shouldn't make any difference if it is wifi or wired.
> Please check your wifi works with PINN before doing attempting a silentinstall.

OK, understood, I will try this.
Can I confirm I am correct in the syntax I have used for wpa_supplicant.conf file, and the location in /recovery ?

Thx
Lee

eroom1966, 2019-12-10:
> It shouldn't make any difference if it is wifi or wired.
> Please check your wifi works with PINN before doing attempting a silentinstall.

OK, understood, I will try this.
Can I confirm I am correct in the syntax I have used for wpa_supplicant.conf file, and the location in /recovery ?

Thx
Lee

procount, 2019-12-13:
It looks ok to me and you say it works under normal circumstances, so it should be ok.
If the file is correctly named & located, then at bootup, PINN will copy it to /settings and rename your file to wpa_supplicant.conf.bak.

I'm assuming this is completely headless?
If it is connecting to the wifi correctly, then the RPi will be getting an IP address. You should be able to find this in your router's DHCP table listing all conencted devices.
If you enable ssh in PINN, then you can connect over the wifi to it and find out why it is not working by looking at /tmp/debug.

Or if you can get a screen and keyboard connected locally, you could diagnose it that way,

eroom1966, 2019-12-14:
Thanks
Tried with screen and keyboard, and actually, I cannot do an install using pinn lite
The install icon does not become active after selecting the raspbian install
Now im confused, seems I have a more general issue
After I have installed an os I can easily config the wifi and disconnect the wired

procount, 2019-12-14:
With keyboard and screen connected can you click the wifi button and connect to your wifi that way, rather than handcrafting it manually?

eroom1966, 2019-12-14:
Yes I tried connecting kb/mouse
Not sure I understand if I am connected, it seems to as I get message about releasing dhcp

procount, 2019-12-14:
Ok.
Boot with screen connected. Press shift as necessary to enter recovery menu.
Watch the toolbar. The wifi icon should go from grey to coloured when it recognises your wifi.
Then watch the title bar. When it connects, the IP address will be displayed.
If you don't get an IP address, click the wifi button and select your ssid from the list. Enter the password and connect.
To check wifi is connected you can click the help button to open the web browser at the help page.

eroom1966, 2019-12-14:
OK, putting in the WiFi details the first time gave me this error:

[image]

I re-burnt my disk and it worked!
Having tried a couple of times, my results seem a bit non-deterministic.

 - is it possible there is a race here between when the network is fully established, and when the network file read occurs ?
BTW, my command line looks like this, could this be ab issue ?

> runinstaller silentinstall select="Raspbian,waitnetwork" no_default_source disableusbimages disablesdimages no_update alt_image_source=http://pilotaware.lode.co.uk/downloads/OGN/net/os_list.json ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline reserve+=800

thx
Lee

procount, 2019-12-14:
You could try adding networktimeout=30 to increase the wait time. Vary the value as required.

eroom1966, 2019-12-14:
@procount 
I seem to have made progress
I was using a slightly older version of PINN
Have updated and can now proceed to install silently over wifi - hoorah !

Unfortunately it looks like the remote reinstall is broken 
If I delete all files in /recovery, then unpack all the files (from the first install), on reboot I get no further than the rainbow screen

This is unrelated to wifi, as I get the same behaviour over eth0

eroom1966, 2019-12-15:
@procount 
So summary is as follows

With latest version 3.3.3 ?
From here
http://downloads.sourceforge.net/projects/pinn/pinn-lite.zip
The WiFi connection and install is resolved- excellent!

Unfortunately, with this release I can no longer perform a remote reinstall, by unpacking the contents once again into the /recovery, I just get a rainbow screen
I am surprised at that, but not sure how to investigate or resolve 

Thx
Lee

procount, 2019-12-15:
Sorry, I must be dim today, but I don't understand what you are doing with a remote reinstall. What are you unpacking, where, why, how etc...?

eroom1966, 2019-12-15:
@procount 

Hahaha no problem, this has turned into a long thread 😳

My installations are remote, but I can access over vpn/ssh

Occasionally I need to perform a complete clean (unattended) reinstall
This is done by logging in
Removing contents of /recovery 
Extracting contents of my personalised pinn installation 
Rebooting

This then performs an unattended reinstall
This has been working perfectly over a wired lan

I also have the requirement to do the same over wifi, this is where I cane unstuck
I found the latest version in sourceforge fixed the wifi issue, but broke my original issue of unattended remote install 

Maybe my original issue was not resolved in the correct manner, but I think this explains my requirements

Thx in advance 
Lee

procount, 2019-12-15:
Are you deleting the /settings partition?

eroom1966, 2019-12-15:
No I am not, when booted under Linux, I dont think I see that mounted
Should I mount manually and remove ?

procount, 2019-12-15:
No!
I thought you removing it would be causing you the problem.
I can only think that there is something different about your "customised PINN installation" that is failing.

eroom1966, 2019-12-15:
@procount 
I think I was messing up - pretty sure I have this working as intended now
over wired and wifi

Thx
Lee

eroom1966, 2019-12-15:
@procount 
OK, almost there !
I have one more issue.

It appears I am getting a copy of /recovery/wpa_supplicant.conf appearing in /boot/wpa_supplicant.conf
I have no idea how that is getting there, I am assuming you may be copying this in during the install ?

This then seems to conflict with my network setup under normal operations

When my OS is re-installed I have a a /etc/network/interfaces file which looks like this
 
> 
> source-directory /etc/network/interfaces.d
> 
> auto lo
> iface lo inet loopback
> 
> iface eth0 inet manual
> 
> auto wlan0
> allow-hotplug wlan0
> iface wlan0 inet dhcp
> wpa-ssid XXXXXXXX
> wpa-psk YYYYYYYY

So I have to perform 
sudo ifdown wlan0
- which complains about already being configured
then
sudo ifup wlan0
- which complains about trying to "/bin/mv /boot/wpa_supplicant.conf"
I have no idea what is happening here - any ideas ?

I can probably work around by using the wpa_supplicant.conf to define the WiFi network, but not sure where /boot/wpa_supplicant.conf comes into play

thx
Lee

procount, 2019-12-15:
Check the partition_setup.sh file for your OS.
That normally copies wpa_supplicant.conf from PINN to the newly installed OS so that wifi works straight away,

procount, 2019-12-15:
Unless you have a really old distro, the current recommendations are to leave /etc/network/interfaces alone and specify your wifi in wpa_asupplicant.conf

eroom1966, 2019-12-15:
Yes agreed, trying that now ...
Thx
Lee

lurch, 2019-12-16:
See https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/headless.md

closed 2020-04-02 by procount