procount/pinn #344: Multiple destinations installs

franferri, 2020-01-01:
I have a RPi 4. An USB SSD enclosure and an SD.
I would like to install 1 OS in the USB and one in the SD.

I can do either, but not both. So, I can install 2 OS in the SD but if I install in the USB, the boot partition disappear.
I can do the opposite. I install in the SD card one then one in the USB and the boot partition disappear.
 
Is there a way of having installed in 2 different destinations different os and keep the boot menu?

procount, 2020-01-01:
Short answer: No. 
NOOBS & PINN are designed to manage 1 storage device only.
If you boot from the SD card, but install to a USB device, then the boot partitions are stored on the SD card but the rootfs are stored on the USB.

If you want to install more than 1 OS, you must install them all at the same time. You cannot install a second one later. The only way to do this is to install multiple empty ProjectSpaces and then replace them with OSes at a later date. Using the Install feature a second time replaces everything that was on the storage device previously.

Slightly longer answer: Yes, if you understand how it all works and you know how to: fiddle about manually with the configuration files, create, format and move partitions and their references.
To do this, you would have to install both OSes to the SD card, then copy the rootfs of one of the OSes onto the USB SSD and fixup the cmdline.txt, /etc/fstab and the installed_os.json file to reference it correctly. 
When you have this working, you can delete the rootfs on the SD card and maybe resize the partitions to make better use of the card.
But this is totally unsupported and if you don't understand any of this, I wouldn't even attempt it.

franferri, 2020-01-02:
I missed the fstab. I just fixed the cmdline.txt config file. Will do it
tomorrow. I had it booting, the one in the Ssd but it went in  emergency
mode.

lurch, 2020-01-03:
I guess when the RPi4 firmware is updated to support booting directly from USB, this will become a non-issue, because you won't need to use an SD-card to boot the OS stored on the USB drive? :slightly_smiling_face:

franferri, 2020-01-03:
Something tells me this is not going to happen. If it was meant to happen,
will be happening already, RPi4 has been around for a while now.
I love PINN btw.

lurch, 2020-01-03:
>> I guess when the RPi4 firmware is updated to support booting directly from USB

> Something tells me this is not going to happen.

Don't worry, it **will** happen, there's just no ETA yet https://github.com/raspberrypi/rpi-eeprom/issues/28

procount, 2020-01-03:
> I guess when the RPi4 firmware is updated to support booting directly from USB, this will become a non-issue, because you won't need to use an SD-card to boot the OS stored on the USB drive?

I'm not so sure. If that were the case, why has OP not installed both OSes to the USB drive?

@franferri  - Why do do you need an OS on each drive? Is the one on the SD card just a backup for those occasions when you don't have your external drive? 🤷‍♂

franferri, 2020-01-03:
This is specific to my use case.

I want to boot from the SD with PINN, and timeout to the SD OS (Raspbian
Lite) that drives a robot. The RPi is in the robot.
Then I have an external USB drive where I have the development environment
which I connect when I need to do changes. I discovered RPi.GPIO library
(python) is not possible to build in mac or windows, so if you want a
proper development environment you have to use RPi itself, and honestly the
4gb with external SSD HD works very well.

When I finish the changes in the SSD I did commit and restart in the SD
mode (unplug the SSD), the robot first boot git pull the changes and
proceeds to work

As simple as that.

procount, 2020-01-03:
Ah, that makes sense now. Have you made the SD Raspbian Lite the "Sticky default" then?

franferri, 2020-01-03:
Yes :D

lurch, 2020-01-03:
> If that were the case, why has OP not installed both OSes to the USB drive?

I was assuming that he wanted to use **all** the space on the SD card for one OS, and **all** the space on the USB drive for the other OS. But @franferri has now clarified that it's more complicated than that :wink: 

> I discovered RPi.GPIO library
(python) is not possible to build in mac or windows, so if you want a
proper development environment you have to use RPi itself

Depending on what you're doing, you may find https://gpiozero.readthedocs.io/en/stable/remote_gpio.html useful :slightly_smiling_face:

franferri, 2020-01-03:
Thanks! I knew remote gpio, but in this case I'm using the local one.

closed 2020-01-15 by procount