procount/pinn #347: New images for Raspberry pi 4?

rhunjadi, 2020-01-13:
Lineage OS 16 for RP4 
It's been released. Can you add it to the raspbarry pi 4 PINN image?

https://konstakang.com/devices/rpi4/LineageOS16.0/

Thank you very much!

procount, 2020-01-13:
The last time I tried to add LineageOS16 I had some problems. I may try again when I have some spare time.

I have to convert each image to NOOBS format manually, which takes a while, so I'm not prepared to do that on a weekly basis, I'm afraid.

rhunjadi, 2020-01-13:
Thank you for your quick replay. 

Hope u'll have succes with LineageOS16.

For the second one I was just having in mind the latest version for RP4. It's no problem...I'll just wait for the final relese or Retropie or try to convert one of the images manualy. It shoudn't be that hard. I understand your pain about weekly continuous (manual) integration process... :)

Norde, 2020-01-14:
You can easily convert a classic image to a noobs/pinn one:

1) Flash system img on a sd-card
2) note size and name of boot and root partition (in Mb) (use > df -h)
3) Tar root partition : 
> sudo tar -cvpf ~/root.tar /mounting_point/to/ROOT-partition/

4) Do the same for boot partition (as tar command doesn't handle well FAT partition, you must first copy boot directory: 
> mkdir ~/boot
> cp -r /mounting_point/to/BOOT-partition/ ~/boot/
> sudo tar -cvpf ~/boot.tar ~/boot/

5) Note the size of created files root.tar and boot.tar (use > ls -lh)

6) Compress both .tar files:
> sudo xz -9 -e -v ~/root.tar ; sudo xz -9 -e -v ~/boot.tar

7) Now using existing noobs/pinns system image as exemple, put created boot and root.tar.xz and edit "partitions.json" file:
- For "label" input, write corresponding one to root and boot partition noted in 2)
- For "partition_size_nominal" size noted in 2) (You can round up)
- For "uncompressed_tarball_size", size of .tar file noted in 5) (You can also round up)

8) Now edit "os.json" file to personnalise title, description and compatibilities

procount, 2020-01-15:
Finally, some success with LineageOS16. Still some work to do, but it's looking more promising now.

rhunjadi, 2020-01-15:
First of all I would like to thank you for your instructions. Ill try them tonight. 
As for the Lineage...You see the light at the end of the tunnel. I really hope that this light is the sun...and not the train! :)

Norde, 2020-01-19:
I updated instructions (previous ones were dated).

procount, 2020-01-25:
LineageOS 16.0 has now been added to PINN

closed 2020-01-25 by procount

mjcc30, 2020-11-20:
Hello ! Good work ! 👍 
There are other images available for lineagos here : 
https://konstakang.com/devices/rpi4/
is it possible to add the android tv version ?
Thank you for your work

procount, 2020-11-20:
I've been working on it. There have been some quite complex changes, but nearly there.

matthuo333, 2021-04-28:
Can I add lineage18.1 (Android 11)? Because when running lineage17.1, when the wired network is not plugged in, the system always restarts. Not in 18.1. For this issue, the related lineage thread has been confirmed.
Or can you tell me how to customize and add a new system?

commit 854be817 (2017-04-23): Pi 3: fix ACT led

Add missing CONFIG_GPIO_BCM_VIRT=y kernel config option.

Closes #347
Closes #349