procount/pinn #348: Sending CEC command with PINN

Norde, 2020-01-15:
While testing a CEC compatible TV I wondered if it could be possible for PINN to orders other devices  using cec-client (at pinn boot to power on devices or switch video/sound sources).

Would including cec-client at buildroot suffice?

If yes I could add a script to pinn_init.sh using config.txt for variables (and update the documentation accordingly).
A simple command like "echo 'cec-code' | cec-client" should suffice.

procount, 2020-01-15:
Maybe. Not sure if it will conflict with PINN's use. Try it and see.

Norde, 2020-01-15:
I need to familiarize myself with the builroot first.
I come back when I have done the tests :)

procount, 2020-01-15:
BTW, I didn't include libcec/ cec_client in PINN for possible licensing reasons.

Norde, 2020-01-16:
According to the official website, libcec  is dual licensed GPLv2/Commercial
http://libcec.pulse-eight.com/

Norde, 2020-01-16:
Indeed.
Thank you for the details and sorry for this questions which had already been answered 😟

closed 2021-01-13 by procount