procount/pinn #349: How to set the language in recovery.cmdline to get US instead of GB

franferri, 2020-01-21:
Hello, modifying the file recovery.cmdline for language=us don't boot PINN with US selected, but rather GB.
Keyboard=es works.

How do I setup the US instead of GB for PINNs boot?

Thanks!

procount, 2020-01-21:
See https://github.com/raspberrypi/noobs/issues/179

franferri, 2020-01-22:
Yes I saw that, but is still not working, PINN last version, language=us is not working at all, I'm using vnc to access to it, and the default are language=British keyboard=es

In a nutshell, language=us don't work, keyboard=es works
I see is a long term bug.

Did you point me to the case because is supposed to be fixed?

procount, 2020-01-22:
> Did you point me to the case because is supposed to be fixed?

No, but because it explains the problem. I don't think anything has changed since then (unless Lurch knows different)
There is no US language for the text, it defaults to UK.

Twice you mentioned "keyboard=es", which is Spanish. Is that what you meant?

The referenced issue mentioned that adding "keyboard=US" _did_ work to change the keyboard layout. have you tried that?

franferri, 2020-01-22:
Yes, Keyboard=es works perfectly.
Language=us dont select US, will be nice to have something like en_us,
since the PINN languages list has US.

The system installs always in en_gb.

Thanks for all the information.
Not good, but not terrible :D

lurch, 2020-01-22:
I still think you're misunderstanding https://github.com/raspberrypi/noobs/issues/179
There **isn't** a US language (in NOOBS or PINN). Selecting the USA flag in the PINN GUI just selects (British) English as the language and a US keyboard layout.

franferri, 2020-01-22:
> I still think you're misunderstanding raspberrypi/noobs#179

Indeed, sorry.

You are right. Thanks for the clarification.

Will be nice to have a way of choosing US tho, since is presented in the
list.

Thanks for the clarification again. I did jump to conclusions too fast.

lurch, 2020-01-22:
If @procount was feeling clever, he could modify the initialisation of the Language-dropdown in PINN so that if the language was `en` and the keyboard was `us`, then it could select the US "language".
But I think even that wouldn't help in your case though, because you're explicitly setting a keyboard of `es` ?

To clarify: I _think_ as far as PINN is concerned, a Language of UK English and a `es` Keyboard will be treated identically to a Language of US English and a `es` Keyboard?

commit 854be817 (2017-04-23): Pi 3: fix ACT led

Add missing CONFIG_GPIO_BCM_VIRT=y kernel config option.

Closes #347
Closes #349
closed 2021-01-13 by procount