procount/pinn #350: Is Void Linux supported?

drwhite729, 2020-01-22:
Can PINN install Void Linux?
* I found Void Linux from https://distrowatch.com/search.php, which directed me to https://voidlinux.org/ (the only active distribution for the Raspberry Pi that supports xfce, has a rolling release model, and doesn't use systemd)
* Void Linux download instructions for the Raspberry PI are here: https://wiki.voidlinux.org/Raspberry_Pi
* Void Linux binaries built for the Raspberry Pi are here: https://alpha.de.repo.voidlinux.org/live/current / void-rpi*
* Void Linux instructions say to prepare an SD card with one FAT partition for /boot and one ext4 partition for /, which is exactly how the Raspbian install files are organized

So, it seems like it should be possible to install Void Linux with PINN.  Indeed, it looks like PINN is at least aware that it exists:
* My RECOVERY partition has a recovery.cmdline file that includes: repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json (which looks the same as what it would be if I were to reinstall PINN)
* that repo_list.json file refers to an os list named PINN located here: http://raw.githubusercontent.com/procount/pinn-os/master/os/os_all.json
* that os_all.json is quite long, but I found entries for os_name: void1, void1_mus1, void2, and void2_mus1; the details for those entries look close, but not quite right
  - the url for all of them is http://http://www.voidlinux.eu, not https://voidlinux.org
  - the download files are coming from https://sourceforge.net/projects/pinn/files/os / void*, and they exist there, but they look very old - from 2019
  - I know some of those files are PINN-specific, but I would have expected the actual *.tar.xz files to be pulled from https://alpha.de.repo.voidlinux.org/live/current 

However, whether the os_all.json list is up to date is actually a secondary issue -- the primary one is that when I boot into PINN and browse the list of operating systems to install, Void (of any kind) isn't in the list.

I have a Raspberry Pi 3 Model B Plus, Rev 1.3 (according to PINN).  Should that be working, or do I need to install Void Linux the hard way?

procount, 2020-01-22:
Welcome back!
For the full list of OSes that PINN supports, see http://pinn.mjh.nz, which is occasionally replicated to my wiki at https://github.com/procount/pinn/wiki/OSes-supported-by-PINN.
Void linux is supported, but the last version I converted was in 2017, which predates the 3B+, so I need to update my void images. If you can't wait, you'll have to do it the hard way.
IIRC, Manjaro is another Arch based distro which supports xfce, but not sure about all of your other criteria. I should be supporting that soon too.

procount, 2020-01-22:
In fact, it looks like I was preparing an update last December.

If you want to try it out, create a file on your PINN sd card called `os_list_v3.json` and copy the following contents into it:

```
{
    "os_list": [
        {
            "os_name":	    "void3",
            "description":  "A Void Linux port for ARM devices RPi3",
            "release_date": "2018-11-11",
            "feature_level":   0,
            "supported_hex_revisions": "1040,1041,2082",
            "supported_models": [
               "Pi 3"
            ],                
            "supports_usb_root": false,
            "supports_backup":  "true",
            "url":              "http://http://www.voidlinux.eu/",
            "group":            "Minimal",
            "download_size":    63208864,
            "os_info":          "http://downloads.sourceforge.net/project/pinn/os_next/void3/os.json",
            "partitions_info":  "http://downloads.sourceforge.net/project/pinn/os_next/void3/partitions.json",
            "icon":             "http://downloads.sourceforge.net/project/pinn/os_next/void3/void3.png",
            "marketing_info":   "http://downloads.sourceforge.net/project/pinn/os_next/void3/marketing.tar",
            "partition_setup":  "http://downloads.sourceforge.net/project/pinn/os_next/void3/partition_setup.sh",
            "tarballs": [
                                "http://downloads.sourceforge.net/project/pinn/os_next/void3/boot_void.tar.xz",
                                "http://downloads.sourceforge.net/project/pinn/os_next/void3/root_void.tar.xz"
            ],
            "nominal_size":  453
        }
    ]
}
```
It might be easiest to put your PINN sd card in a Windows PC and create the file there (do not accept any suggestions from Windows to format any partitions!)

zen0bit, 2021-11-09:
New versions of Void for all rpi (rpi, rpi2, rpi3, rpi4) is out...

Can be added to PINN please?