procount/pinn #351: Bypass OS installation menu?

pandalatte, 2020-01-22:
Running latest PINN on Pi 3.

When I press SHIFT and go to recovery to boot into another (already installed) OS, I'm being presented with the OS installation menu.
I don't want to install any additional OSes :)
Is there a key to disable it and to fast forward to the Boot menu? Thanks!

procount, 2020-01-22:
So, you have more than one OS installed.
If you do not press any keys at startup, then I guess it just boots into one of your OSes?
You need to press shift to get the recovery menu, but really you just want the boot menu?

Well, if you have more than one OS installed, then PINN should present the boot selection dialog at startup everytime, but if it isn't then you've done something.

If you go to the boot selection menu, do you have any of the checkboxes ticked to the left of any of your installed OSes? My guess is you do. This is the sticky default flag that causes PINN to boot directly into an OS unless you press the shift key at startup. Uncheck the box and it should work as expected.

pandalatte, 2020-01-23:
Yes, I run 2 OSes: Raspbian and LibreELEC.

Raspbian is set to sticky.
When I want to boot to ELEC, I have to:

1. press shift (to recovery)
2. press escape (to skip OS installation prompt)
3. select ELEC from the boot menu
4. boot to ELEC

All I want is to *exclude* steps 1 and 2.
Unfortunately, if sticky is not set, PINN boots to the last known OS.
I want to keep Raspbian as the default OS to boot.  Maybe there's an option for that?

procount, 2020-01-23:
There's no option for that at the moment, mainly because there is no way back from the bootselection dialog to the recovery menu.
Do you really mean to exclude step 1) pressing the shift key? If you do that you will always get the boot menu as if you had no sticky.

Or do you just want to change the boot behaviour to always be the default sticky instead of the last os booted?

pandalatte, 2020-01-23:
> Or do you just want to change the boot behaviour to always be the default sticky instead of the last os booted?

Yes, this.
What if I:
1. Remove the sticky
2. Let it boot to Raspbian once (to remember the last booted OS)
3. Set os_config.json or wherever it stores it to READ-ONLY?

Would it honor the R-O attribute ? Gotta try.

procount, 2020-01-24:
You can try, but I think it will mean it will always boot to Raspbian

procount, 2020-01-29:
After installing 2 OSes, the standard PINN boot behaviour is as follows:

1. Option to press shift to enter recovery menu
2. If no shift key pressed, Shows boot menu to select OS to use.
3. If no user input, PINN chooses last OS booted.

With a sticky default set, the idea is to bypass this boot menu since you mostly want to boot one preferred OS, but occasionally you want to boot a second OS in preference. The idea is to speed up the boot process for the normal case, so the use case is as follows:

1. Option to press shift to enter recovery menu.
2. If no shift key pressed, boots into Sticky OS.

To select the second OS:

1. Option to press shift to enter recovery menu.
2. When shift key pressed, enter Recovery menu.
3. Press escape to skip recovery menu and go to boot selection menu
4. select & boot 2nd OS.

To avoid the 2 button presses of shift and escape to select the 2nd OS, I could envisage the following work flow:

1. Option to press shift to enter recovery menu
2. If no shift key pressed, Shows boot menu to select OS to use.
3a. If no user input, PINN chooses the STICKY DEFAULT OS to boot.
3b. If use clicks on 2nd OS, that is booted instead.

So this would delay the startup process as PINN has to wait with a timeout for you to do something to vary the normal boot procedure, but it means less clicks or button presses.
Does this alternative mode sound like what you would like to do?

lurch, 2020-01-29:
> So this would delay the startup process as PINN has to wait with a timeout

This would be a change to PINN's current behaviour, which has the potential to upset existing users?
What if you kept PINN's current behaviour by default, but instead added an option (in cmdline or similar?) to then trigger the new behaviour you're proposing? (i.e. the sticky default OS having a menu with a timeout, rather than booting immediately)

procount, 2020-01-29:
Yes, that's what I was thinking, but I just wanted to check the procedure for this alternative method was ok first. Got to think of an appropriate name for this setting next.

hupfdule, 2021-01-11:
I'd like to add a comment as I "suffer" from the same problem.

I have multiple OS installed. One of them is set as sticky, as this is the default that should be booted most of the time.

Sometimes, I want to select another OS to boot from my list.

Very rarely I want to to install/reinstall an OS.

The current behaviour means, that the rare event of installing/reinstalling an OS (via recovery menu) is favoured by PINN over the more frequent event of wanting to boot a secondary OS.
It should be the other way around.

My expected behaviour for the above mentioned use case is:

- With no use interaction: Boot to sticky OS
- If shift is pressed on boot: Display boot menu
- Allow switching from boot menu to recovery menu

The last two points are currently switched. I think this comes from the assumption that most users would leave the boot menu enabled (without any sticky default) and therefore would only want a way to display the recovery menu _instead_ of the boot menu.

My proposal would be to add a possibility to switch from the boot menu to the recovery menu to allow the above mentioned use case. I would leave the option to switch from the recovery menu to the boot menu in as this would be handy, too.

As this would be a change of behaviour to existing versions of PINN, this could be made configurable.

procount, 2021-01-11:
The problem is that with the existing architecture inherited from NOOBS, the only way to get from the Bootselection dialog to the recovery menu is via a reboot, but I could probably make it configurable so that shift goes to one or the other.

redm123, 2021-05-08:
I just wanted to create a ticket with about the same suggestion as @hupfdule. 

80% boot the same OS -> sticky direct boot
19% boot an alternative OS -> press Shift or whatever to show bootmenau
1% some actual recovery/admin/config task -> enter recovery mode UI

It seems a bit awkward that the most common thing after the booting the sticky OS always sends you through the recovery UI.
Actually I also found this somewhat confusing the first time: a) it's not self explanatory how to boot a non-sticky OS, and then b) the documentation says I need to enter recovery mode to boot an alternative OS (or remove stickiness). So there I was, searching back and forth, and couldn't figure out how to remove the stickiness or boot something else. I only found out by chance when I left the config UI... The suggestion from @hupfdule would solve that, too.
Also keep in mind that there may be other users for certain PINN installation (e.g. in the Pi-connected-to-TV case), who should be able to just use it. Without having to figure the above out or having to read the PINN documentation...

So much for the problem.. solving seems somewhat more difficult, the way NOOBS works..

If there is no other option, it would probably boil down to always showing boot menu and just sticky select a certain OS (basically like all other boot managers do). That would cater the ease-of-use for the average user. For people who want to further optimize boot time add a "no boot menu" option to explicitly hide the menu, with all implications (so explicitly re-enable the way it works now).

> This would be a change to PINN's current behaviour, which has the potential to upset existing users?
> What if you kept PINN's current behaviour by default, but instead added an option (in cmdline or [...]

Well, generally, not upsetting users is a good thing of course. But if a change is meaningful, reasonable, good then it should be done. Possibly upset users should not be an excuse to keep suboptimal or broken behaviour forever. In this case I would make showing the menu with a timeout the default (i.e. current normal behavior, just with a sticky pre-select), as this likely affects more users and users less familiar with PINN. For the possibly "upset" (power) users allow an override to restore the (somewhat destructive) direct-boot-without-menu behaviour.