procount/pinn #352: noobs translation languages alignment

bovirus, 2020-02-06:
@procount 

Checking noobs and pinn project about recovery language files I found some differences

noobs use "ar" and pinn use "ae". Is it the same languages?
noobs include "gl" language (Galixian) and pinn not have it. Could you please include it?
noobs use "zh_Hans_CN.ts" and pinn seems not have it ("zh_CN"). Could you please include it?
noobs use "zh_Hans_TW.ts" and pinn seems to use "zh_TW". is it the same?

Why these differences about language code?

For the new languages please also add the relative icons getting them from noobs project.

Thanks.

procount, 2020-02-06:
Looking in NOOBS at the flag for ar country code and the translation, I think it refers to United Arab Emirates, which has the iso code "AE", not "AR". AR is for Argentina. So I think I corrected it in PINN.

gl is new and only just added. I think it is another Spanish dialect. But gl is the country code for Greenland....

I'm not sure about the chinese ones. I think they refer to the same languages, but I tried to keep the naming consistent.

bovirus, 2020-02-06:
@procount 

Checking on this list

https://stackoverflow.com/questions/7973023/what-is-the-list-of-supported-languages-locales-on-android

ar_AE = Arabic (United Arab Emirates)
gl = Galician
gl_ES = Galician (Spain)

then ae -> ar_AE
gl is ok.

"ast" is asturian and is not in the list.

procount, 2020-02-06:
"ast" is treated as a special case.

bovirus, 2020-02-11:
@procount 
Ok for ast.

For **_recovery_ae.ts_** I suggest to rename this file **recovery_ar_AR.ts** and modify it inside as

language="**_ar_AE_**" sourcelanguage="en"

and modify the **_recovery.pro_** to use **recovery_ar_AR.ts**

to follow the standard

https://lh.2xlibre.net/locales/

For asturian I follow the same ruels as ast_ES but it didn't work.

ykla, 2024-06-13:
Hi,

Please delete the current `zh_Hans_CN.ts` and `zh_Hans_TW.ts` files. A more i18n-compliant naming might be `zh_CN` and `zh_TW`. Currently, the project has two `zh_TW.ts` files, namely `translation_zh_Hant_TW.ts` and `translation_zh_TW.ts`. Please retain only the file designated as "translation_zh_TW.ts" and delete the duplicate file designated as "translation_zh_Hant_TW.ts." It is recommended that you synchronize English updates for "zh_Hans_CN.ts" to display items that require translation. This can be accomplished by deleting "zh_Hans_CN.ts" and creating a new file designated as "zh_CN.ts," which I will translate.

 The Chinese language, which is written in simplified Chinese, differs from traditional Chinese, which is used in Taiwan and Hong Kong. Despite the similarities between simplified and traditional Chinese, there are notable differences in language habits, similar to the differences between American and British English. Chinese is a highly distinctive language, and the differences between it and other languages are considerable. The smallest writing unit in Chinese is the character, which is composed of several strokes. Due to historical reasons, modern Chinese has two writing systems: Simplified Chinese and Traditional Chinese are the two major varieties of Chinese. Additionally, due to regional differences, Traditional Chinese also has its own specific vocabulary and expression habits. Therefore, in general, internationalization (i18n) projects involving Chinese typically include three files related to Chinese: `zh_TW.ts`, `zh_HK.ts`, and `zh_CN.ts`.

procount, 2024-06-13:
Hi ykla,
Thank you for your valuable comments.
Despite having visited Hong Kong and China, I'm afraid my knowledge of the Chinese language is virtually non-existent and I have relied on the original translations from NOOBS and the kindness of volunteers to update & maintain them. 
I've just tried googling for further information and to understand your comments and discovered there are 2 common ways to represent Chinese localisations:
1. Use zh_Hans (simplified) or zh_Hant (traditional), or
2. Use zh_CN (simplified), zh_TW (traditional) or zh_HK (traditional)
These 2 approaches seem to be mixed up within PINN. 
I am more than happy to make the adjustments you suggested, especially if you are kind enough to update the translations.
Apologies if it takes me a while as I am in the midst of debugging Pi5 issues, so please keep an eye on my github for when the changes are made for you to update the translations.
Thank you for your support.

procount, 2024-06-16:
I'm not sure the zh_Hans_CN or zh_Hant_TW languages were actually used within the program.
Nevertheless I have restructured the translation files.
Please have a look at the i352_zh_translation branch