procount/pinn #354: WIndows 95

actraiser30, 2020-02-20:
Hello great to add Windows 98 in the Operating systems, is it possible to add Windows 95 too?

thanks

procount, 2020-02-20:
I think someone created a Win95 demo in 2012, but I don't have an image. If you know of one, I could put it up.
It might be interesting to see if either of these run respectably on a Pi4 these days.

actraiser30, 2020-02-20:
there are instructions in this page under but i cannot make it work since i cannot undertand a lot of things if you can make it will be great

http://sbcgaming.wikidot.com/wiki:win95

thanks

procount, 2020-02-20:
Sorry, but it takes me all my time to maintain the OSes I do have images and conversions for, and add new features to PINN, without having to create the image as well.

actraiser30, 2020-02-20:
ok no problem if you have time after maybe you can try it thanks

lurch, 2020-02-20:
Out of curiosity, why do you need Windows95 rather than Windows98 ?

actraiser30, 2020-02-20:
i have some old games that i was playing back then that i want to try them

HexEvee, 2023-01-28:
Hi. I saw this, and, you said you had windows 98. I never saw this in the menu, could you tell me how to get it?

procount, 2023-01-28:
It's still there under the testing category, but you won't see it if you have a Pi4/400 as it is not compatible. .
It is an old image and therefore only for Pi3 and below.