procount/pinn #355: Problem with DSI HDMI switching

MJ76, 2020-02-23:
Hi

I just downloaded PINN due to is great "auto-switch HDMI/DSI screen " but I have some problems with using it.

My configuration:
RaspPi 4B 4G with attached RaspPI 7" Touch Display and a TV attached via HDMI

My Goal:
If the TV is On during booting Kodi should be displayed on the TV for watching movies. If the TV is OFF during booting Kodi is displayed on the RasPi DSI Touchscreen for Internet Radio, Music,...

I would prefer not to do it via booting but as I read in different forums, this is the only way it currently works.

Anyway. I installed PINN and according to your readme, it should do the job. However, I have problems to configure the PINN correctly: I have seen that there are already some config versions installed, and I dont find the "screenswitch.zip" in your repository, plus there is already a DSI option point in the config of PINN. Hence, I guess you improved this feature and your readme is outdated. Can you give me a short description how I can configure PINN correctly to get to my goal?

Sorry for the troubles.

Thanks a lot. All the best.

procount, 2020-02-23:
You will find screenswitch.zip on sourceforge.
Here is a direct link to download it -> https://sourceforge.net/projects/pinn/files/screenswitch.zip/download

The readme instructions should still work the same, so have another try now you have screenswitch.zip. 

I haven't actually tried it with the new Pi4, but as PINN uses the legacy drivers, it should still work the same.

procount, 2020-02-23:
Oh wait - you confused me with reference to screenswitch!
DSI and screenswitching are 2 different concepts.

For DSI switching priority, you just need to read this section -> https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-auto-switch-hdmidsi-screen-configurations You don't need screenswitch.zip for this. Nor do you need to use the config files - they are to do with screenswitch. You may as well delete them and start again with a new config.txt file that just contains `ignore_lcd=1`.

(The screenswitching is described in the following section -> https://github.com/procount/pinn/blob/master/README_PINN.md#automatic-screen-detection)

MJ76, 2020-02-29:
Thanks a lot! It works now :-)

One more quick question: Is it possible to shorten the 10 seconds waiting time?

lurch, 2020-03-07:
> One more quick question: Is it possible to shorten the 10 seconds waiting time?

https://github.com/procount/pinn/blob/master/changes.md#os-boot-selector ?

procount, 2020-04-02:
As above.

closed 2020-04-02 by procount