procount/pinn #357: Ubuntu Server official Raspberry Pi versions (armhf, arm64)

Lonerider2010, 2020-03-11:
As there are official Raspberry Pi versions of Ubuntu Server (see https://ubuntu.com/download/raspberry-pi), please add them to PINN.

procount, 2020-03-23:
I already have some older Ubuntu versions and I will update them in due course.
Unfortunately, they use U-Boot, which means I have to adapt them to make them suitable for multi-boot, which takes a little time.

Ae3NerdGod, 2020-08-11:
I've got all armhf versions of latest Ubuntu installing and booting reliably stable, they have self extracting kernels. And from what I read I don't think they'll be changing that for armhf.

Arm64 I got to boot, and update to latest kernel fine, problem is the initial dist-update that runs at first boot in the background updates the kernel and ignores post-dpkg commands. Currently trying to perfect a daemon to watch vmlinuz for changes and extract it to vmlinux automagically.

It _almost_ works fine at the moment lol.

@procount couple questions:
Should Ubuntu mate betas be in the testing group or general?

What about arm64?

Should I separate the os_lists on my GitHub for the different distros, ie os_ubuntu, os_manjaro, os_twister, and make a repo_list for all of them? Idk what would be easier for you if you decide to include them in the main repo, nor the legality of twisterOS. Let me know.

procount, 2020-08-11:
The testing group is for beta versions, OSes without complete slideshows, or otherwise untested OS conversions, whether they be arm64 or armhf.

Just put them all in one os_list_v3.json file

Ae3NerdGod, 2020-08-11:
Will do, and what do you consider a "complete" slideshow. Are those from the distro people? Do I need to request them from the people who make the distro or can I just make them myself by downloading images from their websites?

procount, 2020-08-11:
Yes.
You'll be lucky to get the distro makers to do it for you.
Usually a few slides of features and benefits are sufficient.

Ae3NerdGod, 2020-08-11:
Alright. Just as a reminder, both manjaros and twisterOS are done with marketing at 
https://raw.githubusercontent.com/Ae3NerdGod/PINNg/master/os_list_public.json

Let me know if I need to change or fix anything about them.

procount, 2020-08-25:
@Ae3NerdGod - Any further progress on a Ubuntu conversion?

Ae3NerdGod, 2020-08-25:
Yes, everything but Ubuntu mate 18.04 beta 1 versions specifically, as far as I can tell they should boot, but I do not have the correct eeprom to test them or my pi3b+ so I won't be making them public. probably wait for a beta2 or 18.04.5 mate versions before adding those anyway. (I'd flash my eeprom, but this is currently my main way to access my desktop while I'm away from home, so I wanna keep my main OS SSD booting)

But yes,Just adding finishing touches; marketing and icons and typos and they should be available tonight or tomorrow actually depending on upload time.

I've also got working Manjaro 20.08 for all four flavors, i3, kde, xfce, and sway.

Ae3NerdGod, 2020-08-26:
strike that, ive done something that makes 18.04 not work, cant figure out what, fails to find the root partition on sdcards, usb still works. taking it off public for now
20.04, new manjaro, and twister1.5 all work properly

Ae3NerdGod, 2020-08-26:
would it be possible to specify additional txt files available as a tab when you edit an os from the maintenance mode? for ubuntus weird bt/nobt cmdlines and user/system cfg.txt's ?

procount, 2020-08-26:
I'll add it to my list.

Lonerider2010, 2021-01-28:
When I updated PINN, I could install Ubuntu 20.10 desktop on my Raspberry Pi 4. That works really fine.
So thanks a lot @procount for this support, it is great!