procount/pinn #360: LineageOS Installation Issues

eleding24, 2020-03-28:
How do I install LineageOS on rpi4, because when try to install lineageos, it says that I need another partition named "boot"

procount, 2020-03-28:
You'll have to be more specific about your problem.
I just installed Lineageos16_rpi4 on my Rpi4 and 7" RPF touchscreen and it installed fine.

eleding24, 2020-03-28:
I tried installing Lineageos16 on my rpi4 with two other oses (Raspbain Lite and Lakka) and the installation has an error needing the partition "boot"

procount, 2020-03-29:
Oh I see. I think that'll be because LineageOS requires the partition names to be preserved, and one of the other OSes uses the same partition name and is installed first.

A possible (untested) workaround is as follows:
Start again and install LineageOS along with ProjectSpace1 and ProjectSpace2.
Then replace ProjectSpace1 with Raspbian Lite, and ProjectSpace2 with Lakka.

lurch, 2020-03-31:
> LineageOS requires the partition names to be preserved

Oh dear! That'll break so many of the assumptions in PINN / NOOBS. I wonder if you could fix it to look for the correct PINN-assigned partition names with the partition_setup script?

procount, 2020-03-31:
It's not the only one. Ubuntu has also recently gone down this route of referencing partitions by labels.
I added an extra field to os.json to indicate those OSes that do this, so it will prevent PINN from installing the OS by renaming the partition labels. Maybe I could use this to change the installation order of OSes, but it still won't be guaranteed to install. It certainly prevents multiple installations of the same OS.
The partition_setup script could be used to fix them all up, but the trouble is, I have no idea how many references there are to these partition labels, apart from the usual cmdline.txt and fstab.

lurch, 2020-03-31:
Maybe it's worth reaching out to the distro-developers directly, asking them for assistance / guidance?  And don't forget to mention that PINN makes it easy for users to try out their different OSes :smiley:

procount, 2020-04-01:
My mistake!
LineageOS does not require fixed partition labels, so those constraints got added in accidentally (probably copied from Ubuntu).
I will fix it in the next release of LineageOS coming soon.

lurch, 2020-04-02:
Ahh, so it looks like @eleding24 was actually seeing the error message come from PINN itself, but @procount thought that they meant the error was coming from LineageOS (_after_ PINN had installed it)? :upside_down_face:

procount, 2020-04-02:
No, I knew they were coming from PINN  during installation, but I hadn't realised that I had put the constraints in by mistake.

Anyway, I have just released the Feb versions of LineageOS for use with PINN.
These should install happily with other OSes (Raspbian Lite, Lakka etc).
The TWRP Recovery mode is also fixed up for use with PINN.

closed 2020-04-02 by procount