procount/pinn #363: manjaro arm pi4

jonzen55, 2020-04-03:
pelase add manjaro arm if possible

procount, 2020-04-03:
I tried 2 versions of the 20.02 version on my Pi4 - KDE & xfce.
They booted up to the point where it asked me to select the keyboard layout, but none of my USB keyboards worked.
Strange because I'm sure a 19.08 version worked previously.

jonzen55, 2020-04-04:
[pd@pdrpi4 ~]$ screenfetch
                             pd@pdrpi4
                             OS: Manjaro-ARM 20.02
                             Kernel: aarch64 Linux 4.19.97-1-MANJARO-ARM
using 20.02 now with logitech usb wirelesss
also tried ipazzport mini keyboard  works fine

procount, 2020-04-04:
I tried another Pi4 and it worked. Must find out what the difference is.....

procount, 2020-04-05:
I've just released Manjaro 20.02 KDE & XFCE versions.
(Sourceforge download seems a little slow at the moment).

closed 2020-04-05 by procount

lurch, 2020-04-06:
The output above indicates that it's using an `aarch64` kernel, i.e. running in 64-bit mode. I can't remember all the details, but IIRC there's still some issues in some of the 64-bit drivers which means that a 1GB Pi4 sometimes works better in 64-bit mode? :man_shrugging: 
(or maybe that's just hearsay / outdated-info ? )

procount, 2020-04-06:
> The output above indicates that it's using an aarch64 kernel, i.e. running in 64-bit mode. I can't remember all the details, but IIRC there's still some issues in some of the 64-bit drivers which means that a 1GB Pi4 sometimes works better in **64-bit mode**?

Did you mean to say it sometimes works better in **32-bit mode**? (or am I misunderstanding what you are trying to say?)

But yes, this is a 64-bit distro. Manjaro have dropped all 32-bit ARM versions. I do have a copy of a 19.06 version for the RPi3, but I'm not sure if it is worth converting seeing as it is deprecated now.

If you are referring to my Pi4 that didn't work, it was a 1.4 B0 version, so maybe the firmware wasn't updated or something 🤷‍♂

lurch, 2020-04-06:
> Did you mean to say ...

Just edited my previous comment for clarity :slightly_smiling_face: 

> If you are referring to my Pi4 that didn't work, it was a 1.4 B0 version, so maybe the firmware wasn't updated or something

Yeah, I guess that's also a possibility? :man_shrugging:

procount, 2020-04-06:
Yes, it could be the bounce buffers.
Strange because it normally doesn't have a problem. I'll have to check the eeprom.