procount/pinn #364: PiNN "time" set menu

typxxi, 2020-04-06:
I am using PiNN with a "preconfiguration" from raspberry and language / keyboard to wpa_supplicant.conf which I had downloaded yesterday, so a complete new system.

If  I boot into PiNN and choose the option time (t) I get the current time displayed - but the UK one and not the german one even I am online. 

Could you adjust the time to that from the web ? 
Maybe you need to add a location/timezone=DE option in the recovery.cmdline to run the right timezone that can also be used for the booting OS ses like Raspbian. 

Maybe this feature was for "offline" installation some where remote ... and I missed the purpose but when I pressed it I was astonished about the "wrong" time 9:50 instead of 11:50 ...

Then I checked the language and keyboard layout for right flags ... that were already running.

At least it would be more consistent if there would be a feature in the file for "location"

lurch, 2020-04-06:
AFAIK the timezone in Germany is only 1 hour ahead of the UK, not 2 hours?
So perhaps PINN is in fact using "UTC time" rather than "UK time"? :man_shrugging: 

I believe this is actually the behaviour you'd want, because it means the timestamps on the files that PINN writes will be in UTC (which is the desired behaviour) and then the timezone-features in whatever OS you're using will then convert the UTC-timestamps into local time.

procount, 2020-04-06:
PINN only needs the time for 2 purposes:
1. to allow https connections to be established.
2. to automatically insert the date in any filenames of OSes that are backed up.

PINN does not use NTP, but gets the time from the fileserver, which is normally UTC. 
For point 1, only an approximate time is required. I forget the details, but I think it is of the order of days, rather than hours.
For point 2, the timestamp can be overridden in the filename anyway, so it's just for convenience.

So the main purpose for manually setting the time is if you are offline and want to backup an OS. I'm not sure providing a timezone and all of it's complexities is worthwhile for this function.

typxxi, 2020-04-13:
I had 2 hours of difference and that might be depending on summer time or not. But at the moment I saw that difference of 2 hours and was shocked or astonished believing in that time for a second or two. 

Maybe it is possible - if wifi is available - to use the right time cause first step is establishing wifi connection and since then everything is available to display the right time to the user  cause the system knows language, nation ... Would be great.

lurch, 2020-04-14:
> Maybe it is possible - if wifi is available - to use the right time

As @procount explained, it does get "the right time", it's just that it's displaying the [UTC time](https://en.wikipedia.org/wiki/Coordinated_Universal_Time) and not localising it to your local timezone.

>  cause the system knows language, nation ... Would be great.

Converting the UTC time to your local timezone is a [lot more complicated](https://en.wikipedia.org/wiki/Time_zone#List_of_UTC_offsets) than you might think :wink:

procount, 2020-04-14:
And on top of the timezones there are all the daylight savings times, which don't all change on the same date.
Sorry, but I'm just not going to go there.

lurch, 2020-04-14:
@procount I wonder if mentioning somewhere in the PINN UI that it's using UTC time, would prevent any confusion about "PINN being set to the wrong time" ? :man_shrugging:

procount, 2020-04-14:
Was just thinking I should add an indication somewhere......

procount, 2020-08-25:
UTC indication added

closed 2020-08-25 by procount