procount/pinn #365: system partition too small for lineageOS 16 - pi4

montgomeryb, 2020-04-12:
Hi,
I replaced an os with the latest lineage 16 for pi4 - that all seemed to go fine. 

I'm pretty sure I'm following the correct instructions from https://konstakang.com/devices/rpi4/LineageOS16.0/ - when I boot into TWRP to flash the gapps, it fails with an error 70.

Reviewing the log, it seems that the system partition is about 65K too small. I tried deleting the boot animation that should have given plenty of room, but it still failed. I'm thinking it's trying to update a disk image? I'm not sure, but is the size of the system partition set by PINN and something that can be increased?

procount, 2020-04-13:
I think all except the data partition are fixed sizes, but I can increase the size of the system partition to see if it makes a difference for you?

montgomeryb, 2020-04-13:
Thanks, I'd be happy to try it. I'm pretty sure I've seen steps somewhere for adding a custom / local image. I'm guessing that would be the approach here, download it, make it available (eg on usb) to pinn and then reinstall the partition.

procount, 2020-04-13:
I've increased the system partition size from 1008MB to 1010MB, so an increase of 2MB should be sufficient.
Please try downloading/installing it again and let me know how you get on.

montgomeryb, 2020-04-15:
Success - after a few false starts. I ended up downloading the image to a usb drive, and having seen the changes you made to partitions.json, I changed it to `"partition_size_nominal" : 1110,`

I realized that in fact with the initial failure, it was ~65MB too small - I missed that the report was in KB. It also seems that when reinstalling an image, pinn doesn't recreate the partitions? I tried with the modified image on the USB but it still seemed to have the same free size. 

After reinstalling on a fresh sd card, with the modified, downloaded image it worked fine. I included the resulting log file that shows about 18MB free. [open_gapps_log_good.txt](https://github.com/procount/pinn/files/4478644/open_gapps_log_good.txt)

procount, 2020-04-15:
Great.
Yes, re-installing (or replacing) does not adjust existing partition sizes. PINN is not gparted! So you have to install a fresh copy.
I have increased the size of the system partition in the Pi3 and Pi4 versions of LineageOS now.
Thanks for confirming.

procount, 2020-04-15:
I must try installing GAPPS myself. Are the instructions in the FAQ sufficient and easy to follow?

montgomeryb, 2020-04-15:
I hadn't initially thought too much about it - but I figured that wiping
out an installation would delete those partitions and leave a 'hole' on the
disk that a new installation would fill. I confirmed that it didn't when I
tried installing a different repo - and it wouldn't because of a different
number of partitions.

As for gapps, it was easier. First try I downloaded from the lineage
browser but then I copied the zip file to the usb drive for further
iterations. Essentially from what I remember, the steps were to: 1) Enable
developer options, 2) Enable root access, 3) enable local terminal
application 4) Return to home screen and then to apps to open terminal 5)
su and then run rpi4-recovery.sh and reboot 6) from TWRP install zip on usb
7) mount boot 8) goto advanced and open terminal 9) run recovery.sh boot
and reboot

The page said you had to register the device on google to enable gapps but
I didn't seem to have to do that (maybe they now accept RPI4 as a valid
device?), so far I have only installed facebook messenger (Haven't even
tried running it yet).

Thanks.

procount, 2020-04-24:
Closing as solved.

closed 2020-04-24 by procount

Firifire, 2021-10-07:
With newer Gapps, I believe the size required for system is more now so I am having this issue now