procount/pinn #366: PiNN does not allow to "pishrink" its images

typxxi, 2020-04-13:
After years of using PiNN since Zero 1.3 and the first PiNN release i ran into trouble caused by running out of 8 GB SDcard. I had to buy new ones and then my dd created images did not fit on the new SD cards.

So I searched for a solution cause my systems are a lot smaller than 8 GB so there must be space. 
The solution I found was pishrink.sh script ... that worked for a friend for years , but not for me.
It took us 4 continous day to isolate what is causing the problem cause first i thought it was on my systems, sd cards, corrupted installations (that still were running well). 

After tons of new installations and checks to make sure that it was not a mistake on my side I can sum it up what took me 4 days to find out:

PiNN Raspbian installations lead to: Error: Can't have overlapping partitions.
pishrink.sh: ERROR occured in line 309: parted failed with rc 0

From that point on I checked a new Noobs Installation and I got a similiar error and then went the last step creating a new raspbian system from scratch or better raspberry foundation which I had not done for 4-5 years . Again: this was another buster installation and it worked once, twice and a third time before I could believe it ... that I had lost 4 days.

I like to use piNN and its multiboot options and many of the details but I am sure that I might run again into issues like this that an IMG I made from a live system does no longer fit on a new SD Card cause it might be 512 K smaller and therefore I might need a shrinked IMG again. 

Therefore I did all these test with 5 or 6 sd cards on all my Raspberry from zero, 2, 3 to 4 I have .
I hope you can figure out a solution with the PiShrink guy ... cause this is a heavy point people get aware years later when 4 or 8 GB cards are no longer available and it is no solution to move to 16 GB cause this will lead to 16 GB image files as backup ... 

Here I had opened an issue on the PiShrink site with a longer explanation of my easter 2020 ... cause it might need a cooperation to solve this once and forever. It covers more details and the history where I came from and what steps I made to get to this point. I think it is caused by a NOOBs mechanism that is used PiNN too and might need changes in noobs too or Phishrink. 
Some more details here - Thanks for the support

https://github.com/Drewsif/PiShrink/issues/130

procount, 2020-04-14:
According to the answer you got from pishrink, I suspect pishrink is designed for Raspbian images only and assumes the SD card has only 2 partitions

A noobs or pinn SD card is partitioned differently so it is not surprising it won't work.

Until pishrink supports a noobs/pinn partitioning layout (see wiki) I suggest you use PINN's ability to backup each OS individually to a USB stick. This will already be compressed so there is no need to shrink it further as it will resize to any target SD card on restoration. Or use PINN's clone SD card facility to copy the whole card to another.

procount, 2020-04-14:
@lurch, yes "SD card Copier" works with noobs & pinn and will expand/shrink only the LAST partition to fit slightly different sized SD cards, but also it only copies the first 9 partitions. See my long outstanding issue https://github.com/raspberrypi-ui/piclone/issues/19.

PINN'S clone SD feature works the same yet copies all partitions but doesn't generate a new disk uuid.

procount, 2020-04-24:
Closing as this is a PiShrink issue.

closed 2020-04-24 by procount