procount/pinn #367: PINN won't boot to any OS

yesno123, 2020-04-16:
Hi, I've installed LibreElec,Lakka and Raspbian, but none of them boot, when I select any of them I eventually just end up getting the Check cable signal message from my monitor.. I can boot into the PINN installer, recovery no problem, and have tried reinstalling the OSs but it's the same result, any help would be appreciated

procount, 2020-04-16:
Are you using a Pi4?
More than likely it's the new vc4_kms drivers not working well with your HDMI monitor. What is the default resolution?

In PINN, go to the maintenance menu. Select each installed OS in turn and edit the config.txt file. Add or change the following settings:

hdmi_group=2
hdmi_mode=28
hdmi_ignore_edid=0xa5000080

Or you could try commenting out the fkms driver.

lurch, 2020-04-17:
...and if you have a spare SD card, you could also try doing a regular "image install" of Raspbian using [Raspberry Pi Imager](https://www.raspberrypi.org/downloads/) to see if that behaves any differently to the PINN-installed version?

yesno123, 2020-04-17:
> Are you using a Pi4?
> More than likely it's the new vc4_kms drivers not working well with your HDMI monitor. What is the default resolution?
>
> In PINN, go to the maintenance menu. Select each [...]

Got it in one! Thanks a million!

closed 2020-04-17 by yesno123