procount/pinn #368: Arch Linux Arm - Raspberry Pi 4 support

aTosser, 2020-04-23:
Can we add the arch-linux-arm for rpi4 builds?

https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-4

lurch, 2020-04-23:
Related to the discussions in #58 ?

procount, 2020-04-24:
Certainly it was the case back in 2018 when I last looked at their Arch version for the RPI3, but that was an aarch64 version. I've had a look at the current RPI3 and RPI4 versions and it seems both have gone back to 32 bit, so they don't use U-Boot anymore for RPi. I'll try and find some time to convert them.

procount, 2020-04-30:
Updated all Arch

closed 2020-04-30 by procount

33Fraise33, 2020-05-08:
Hello, 

has this been added now? as I don't see arch support for the RPI4 in the OS list: 
https://github.com/procount/pinn/wiki/OSes-supported-by-PINN

procount, 2020-05-08:
Yes. Sorry, I don't always remember to update that list. For a dynamic up to date list, go-to pinn.mjh.nz

33Fraise33, 2020-05-08:
> Yes. Sorry, I don't always remember to update that list. For a dynamic up to date list, go-to pinn.mjh.nz

Thanks for the reply!

procount, 2020-05-08:
I will do. But if you already have PINN & are connected to the internet, you will always see the latest available versions ready to download. They automatically appear.