procount/pinn #370: Operating Post Install

lsrs7, 2020-04-28:
Hi,

I'm using PINN to install Ubuntu-mate onto sd-card for Raspberry (headless) and want to post install the operating system (ubuntu mate) without configuring all day long the os (similar to preseed). Is there any elegant way of doing so?

Cheers,
Lutz

procount, 2020-04-28:
I'm not that familiar with preseeding, but I understand it is a means of providing a set of predetermined answers to the installation questions. In this sense it is more to do with how to configure the OS after PINN has installed it, and you are asking how best PINN can facilitate this?

I think this is more dependent on the OS itself, as most of the OSes don't have installation questions. Manjaro is one that springs to mind as having one, but I wouldn't know how to preseed that. And I think preseeding only really makes sense if you're going to have to install the OS multiple times.

So PINN can help in 2 ways:

1. The simplest way is perhaps to install the OS as normal and answer the installation questions manually. Once you've done this you can use PINN to backup the OS to USB. In future, if you need to install it again, just install from the backup and you have all the questions answered already.

2. The above is good if you want to answer the questions the same way each time., but if you want to have different preseeded answers, then you'll have to rely on the OSes preseeding method. You can use PINN's "flavours" to configure the OS in this case. See https://github.com/procount/pinn/blob/master/noobsconfig.md for details. This allows you to copy files onto the OS, such as a preseed file, or run on boot scripts, before the OS is first executed. You can provide a "flavour" for each different preseed file, but all flavours will share the same OS installation.

procount, 2020-04-30:
You would write a script, which noobsconfig copies to the appropriate partition of the OS after it has been installed. As a simple example for Raspbian, you could modify /etc/rc.local which is a script file that is executed when the OS boots (different OSes do it in different ways. you could use a systemd task for example). 
Then, when the OS boots for the first time, it runs your script and configures the OS however you need it done. Finally the script would delete itself as you don't want it to run on subsequent boots.

You can create an elaborate framework of scripts, so that some run only once on first boot, others run on every boot etc. For a comprehensive example, you can see https://pihw.wordpress.com/guides/pi-kitchen/

But typically, most of the images I provide through PINN are an image "snapshot" that is taken after the OS has been installed, whereas "pre-seed" files are usually applied during installation, for example, where you install an OS from a bootable USB drive on a PC and it is customised as part of the bespoke installation process. That's not normally how it is done on a Pi.

noobsconfig solves the problem of "how do I take a standard image and inject some standard files,scripts or packages onto a partition, particularly when I can't access the EXT4 formatted root partition from a Windows PC?"

lurch, 2020-04-30:
> You can create an elaborate framework of scripts, so that some run only once on first boot, others run on every boot etc.

There's also [PiBakery](https://www.pibakery.org/) which is a similar idea, but that's specific to Raspbian image installations and so not compatible with PINN.

Ultimately, I think any "preseeding" questions are really something you need to ask the developers of the particular OS about, rather than the developer of PINN?

procount, 2020-11-29:
No activity, so closing

closed 2020-11-29 by procount