procount/pinn #374: PI 4 with PINN V3.3.4 installing Raspbian Lite to a PNY SSD causing black screen

PeterHurn, 2020-05-08:
PI 4 with PINN V3.3.4 installing Raspbian Lite to a PNY SSD causing black screen

Odd one as no graphics involved, the boot starts are normal and proceeds almost to the usual login prompt then the screen goes blank, waited for 30 minutes but nothing appears, the full version of Raspbian works with the GUI.

Any Ideas?

procount, 2020-05-09:
I think this might be an incompatibility problem with your monitor and the new graphics drivers.
The graphics drivers switch several times during boot. Maybe Raspbian works because it boots into the desktop, whereas Lite is still in console mode.

One thing to try in PINN is to go to the Maintenance Menu and highlight your installed Raspbian Lite. Then select Edit config and change (or add) as required the following options:

```
hdmi_group=2
hdmi_mode=28
hdmi_ignore_edid=0xa5000080
```
This will force 1280x800 mode and ignore any information from the monitor.

Have you tried installing Raspbian Lite with RPI Imager? See if that gives you the same result, or if you end up at the login screen.

lurch, 2020-05-10:
Lots more info at https://www.raspberrypi.org/documentation/configuration/hdmi-config.md and https://www.raspberrypi.org/documentation/configuration/config-txt/video.md and https://www.raspberrypi.org/documentation/configuration/cmdline-txt.md

PeterHurn, 2020-05-11:
Thanks, apologies, I should have checked the config.txt, I have uncommented the HDMI safe options and everything is working fine.

closed 2020-05-11 by PeterHurn