procount/pinn #375: Bootselection by gpio

beatz0001, 2020-05-13:
Is it possible to switch between (2) installed os' by connecting a switch to gpio's of the pi?

Pos.1 > Boot OS1 (or Standard OS)
Pos.2 > Boot OS2

procount, 2020-05-14:
I've not tried this myself.
I suppose you could write a small shell script to do this. It must be called `pinn_init.sh` and stored on the PINN recovery partition (i.e. /dev/mmcblk0p1)
You could get this to read the status of a GPIO pin and alter the setting of `default_partition_to_boot` in `/settings/noobs.conf` accordingly. For typical 2 partition OSes, you would change the value between 6 and 8 (see `/settings/installed_os.json` for the actual values used for your OSes).
See https://www.ics.com/blog/gpio-programming-using-sysfs-interface for details on how to use GPIO using sysfs.
Note that PINN uses the busybox provided shell (/bin/sh - based on the simpler ASH, not BASH).

lurch, 2020-05-14:
Maybe https://github.com/raspberrypi/noobs/pull/389 helps? :man_shrugging:

procount, 2020-05-14:
Good find! (Memory of an elephant again)
Is there actually an implementation in that PR (somewhere in the 5334 files that have changed)? Looks just like a request. Even if there is, I don't remember it being implemented. If not, it would require a rebuild of NOOBS/PINN, whereas the solution above can be done by a user without modifying PINN.

lurch, 2020-05-14:
> Is there actually an implementation in that PR

I dunno - perhaps there _was_ an initial implementation that got lost in a "git rebase" that went wrong somewhere? :shrug: The initial request was in https://github.com/raspberrypi/noobs/issues/366

> whereas the solution above can be done by a user without modifying PINN.

:+1:

beatz0001, 2020-05-16:
Thanks for your input.

I've built it and used the script from here: https://github.com/procount/pinn/issues/187

Afterwards I've installed two instances of picoreplayer. Everythings looks good. The output of pinn_init.sh confirms the selection of the different os's (also shown in the bootselection-window of 
pinn).

BUT: It ends up with the same configuration of picoreplayer ?? In the output of the shell (bootprocess) is always the recovery of picoreplayer configuration from mmcblk0p9 mentioned.

How is that possible?? I've also reinstalled both os' but the result is the same.

There are two identical installations of picoreplayer (renamed with alias) made with project spaces.

OS1 partition 6/7
OS2 partition 8/9

Any idea?

procount, 2020-05-16:
Perhaps you could post your script here, along with with any config.txt and cmdline.txt files from picoreplayer to see if we can spot anything wrong?

beatz0001, 2020-05-16:
### **RECOVERY**

**cmdline.txt > empty**

**pinn_init.sh >**

`dir=$(pwd)

cd /sys/class/gpio

echo 18 > export
echo in > gpio18/direction
bootselection=$(cat gpio18/value)

mkdir /tmp/bootsel
mount -t auto /dev/mmcblk0p5 /tmp/bootsel

cd /tmp/bootsel

if [ "$bootselection" = "1" ]
then
   echo "Local LMS selected! Rock on!"
   rm noobs.conf.bak
   cp noobs.conf noobs.conf.bak
   rm noobs.conf
   cp noobs.conf.llms noobs.conf
   cat noobs.conf

else
   echo "Remote LMS selected! Don't stop the Rock!"
   rm noobs.conf.bak
   cp noobs.conf noobs.conf.bak
   rm noobs.conf
   cp noobs.conf.rlms noobs.conf
   cat noobs.conf

fi

cd $dir

sync

umount /tmp/bootsel
rmdir /tmp/bootsel

sleep 10`

**recovery.cmdline >**

`repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json ssh bootmenutimeout=2 quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline loglevel=2 sdhci.debug_quirks2=4
`

### **SETTINGS**

**installed_os.json >**

`[
 {
  "bootable" : true,
  "description" : "A Squeezebox Audio Device",
  "download_size" : 39818320,
  "folder" : "/settings/os/picoreplayer",
  "group" : "Media",
  "icon" : "/settings/os/picoreplayer/icon.png",
  "name" : "picoreplayer=picoreplayer-rlms",
  "partitions" : [
"/dev/mmcblk0p6",
"/dev/mmcblk0p7"
  ],
  "password" : "piCore",
[... 288 lines cut ...]
#pCP will retain these settings during insitu-update
#---Begin-Custom-(Do not alter Begin or End Tags)-----
#---End-Custom----------------------------------------

# NOOBS Auto-generated Settings:

`

**os_config.json >**

`{
 "bootable" : true,
 "description" : "A Squeezebox Audio Device",
 "flavour" : "picoreplayer",
 "group" : "Media",
 "imagefolder" : "/settings/os/picoreplayer",
 "keyboard" : "de",
 "language" : "de",
 "partitions" : [
"/dev/mmcblk0p8",
"/dev/mmcblk0p9"
 ],
 "release_date" : "2017-09-23",
 "videomode" : 0
}`

beatz0001, 2020-05-16:
[image]

beatz0001, 2020-05-16:
How can i mask # in comments? Every codeline with # at the beginning will be shown very large...

procount, 2020-05-16:
Add 3 back ticks around text. I've done it for you around the whole text but you might want to edit to just put them around individual files.

beatz0001, 2020-05-16:
Here are some pictures from the boot process...

beatz0001, 2020-05-16:
OS1

[image]

closed 2020-05-16 by beatz0001

beatz0001, 2020-05-16:
**### OS1**

[image]

[image]

[image]

beatz0001, 2020-05-16:
### **OS2**

[image]

[image]

[image]

beatz0001, 2020-05-16:
Everthings seems to be like it should... (to me at least)

But in the last picture(s) there's in both cases the identical line:

> Restoring backup files /mnt/mmblk0p9/tce/mydata.tgz /

How is this possible? And how can I change this behaviour?

beatz0001, 2020-05-16:
I edited the config files a llittle bit more. 
Now you can see it like I wanted it to be shown.

Thanks für the tip.

procount, 2020-05-16:
Must be a Picore thing.
I note there's a line that is scanning for partitions to build /etc/fstab, so check out those files. 
Maybe Picore is not expecting 2 instances on 2 separate partitions :man_shrugging: 
Maybe it just uses the latest backup it finds.

beatz0001, 2020-05-16:
What i don't understand: How is Picore1 able to know about any configuration on partition 9?

Do you there's a workaround for this?

procount, 2020-05-16:
Post the contents of /etc/fstab from os1 & os2.
(Nothing to stop it loading the partitions of another OS, especially if it is scanning all partitions)

lurch, 2020-05-17:
I agree with @procount - looks like everything is set up correctly from the "PINN side" but something in the "PiCore side" is picking up the wrong partition. Maybe you need to get in touch with the PiCore team to ask them to make it more compatible with PINN? :man_shrugging:

vitouXY, 2020-06-20:
> 
> > Restoring backup files /mnt/mmblk0p9/tce/mydata.tgz /
> 
It seems that piCore when booting and looking for the correct partition uses the last result ...

Look at the scripts: `tc-config`, `tce-setup`, `tc-functions` and `tce-setdrive` in this repository: [[tinycorelinux/Core-scripts]](https://github.com/tinycorelinux/Core-scripts).
Or search for: `/etc/sysconfig/tcedir`.

Suerte :v:

jharris1993, 2021-03-29:
There is an active solution to this located on PINN's sticky, authored by me - (99% of the credit should go to procount for his patient help!)
[https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&p=1843314#p1843314](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&p=1843314#p1843314)

With the attached script and associated config.txt file, (which you put in PINN's root directory, "RECOVERY"), you can use GPIO pins to select operating systems to boot.

Note that (at the present time) ***this only works if you are using a SD card***.  There are still issues with getting scripts to run when on USB media.

This script assumes you are using one of the last six physical GPIO pins, (35 to 40), to select one of five O/S's.  You can reduce that number or use any other GPIO pin you want so long as it's not being used for anything else.  (See the very important ***WARNING*** within that posting.)

I hope this helps.