procount/pinn #376: Joystick can't enter recovery mode

Astra3, 2020-05-16:
I'm using Xbox 360 wired gamepad and whatever I press, PINN menu isn't shown and the default sticky OS is booted. Also woth noting that only left analog and mouse were working until I supplied joy_keys.json.

procount, 2020-05-17:
could you capture some logs?
Please boot PINN. Press some joystick buttongs, move the ananlog joysticks that work etc.
Then see the [wiki-troubleshooting page](https://github.com/procount/pinn/wiki/Troubleshooting) on how to capture them.

Astra3, 2020-05-20:
I booted, pressed some buttons on joystick and entered recovery mode by using keyboard and then pressed some more joystick buttons. As I said, I have to use custom joy_keys.json to make the buttons even do something, so I'm including it as well. Without it, only left analog works for mouse.
[debug.txt](https://github.com/procount/pinn/files/4656690/debug.txt)
[dmesg.txt](https://github.com/procount/pinn/files/4656691/dmesg.txt)
joy_key.json:
```json
{
    "calibration": [
        {
            "threshold": 1,
            "step": 1
        },
        {
            "threshold": 10000,
            "step": 10
        },
        {
            "threshold": 20000,
            "step": 100
        }
    ],
    "any": {
        "any": {
            "mouse_left":   "LAnalogLeft",
            "mouse_right":  "LAnalogRight",
            "mouse_up":     "LAnalogUp",
            "mouse_down":   "LAnalogDown",

            "joy_left":     "RAnalogLeft",
            "joy_right":    "RAnalogRight",
            "joy_up":       "RAnalogUp",
            "joy_down":     "RAnalogDown",

            "mouse_lclick": "X",
            "Key_Escape":   "B",

            "Key_Enter":    "A",
            "Key_Left":     "D-padLeft",
            "Key_Right":    "D-padRight",
            "Key_Up":       "D-padUp",
            "Key_Down":     "D-padDown"

        }
    }

}
```

procount, 2020-05-20:
Thanks for the logs.
It looks like you are running an old version of PINN v3.3f from about July 2019.
Have you tried updating to the latest version of v3.3.4?

Astra3, 2020-05-20:
No, menu says it's running v3.3.4 and when I tried to manually update, no updates were availible. I've downloaded pinn-lite package which I moved to the SD card on 13th May 2020, so I didn't think I was running on a version from July 2019 but I double-checked just to be sure.

procount, 2020-05-20:
Sorry my fault, I was viewing it on my phone and it asked if I wanted to download debug.txt again, so I thought it was already downloaded. Must have been an old one. I'll take another look.

procount, 2020-05-20:
This is the current default mapping file:
```
{
    "calibration": [
        {
            "threshold": 3000,
            "step": 1
        },
        {
            "threshold": 10000,
            "step": 10
        },
        {
            "threshold": 20000,
            "step": 100
        }
    ],

    "VKeyboard": {
        "any": {
            "Key_Left":     "Y",
            "Key_Right":    "RB",
            "Key_Up":       "LB",
            "Key_Down":     "B",
            "Key_Space":    "Back"
        }
    },

    "any": {
        "any": {
            "mouse_left":   "LAnalogLeft",
            "mouse_right":  "LAnalogRight",
            "mouse_up":     "LAnalogUp",
            "mouse_down":   "LAnalogDown",

            "joy_left":     "RAnalogLeft",
            "joy_right":    "RAnalogRight",
            "joy_up":       "RAnalogUp",
            "joy_down":     "RAnalogDown",

            "Key_Left":     "Y",
            "Key_Right":    "RB",
            "Key_Up":       "LB",
            "Key_Down":     "B",
            "Key_space":    "Start",

            "mouse_lclick": "A",
            "Key_Escape":   "X",
            "Key_Enter":    "Start"

        }
    }

}```
But of course, your buttons may not map exactly the same as these.

Astra3, 2020-05-20:
Yes, my mapping is just eddited version of the default mapping (the default mapping file worked fine too once I included it). Problem is, that none of the buttons can be used to enter recovery. They can be used to control the menu, but they can't be used to enter recovery.
That's a huge problem for me due to the way, my Pi is set up, because I have to use sticky OS and if I want to use another OS, I have to enter recovery.

rickwookie, 2020-12-27:
Has there been any update on this. From what I can make out, mappings for either mouse_lclick or Key_Enter always get ignored, so while you can navigate the boot selector, you can never actually select and boot from it.

procount, 2021-01-12:
This should be solved in the next version of PINN. 
I have completely re-written the Joystick code which will enable more flexibility in button mapping and it can be used in place of the shift key to enter the menu.
@Astra3  - As you have an Xbox controller (which I don't have), would you be interested in testing the next release so I can produce a mapping file ahead of release? That applies to anyone else too who has a different type of controller.

rickwookie, 2021-01-12:
I have a controller from the original Xbox and I would like to test the new code.

procount, 2021-01-13:
Excellent! I'll let you know when it is available. Thanks.

Astra3, 2021-01-13:
Sure! If I won't be stupid enough to do it it's no problem.

procount, 2021-01-18:
I have uploaded https://sourceforge.net/projects/pinn/files/testing/pinn-354a.zip/download
Please download and install to a blank FAT32 formatted SD card as normal.
Boot PINN with your joystick disconnected and wait for it to finish initialising.

I need you to record some information from the debug log. You can access this in 3 ways:
1) Go to the Maintenance Menu, check the box next to PINN, then click on the Fix button, select Show Debug Log, and click ok. You will have to click refresh often to get the latest log.
2) Press ctrl-alt-f2 to get to the command shell. Use root/raspberry to login. Then type `tail -f /tmp/debug`. The debug logs will be displayed with no need to refresh. (To exit, press ctrl-C, then ctrl-alt-f1 to return to PINN)
3) Login with ssh Use root/raspberry to login. Then type `tail -f /tmp/debug`. The debug logs will be displayed with no need to refresh. 
Method 1 may be easiest (but be aware that a button mapped to the Escape key may close this window!) Method 3 will allow you to cut and paste from your ssh client into a post. Your choice!

Once you can see the logs, plug your joystick into the USB port.
It should display the name of your joystick such as "Sony_PLAYSTATION_R_3_Controller_7A_13B.json" in the logs. This is the name of the joystick configuration file it will try to load. If it doesn't exist, it will try to load "joy_keys.json". If this doesn't exist, it will use an internal set of mappings (the same as joy_keys.json).
You now need to press each key on your joystick and record the joystick messages. 4 pieces of information are required:
1) The button or joystick you pressed/moved
2) Whether it was reported as a btn or as an axis
3) The number of the button/axis
4) The value of the button/axis when pressed. (The sign is also important)

Please post the name of the joystick as identified, plus the information requested above for each of the buttons/joystick axes.
(A picture of your joystick and the names of each button might also be helpful, as I believe Xbox, Nintendo and Sony might use the same key names, but in different positions).

If you are feeling brave, you could create a new joystick configuration file by copying one of the provided joystick mapping files to the required configuration file name, and entering your mappings into the Inputs section, using the original file as a guide.
Buttons must have a _btn suffix, and joysticks must have an _axis suffix. The next 2 numbers indicate the button/axis number, and the sign of the value (use 1 for positive values and -1 for negative values. Buttons are always 1). Feel free to post this file in addition.

Good luck and thanks for testing.

procount, 2021-01-18:
Sorry, I think I broke SSH in this beta version!
To fix it, go into the command shell and do:
```
mkdir -p /settings/dropbear
sync
rebootp
```
Then you should be able to connect over ssh.

rickwookie, 2021-01-18:
> Sorry, I think I broke SSH in this beta version!
> To fix it, go into the command shell and do:
> 
> ```
> mkdir -p /settings/dropbear
> sync
> rebootp
> ```
> 
> Then you should be able to connect over ssh.

Still getting SSH connection refused even after doing this.

rickwookie, 2021-01-18:
So as we don't get bogged down in the SSH not working for now, and to move this along. I tried by two controllers, and they both seem to be identified with bespoke mapping files (sorry for the poor quality images, but without SSH I just photographed the TV):
[image]
[image]
The issue with the old Xbox controller is that there's no real dead-zone for the analogue stick, so it's practically impossible to get the cursor to stay in one place and not have it shoot off across the screen. The other issue I noticed that applies to both controllers is that the key repeat delay is too short (not the same as for the keyboard for example) so it's very difficult to press the button and only have it fire once (the result being that you nearly always end up selecting and then imediately deselecting something again).

Apart from that, this is a big step forward, as I can now actually click on things in the menu!

I you would like me to send you the debug tail for the controller, let me know, but I'm not sure it's worth doing since both controllers seem to identify with their correct mapping files anyway (and it's very difficult to do the Xbox one anyway since the output is flooded with the analogue axis lines that never "zero").

Astra3, 2021-01-18:
> > Sorry, I think I broke SSH in this beta version!
> > To fix it, go into the command shell and do:
> > ```
> > mkdir -p /settings/dropbear
> > sync
> > rebootp
> > ```
> > 
> > 
> > Then you should be able to connect over ssh.
> 
> Still getting SSH connection refused even after doing this.

Did you add `ssh` in `recovery.cmdline`? it works fine for me.

procount, 2021-01-18:
Ahh! I completely forgot about deadzones, because all 3 of my joysticks read 0 in the middle. I will need to add that back again.

There are no default mapping file for the XBOX controllers (yet!). The json filename displayed is the one it is EXPECTING to find.
At the moment, your system will just be using joy_keys.json, but there might be better mappings that could be created if we knew all the buttons and axis and how they are mapped. So this information would be very useful, as described in my previous post. 

Some joysticks also have different modes, so the D_pad for example could present itself as buttons, or as analog axes. The joystick then presents itself as having a different number of buttons and axes, which is why I include that in the mapping filename.

Astra3, 2021-01-18:
Alright, I tested everything I could. The number of buttons and axis are basically the same as `xpad` reports them, so I figured out all numbers just by using `jstest-gtk`, however, log is provided as terminal.log file.
[terminal.log](https://github.com/procount/pinn/files/5830706/terminal.log)

The only difference between mapped joysticks is the fact that Xbox 360 uses R2 and L2 as triggers, not buttons and I tried to make it work but I couldn't. This is the mapping file I created and it works well excluding RT and LT triggers and analogs since I can't really test those due to the missing deadzone, mouse was moving all over the place until I remapped it to Dpad. The deadzone @rickwookie just mentioned, +/- 5000 will be good enough.
[Microsoft_X-Box_360_pad_8A_11B.txt](https://github.com/procount/pinn/files/5830722/Microsoft_X-Box_360_pad_8A_11B.txt) (had to change the file extension to .txt for github to let me upload it).
The triggers read -32767 when **not** pressed and 32767 when completely pressed, each of them has different axis (I mapped both in the .json file). Maybe the fact I changed the names of the axis disrupted the file reading process, I can't be sure about that.

Also the last thing I noticed was that A button, when mapped to left click, did not allow for dragging and holding windows with mouse, but I suspect this has not been implemented yet.

procount, 2021-01-18:
@Astra3  - thanks for that. Those L2.R2 Triggers are weird. I'm not sure if I have catered for that scenario yet.

procount, 2021-01-18:
@rickwookie - It might be easier to take your SD card out of the PI and put it in a PC.
In the root folder you will find several JSON files. Take one (e.g. "Sony_PLAYSTATION_R_3_Controller_7A_13B" and copy it to the same name as PINN provided in the logs. Then edit the inputs section to match the button and axis numbers you found from the logs.

If you really want to do it on the RPi itself...
```
#ctrl-alt-f2 to get into the shell
#root/raspberry to login.

mount -o remount,rw /mnt
cd /mnt
cp Sony_PLAYSTATION_R_3_Controller_7A_13B.json Microsoft_X-Box_pad_v1__US__8A_10B.json
nano Microsoft_X-Box_pad_v1__US__8A_10B.json
#edit the inputs section
#ctrl-X, Yes, enter to quit
sync
```
If you unplug and replug your joystick, it should use the new mappings.

Astra3, 2021-01-18:
> @Astra3 - thanks for that. Those L2.R2 Triggers are weird. I'm not sure if I have catered for that scenario yet.

Most modern joysticks do use triggers instead of buttons in the back. It would be nice if they could be read as pressed at values from -10000 to 32767 or at least from 0 to 32767. It isn't necessary for this use case to detect, if the triggers are not pressed, it is only necessary to detect if they are pressed.

procount, 2021-01-18:
Don't forget, I need to detect when they are _not_ pressed in order to determine when they are _next_ pressed, Otherwise they will be pressed forever!

procount, 2021-01-18:
So the difference between a trigger and a normal axis, is that an axis is centred in the middle at 0. But a trigger by default is at the negative end (-32768) and moves through 0 towards the other end @32767 you more you squeeze it?

procount, 2021-01-18:
I have uploaded https://sourceforge.net/projects/pinn/files/testing/pinn-354b.zip/download
You can now add an optional 3rd number in the inputs section for the deadzone.
e.g.
```
        "LAnalogLeft_axis":     [ 0, -1, 5000],
```
An absent value, or 0 means no deadzone.
A positive value gives a deadzone of +/- deadzone around 0.
A negative value gives a deadzone from -32768 to the deadzone value. Values below are zero, Values above are 1.
(repeat time is not altered yet).
Let me know how you get on.

Astra3, 2021-01-18:
> So the difference between a trigger and a normal axis, is that an axis is centred in the middle at 0. But a trigger by default is at the negative end (-32768) and moves through 0 towards the other end @ 32767 the more you squeeze it?

Yes exactly. 0 is when the trigger is held in the middle, -32767 when not pressed and 32767 when pressed all the way in. Also I've seen some cheap joysticks not max out at 32767 so try to make it register as pressed at lower values. Some people also don't press the triggers all the way, so if it would register as pressed at values from -5000 to 32767 (or making this customisable in mapping file) it would be incredible.

As for the deadzone build I'm going to test it in 9 hours, can't really do it right now.

Astra3, 2021-01-19:
Ok, so I tested the new build. My Raspberry Pi 3B did not boot it and was stuck on the "rainbow graphic test" thing and after a few seconds I noticed that the build misses some files, so I just filled all missing files from the previous build and it works fine.

After I put all dead zones in the .json file and somehow *fixed* the build, mouse through left analog worked nicely. I have not tested right analog but since it was included in debug log file as well, I assume it works fine as well. This is the debug log, it is still flooded [terminal.log](https://github.com/procount/pinn/files/5834572/terminal.log)

I also did try to install some random small OS and tested if joystick can be used to enter recovery mode - yes it can. Basically, everything except triggers works fine.

procount, 2021-01-19:
Oh. I thought I'd zipped exactly the same but updated files. I will need take a look at that.
Did you try a negative deadzone value for a trigger, like -10000?  A negative deadzone should turn an axis into a trigger. How does it behave?

procount, 2021-01-19:
I'm uploading pinn-354c.zip which I hope includes everything, this time.

Astra3, 2021-01-19:
I tried the triggers and I found it to be a bit funny. This is a log file of me pressing triggers [terminal.log](https://github.com/procount/pinn/files/5834894/terminal.log). From what I understand, for some reason, they are being recognised as A button. I tried putting -1 and 1 for the axis, neither of those changed the behaviour. However I assume +1 should be the right option because the axis maxes out at positive number. This is the mapping file I ended up with [Microsoft_X-Box_360_pad_8A_11B.json.txt](https://github.com/procount/pinn/files/5834913/Microsoft_X-Box_360_pad_8A_11B.json.txt). As you can see, triggers are not mapped to anything and they do still pres A button. 

This is a video showcasing what is happening:
https://user-images.githubusercontent.com/18398260/105015219-9a1e8e80-5a41-11eb-94f9-4760dacd984a.mp4

procount, 2021-01-19:
Thanks for the logs and video. It's difficult to figure this out without a suitable joystick, but these should help. I'll see if I can see what's going on.

rickwookie, 2021-01-19:
> I'm uploading pinn-354c.zip which I hope includes everything, this time.

It's password protected?

procount, 2021-01-19:
No. Check you downloaded the correct file, because most of the rest are.

rickwookie, 2021-01-19:
Pretty sure it's the correct file. Was just these files with the password, so took those from pinn-354a.zip (they're not in the 354b.zip archive):
[image]

procount, 2021-01-19:
you don't need those. you can safely delete them.

rickwookie, 2021-01-19:
Ok, yes, so it didn't boot. Started again with all the files from a, then copied (the non password protected) files from c over the top. Now it boots, so I'll test now.

rickwookie, 2021-01-19:
> Did you add `ssh` in `recovery.cmdline`? it works fine for me.

I hadn't, but I have now. Thanks.

procount, 2021-01-19:
This is the problem with me cobbling together beta versions, I sometimes forget to include all the right files, because it doesn't follow a proper release procedure 😄

Astra3, 2021-01-19:
> This is the problem with me cobbling together beta versions, I sometimes forget to include all the right files, because it doesn't follow a proper release procedure

I, as a rookie in developing code, know this as well. This work you do is incredible and the joystick is functional even though triggers are not working. I'm really excited once you will figure this out and even more once this makes it into master branch.

Anyway for the triggers, any modern joystick, for example Dualshock 4 or anything newer than Xbox 360 uses triggers as axis and not as buttons. Once the axis passes the threshold set in config file, the input should be registered as pressed and once it falls back, it should be detected as not pressed. At least from my point of understanding but I know how writing code works.

So, if you'll come up with something I'm eager to test it out.

procount, 2021-01-19:
Yes, I think they need to be treated differently than an axis. I tried to just bolt it on but there are too many dependencies.

rickwookie, 2021-01-19:
OK, so here's the first gen Xbox controller output:

IDENTIFIER: Microsoft_X-Box_pad_v1__US__8A_10B

GREEN A:
Press - 	"Joystick btn  #0 Value: 1"
Release - 	"Joystick btn  #0 Value: 0"

RED B:
Press - 	"Joystick btn  #1 Value: 1"
Release -	"Joystick btn  #1 Value: 0"

BLUE X:
Press - 	"Joystick btn  #3 Value: 1"
Release -	"Joystick btn  #3 Value: 0"

YELLOW Y:
Press - 	"Joystick btn  #4 Value: 1"
Release -	"Joystick btn  #4 Value: 0"

WHITE:
Press - 	"Joystick btn  #5 Value: 1"
Release -	"Joystick btn  #5 Value: 0"

BLACK:
Press - 	"Joystick btn  #2 Value: 1"
Release -	"Joystick btn  #2 Value: 0"

BACK:
Press - 	"Joystick btn  #6 Value: 1"
Release -	"Joystick btn  #6 Value: 0"

START:
Press - 	"Joystick btn  #7 Value: 1"
Release -	"Joystick btn  #7 Value: 0"

D-PAD UP:
Press -		"Joystick axis #7 Value: -32767"
Release -	"Joystick axis #7 Value: 0"
		"Joystick axis #7 Value: 1"
		"Joystick axis #7 Value: -1"

D-PAD DOWN:
Press -		"Joystick axis #7 Value: 32767"
Release -	"Joystick axis #7 Value: 0"
		"Joystick axis #7 Value: 1"
		"Joystick axis #7 Value: -1"

D-PAD LEFT:
Press -		"Joystick axis #6 Value: -32767"
Release -	"Joystick axis #6 Value: 0"
		"Joystick axis #6 Value: 1"
		"Joystick axis #6 Value: -1"

D-PAD RIGHT:
Press -		"Joystick axis #6 Value: 32767"
Release -	"Joystick axis #6 Value: 0"
		"Joystick axis #6 Value: 1"
		"Joystick axis #6 Value: -1"

LEFT STICK:
Left -		"Joystick axis #0 Value: -32767"
Right -		"Joystick axis #0 Value: 32767"
Up -		"Joystick axis #1 Value: -32767"
Down -		"Joystick axis #1 Value: 32767"
Press -		"Joystick btn  #8 Value: 1"
Release -	"Joystick btn  #8 Value: 0"

RIGHT STICK:
Left -		"Joystick axis #3 Value: -32767"
Right -		"Joystick axis #3 Value: 32767"
Up -		"Joystick axis #4 Value: -32767"
Down -		"Joystick axis #4 Value: 32767"
Press -		"Joystick btn  #9 Value: 1"
Release -	"Joystick btn  #9 Value: 0"

LEFT TRIGGER:
Fully Pressed -	"Joystick axis #2 Value: 32767"
Fully Released -"Joystick axis #2 Value: -32767"

RIGHT TRIGGER:
Fully Pressed -	"Joystick axis #5 Value: 32767"
Fully Released -"Joystick axis #5 Value: -32767"

procount, 2021-01-19:
Perfect! Thank you.

rickwookie, 2021-01-19:
Why are the triggers firing the "A" button when that's not a mapping?

The debug shows:
"Joystick axis #5 Value: 27091"
found joy  0  ( A_btn ) in  "any"  :  "any"  as  4101  ( mouse_lclick )
Inject Mouse Code:  4101  Value:  27091
and
"Joystick axis #2 Value: 32767"
found joy  0  ( A_btn ) in  "any"  :  "any"  as  4101  ( mouse_lclick )
Inject Mouse Code:  4101  Value:  32767

Yet in my custom mapping file (atached) I don't have axis 2 or 5 mapped. Do I need to somehow map those unused axes to null?

[Microsoft_X-Box_pad_v1__US__8A_10B.zip](https://github.com/procount/pinn/files/5837122/Microsoft_X-Box_pad_v1__US__8A_10B.zip)

One other question - how does one invoke the virtual keyboard using only the game controller input?

procount, 2021-01-19:
1. Because my first implementation of triggers was too simplistic! 😉 Don't bother with triggers for now, the implementation will change.
2. Use the joystick axis you mapped to the mouse and the buttons you mapped to mouse_lclick, Or you can use the joystick buttons mapped to the cursor keys along with space and enter.

rickwookie, 2021-01-20:
I get that triggers aren't working, but why are axes 2 and 5 being mapped when they're not mapped in my json file? Is there a fallback for any unmapped/un-overridden axes and buttons?

If I add:`,
        "Null1Left_axis":       [ 2, -1],
        "Null1Right_axis":      [ 2, 1],
        "Null2Left_axis":       [ 5, -1],
        "Null2Right_axis":      [ 5, 1]`

to the end of the inputs list, will that cure the unwanted mouse_lclick catches?

procount, 2021-01-20:
Maybe if you add another dummy input at the top of inputs section, before the A_btn, with an invalid id , that might fix it for now.

rickwookie, 2021-01-20:
> Maybe if you add another dummy input at the top of inputs section, before the A_btn, with an invalid id , that might fix it for now.

That doesn't seem to fix it either (see attached for what I tried, maybe I did it wrong)
[Microsoft_X-Box_pad_v1__US__8A_10B.zip](https://github.com/procount/pinn/files/5842176/Microsoft_X-Box_pad_v1__US__8A_10B.zip)

It's quite annoying having a gazillion left-clicks come through whenever you even lightly brush one of the huge triggers under the controller.

Re my question about the virtual keyboard. It would be really good if it could be toggled by a mapped controller input, rather than the current method of only having it available in specific windows via a checkbox. That way, using a game controller as the only input could be truly achievable since the virtual-keyboard could then be invoked anywhere (even in the web browser for example). Maybe even allow the joy_left/right/up/down mappings (that don't appear to do anything at the moment) to be used to move the virtual keyboard around the screen to get it out of the way when it's obstructing something.

procount, 2021-01-20:
I understand. It needs a redesign for the triggers as I've not had to deal with them before. That may take a while. I have one idea that might fix the gazilion left clicks for the time being. I'll see what I can do.

Regarding the virtual keyboard, it is part of the PINN recovery program, not a separate program, so it is plumbed into sending keypresses to itself, and won't work well with external programs like the web browser, unfortunately.
I haven't figured out a decent way of moving the keyboard yet. It would need to work with a mouse and CEC controller as well. But there is a slider to the right of the keyboard so you can make it more transparent.

procount, 2021-01-20:
I was thinking you could map a button to Alt-V to bring up the keyboard, but I haven't implemented any modifiers, such as shift, ctrl, alt etc. Another thing to think about!

rickwookie, 2021-01-20:
Modifiers could be great for other purposes, but why not just have a custom Key Code such as "v_keyboard" to toggle it, since assigning something like Alt-V for this purpose is counterintuitive (if you can ever physically press the combination Alt-V, then you probably have a keyboard connected, right?!)

procount, 2021-01-20:
That's just the standard way the gui maps accelerator keys for a keyboard. The gamepad just needs to emulate the keyboard.

rickwookie, 2021-01-21:
> This should be solved in the next version of PINN.
> I have completely re-written the Joystick code which will enable more flexibility in button mapping and it can be used in place of the shift key to [...]

Rolling back a little. Can we just establish if the op is really requesting access to the SETUP menu, rather than the boot selection menu, with a game controller. The reason I ask is that from re-reading, it appears as though the issue was being able to chose an os on boot (the entire point of PINN right?) using just a game controller. What we have now with this new implementation is the setup (not boot selection) menu will get invoked if ANY game controller input is detected during the "press shift" screen. If access to the setup IS required using just a controller (I'm more inclined to hook up a keyboard for maintenance tasks my self) then does it not make more sense to have it only invoked using a specific mapped button?

The next issue... what happens when more than one controller is connected. I've configured the json for my Xbox controller, but if I have my Thrustmaster Dual-Shock plugged in too, it will always respond only to that controller (which I haven't set up yet) and there's no input from the Xbox controller working. Is it not possible to respond to more than one controller in the same way as more than one mouse or keyboard works?

procount, 2021-01-21:
The original issue was that pressing a gamepad button did not work like the shift key to stop the booting of the default OS (however your PINN is configured to do that). What PINN then displays is a secondary matter, and there is another issue about the default being the bootselection screen instead of the main menu,
My aim here is to get as many variations of gamepads mappable so that they can work with PINN, including those pesky triggers. The implementation is to emulate the mouse and/or keyboard. Currently there is no way to go straight to the bootselection directly with the keyboard or mouse.

PINN only looks for /dev/js0, which is mapped to the first detected controller. If you have 2 connected, it could be a bit random as to which one is used (in the same way as /dev/sda is mapped for USB devices). I shall have to see if it is possible to implement 2 controllers in parallel.

Astra3, 2021-01-21:
I want my Pi to have the ability to run headless, so I chose Ubuntu MATE as default sticky OS, which then includes SSH and VNC server. But when connected to some screen, I want to choose a different OS, LibreELEC or RetroPie. Although most TVs do support CEC and I use that, some don't (and every monitor doesn't as well) and RetroPie and LibreELEC can be controlled completely through gamepad, but PINN can't (now it can). This required me to take my mouse and use that to choose OS, then disconnect it and don't use it again. This change would spare some space in my backpack when taking Pi with me. Also it would allow me to do emergency maintenance if needed. The fact that PINN now supports 4 input devices (mouse, keyboard, CEC and joystick) and only one of them is required to do anything in the UI is neat, including stopping boot of the default OS.

On mouse it's just click to access menu, click on Exit, then choose OS and then click on boot. With keyboard hold Shift to access menu, press Esc, arrows to select, enter to boot. On CEC (with default config), back arrow is Esc, channel up/down are arrows and Play button is Enter and any button to access menu.

rickwookie, 2021-01-21:
Ah ok, I must have mine set slightly differently, since mine always shows the boot menu, with a short countdown timer that will then boot my sticky os if I don't press anything (this is why not having the deadzone was a nightmare). I realise this is not the most efficient way to boot if you're nearly always booting into the same os and only occasionally want to boot something else, as there's the "press shift" delay on top of the boot selector timer. Now I know that I can get to the boot selector FROM the recovery menu, I'll probably change that behaviour (do I just set the timeout to zero?) for my main TV since that's nearly always LibreElec, and only occasionally 

In my use case it could be 50/50 as to whether I'm booting into LibreElec or RetroPie,simce as the Pi4 is connected to my car's HDMI input (which has no CEC). Up until now I had to have a keyboard in the car as well as the game controller, just to be able to select the os. This change is therefore a massive improvement.

I appreciate it's a completely different issue that the menu order is recovery/setup and THEN boot selection, when having it the other way around, where recovery/setup was an option from the boot selection menu would make more sense.

> PINN only looks for /dev/js0, which is mapped to the first detected controller. If you have 2 connected, it could be a bit random as to which one is used (in the same way as /dev/sda is mapped for USB devices). I shall have to see if it is possible to implement 2 controllers in parallel.
Brilliant, thanks. In the mean time I'll just set up the mapping for the few different controllers I have. The controller that get's chosen seems to be 100% repeatable in my testing so far anyway (obviously can't know if it is alphabetical, or based on hardware vid/pid, etc.)

procount, 2021-01-21:
I hope it is sufficient for me to just upload recovery.rfs to the testing folder, which you can download and copy to your PINN partition. This is v3.5.4d
I've tested triggers by putting an elastic band around my joystick to force it to -32768. Seems to work for me, but please test on a real thing.
The format for a trigger is the same as before: an _axis with value 1, and a negative deadzone.
It's also now possible to map 2 controller buttons to the same key by making them a list.
e.g. "Key_Up": ["DUp_axis","RAnalogUp_axis"],
(It won't work if you map them as 2 separate entries).

Astra3, 2021-01-21:
I have detected three issues. First one is when I mapped mouse to both analog inputs. When the input value is or nearing 32767 or -32768 mouse doesn't work, but when it's something in between it works fine. You may be able to reproduce this behaviour, if not, I can try to make another video. When mapped only to one stick, it works fine, be it left or right.

As for the triggers...the issue with them spamming left mouse click persists. It only seems to do so when moving the trigger behind the specified threshold, it does not happen with value lower than the threshold, i. e. if I'm keeping the value between -32768 and -30000 nothing happens, but when moving with it over -30000 it starts this behaviour and spams the button until the trigger goes again below -30000 or stays at a solid position.

And during that, I found another problem. Joystick mappings can't be used to type to text fields. They can be used in the UI, for example I can map C button and use it to clear selection, but I can't type it to any text field.

This is the current mapping file I used: 
[Microsoft_X-Box_360_pad_8A_11B.json.txt](https://github.com/procount/pinn/files/5851261/Microsoft_X-Box_360_pad_8A_11B.json.txt)

procount, 2021-01-21:
1. Is this when you move both joysticks at the same time? Or are you only moving them one at a time. 
2. Need to check. Maybe mouse_clicks need no auto-repeat
3. I hadn't expected people to want to do that. You need to use the virtual keyboard for that

Astra3, 2021-01-21:
> 1. Is this when you move both joysticks at the same time? Or are you only moving them one at a time.
> 2. Need to check. Maybe mouse_clicks need no auto-repeat
> 3. I hadn't expected people to want to do that. You need to use the virtual keyboard for that

1. No, I was moving only one.
3. Yes I thought it was not implemented that way. To be honest I was not really sure it was not intended but I decided to mention it.

procount, 2021-01-22:
New recovery.rfs file available for you to try....

[7 more replies]

closed 2021-02-17 by procount