procount/pinn #380: Install button shall be disabled after 1st installation

schauveau, 2020-05-30:
The current behavior of the **Install** button is quite dangerous. It is very easy to erase all installed OS by mistake. The problem is of course that **Install** will not only install the selected OSes but it will also remove the previous installation. There is a warning about "removing all data on the SSD"  but it is likely to be ignored since it is basically identical to the warning given when replacing an existing OS. 

For example, it is easy to imagine that an unsuspecting user will enter RECOVERY to try the latest OS he just watched on Youtube. Fortunately, when he configured PINN 6 months ago, he was wise enough to install one or two ProjectSpace for future experiments.  So he disables a ProjectSpace to free some disk space, enables the new OS and of course press Install! Hoops! Facepalm! 

Another situation: You just spent days building the perfect Retropie arcade cabinet with hundreds of roms perfectly configured on a 1TB SSD. This is a very robust cabinet and you let you kid play with it.  After a few minutes, he manages to crash a game and after seeing the boot message he presses SHIFT to open the recovering screen. How much time will it take before he notices the Games panel, click on the gamepad icon of recallbosOS-rpi2 and then press the Install button?

Still not convinced? How many keypresses are needed to destroy all partitions from boot? The answer is 5:  SHIFT + C(lear) + ENTER (to select at least one package) + I(nstall)  + Y(es). That is not a lot and those are not even difficult keys. A baby playing with a keyboard could easily produce them. 

I understand that those scenarios are just that ... scenarios. They are unlikely to happen in real life but I really feel that the current behavior is too dangerous to be acceptable. You should reconsider the behavior of the **Install** button. 

So I propose the following UI improvements:
- A new button **Destroy All" should be introduced in the main menu (with a scary icon) 
- When no OS is installed, **Install** is enabled and  **Destroy All** is disabled. 
- When OS are installed, **Install** is disabled and  **Destroy All** is enabled.
- Clicking on "Destroy All" should open a very scary window explaining that EVERYTHING WILL BE DESTROYED. Also, the action should be confirmed for instance by entering a specific string such as **YES** in a text box (similar to the Write action in cfdisk or to the removal of a repository in github).   
- Finally, the RECOVERY menu shall be protected by a password. Even a non-encrypted password in recovery.cmdline would be better than nothing.   

Also, for added security, I would appreciate to have the ability to forbid Reinstall and Replace on some OSes because there are a few partitions that I really really do not want to remove by mistake.

schauveau, 2020-05-30:
My mistake... I wrote that everything can be destroyed with 5 keypresses but that can actually be achieved with 3 keys: SHIFT + i + y.

lurch, 2020-05-30:
Or you could take out the SD card and cut it in half with a pair of pliers? Zero keypresses needed! :laughing: 

If you're worried about how "easy" PINN makes it to install other OSes (and erase existing OSes) maybe you should be using image-installs of those OSes rather than PINN-installs?

The optional password-protection is an interesting idea though. But given how easy it would be to bypass that (simply put the SD card in a Windows PC and look at recovery.cmdline), I wonder if people might get upset when they realise that the "password protection" is much less secure than they assume it is? :shrug: 

(No offence intended, please accept my comments as tongue-in-cheek)

schauveau, 2020-05-30:
@lurch No offense taken! I am not really worried about people destroying my Pi intentionally. My real problem is that the PINN UI is counter intuitive and error prone. I am more worried about human errors. I would not care much about losing my current rpi2 used as a Kodi server but, being 2 or 3 keys away from total annihilation will not be acceptable for my upcoming rPI 4 that I am planning to use for more serious things.

I am not sure to understand what you mean by using 'image-installs of those OSes". If you are talking about the disk images provided by the OSes then yes I could easily install them manually. However, there is still the problem of having a reliable and simple method to handle multiple boots on the same disk. I looked at NOOBS, BerryBoots and PINN and, apart from the issues mentioned previously,  PINN is the only one that I do not dislike too much.

And of course, keeping the password in clear text is not an ideal good solution. It would be far better to pass a hashed password tin recovery.cmdline but, anyways, that wont help much is a hacker can get access to the boot partition.

schauveau, 2020-05-31:
I am replying to myself because I just figured out that I was wrong about the difficulty to implement multi-boot on the Pi. The official boot sequence documentation gave me the false impression that booting was only possible on the first partition (with some black magic with recovery to boot on other partitions) but I just figured out the LINUX_REBOOT_CMD_RESTART2 system call (i.e. rebootp). So I will probably follow @lurch advice and do everything manually without relying on PINN and co. 
Nevertheless, I still think that you need to do something about your **Autodestroy** ... hummm sorry ... **Install** button.

procount, 2020-05-31:
I have a similar item already on my roadmap so it will get done at some point.

procount, 2021-02-17:
Added a new scarey skull icon for the install button when 1 or more OSes are installed, plus another dialog box to help prevent the AutoDestory scenario (v3.5.5)

aripollak, 2021-02-19:
Would it be possible to have a boot cmdline option to disable erasing/reinstalling OSes? It doesn't have to be on by default, but I'd find it really useful to turn it on so i don't accidentally erase a partition. And if I really really wanted to, I could remove that option from the cmdline.

procount, 2021-02-19:
This was just a first step. I have a more comprehensive solution in mind for the future. But your suggestion might be another step towards my goal.

jharris1993, 2021-02-22:
Greetings!

1.  I would like to object to the skull icon for two major reasons:
   (a)  It's in poor taste and does not reflect well on the tremendous amount of work done on this piece of software.
   (b)  It's not obvious that the "skull" means "install"

2.  I absolutely agree with the original poster's idea that the current UI design for PINN is both counter-intuitive and dangerous.
    (a)  The original UI design comes from a time when the loader was expected to handle one O/S at a time, and the Raspberry Pi was more of an "Arduino on Steroids" than the system it has become today.
    (b) It repeatedly violates the [Principle of Least Astonishment](https://en.wikipedia.org/wiki/Principle_of_least_astonishment), causing a lot of user grief and angst.

3. Until such time as the UI gets a thorough re-design, I would propose a different way of handling this issue:
**_If this is not the first install_**: (_i.e._ If this is not the first time PINN has been run on this media.)
    (a)  Display a _**LARGE**_  danger-prompt, (a skull-and-crossbones would be appropriate here), once the user selects "Install",
    (b)  Implement the confirmation method used by other pieces of software to absolutely confirm that the user knows what he's doing. (_i.e._  Require the user to manually enter a confirming phrase, something like "I know what I am doing" in order to proceed.

The actual problem is that the UI is an artifact of an older, outmoded design and is no longer appropriate for what PINN has evolved into.  The only true solution is an - eventual - re-design of the entire UI.

Note that I do not intend to complain about the tremendous amount of work put into PINN by its maintainers.  It is a tremendous piece of work that is appreciated.  What I do want to say is that it has "outgrown" its original UI paradigm.

jharris1993, 2021-02-22:
Update:

After re-reviewing the title, I agree that the title suggests the best, and most effective, way of solving this issue.
Viz.:
If PINN is being run again after any install, the install button should be disabled.

The basic problem is that the generally accepted meaning of "install" is to **_add_** something, not remove something.  For example, "installing" an extra horn or headlights on your car doesn't imply destroying and removing the entire car to do so.  Likewise, "installing" a window-screen or awning doesn't imply deleting your entire house or apartment.

Even in the context of "installing" an operating system, "install" allows you the option of either adding the new O/S to the existing system, or, (optionally), replace it entirely.

Ergo, either one of two things need to be done:
1.  Disable the "install" button.
2. Have the functionality of the "install" button become "replace".

jharris1993, 2021-02-22:
>  I have a more comprehensive solution in mind for the future.

Would you mind mentioning what some of these ideas might be?

Thanks!

procount, 2021-02-22:
Like I said above, this is a first step towards a more comprehensive solution, so it will evolve.

It was a quick win that I could do immediately, as I took the point that it is too easy to wipe out all your OSes with a misplaced click. I felt it was better to put something in this release, rather than wait for a complete UI redesign. 

Sorry if you don't like the placement of the skull icon. To me it means poison or danger so it is appropriate (I couldn't find a free icon of skull & crossbones which is what I would have preferred). Although I do prefer the icon that is shown before you install any OSes. However, it does have the words INSTALL written underneath it, so its meaning should be fairly intuitive. Maybe a big red overlaid exclamation mark might have been more acceptable?

One of my goals all along has been to retain compatibility with NOOBS, mainly so that I can take advantage of any necessary changes to support the latest firmware, kernel or other necessary features that are needed for the latest hardware models. But it has also allowed me to feed back PRs back into NOOBS to fix and improved things there where they are appropriate. Now that NOOBS is not being majorly developed, I could perhaps break free a bit more, although I doubt I will have enough free-time for a complete UI re-design!

> Ergo, either one of two things need to be done:
>
> 1. Disable the "install" button.
> 2. Have the functionality of the "install" button become "replace".

Option 1 is my preferred solution, but this requires a disruptive change in the architecture, which will take me a while.
I'm not sure just changing the name to Replace will help. Won't that confuse with the other button called Replace on the maintenance menu? 

> The basic problem is that the generally accepted meaning of "install" is to add something, not remove something.

I disagree when it comes to software. In this case "install" nearly always means "add" on the first invocation and "replace", on the second invocation, which is what is happening here. But the major limitation is that NOOBS was not designed to ADD additional OSes.

> Would you mind mentioning what some of these ideas might be?

Disabling the install button is the main one, but it needs integrating with the Replace/Reinstall functionality and the ability to enable it for the whole drive, or individual OSes. It requires some thought how to implement it properly. There are other ideas too, but I don't know how to do them yet.

I hope that answers your comments, but they keep changing and being added to, so my answer might be out of date already! :wink:

procount, 2021-02-22:
Referring to your last thread post: "Instead of simply adding a "scary icon" - which doesn't communicate what you want - simply disable the "Install" button after PINN has installed something for the first time."

How would the user enable that functionality again? I know they can set the runinstaller option again (through the cmdline or via the Gui), but that does more than just wiping the extended partition.

jharris1993, 2021-02-23:
>Referring to your last thread post: "Instead of simply adding a "scary icon" - which doesn't communicate what you want - simply disable the "Install" button after PINN has installed something for the first time."
 >
>How would the user enable that functionality again? I know they can set the runinstaller option again (through the cmdline or via the Gui), but that does more than just wiping the extended partition.

Procount,
I totally agree that the best way to accomplish all of these things is to break from NOOBS and create a completely independent utility with a re-designed UI.  IMHO, 99.999% of the issues spring from trying to turn a sparrow into an eagle, but retain backward compatibility to being a sparrow.

Correct me if I am wrong, but the "install" process is - in essence - identical to the first-boot functionality except that the preliminary formatting and setup is already done for the first two PINN partitions.

In essence, running "Install" performs the same functionality as an initial install after copying PINN to the installation media for the first time - it nukes everything (except for PINN's first two partitions), and creates a fresh, new install of whatever is selected.

Since the "install" functionality is (essentially) identical to the "first run" functionality, the way to solve this problem is to make "Install" only available on first run.  In order to re-enable the functionality the user would have to either re-install PINN on the media or modify the command-line.  Since neither of these are trivial actions, it essentially prevents the user from accidentally destroying an existing installation.

Like I have said before on a number of occasions, PINN's big problem is that it's trying to be too many things at the same time.  You want to maintain strict compatibility with NOOBS, yet you want to extend the program in ways that NOOBS never imagined.

Now that NOOBS is being/has been depreciated, I think it's time to sit down and carefully consider where PINN is going in the future - what features are essential, what features are "nice to have", and what features are more trouble then they are worth.

Going back to my first statement, since "Install" is only relevant to a first-install scenario, it should be limited to first-install.

>I disagree when it comes to software. In this case "install" nearly always means "add" on the first invocation and "replace", on the second invocation, which is what is happening here. But the major limitation is that NOOBS was not designed to ADD additional OSes.

Perhaps this is true within PINN.  With the exception of PINN, I cannot think of any instance where "install" means something other than "add".   Even in the case of a Windows installation, (assuming that there's a pre-existing installation), the installer asks if you want to replace or add.  In the case of Linux installers, (particularly the Ubuntu family of installers), you are given the choice of "replace", "add", or "do something else" if the installer detects something is already there.  In all these cases, "add" is the implied functionality with "replace" being active only if something else exists.

Unfortunately, this is a discussion that is way too long and involved for this one thread and ultimately a UI/paradigm redesign is what is going to solve the problems.  Perhaps a UI discussion thread is in order?

procount, 2021-12-06:
P3.7.4 - The scary skull is removed.
If you choose the Install button, and OSes are already installed, a warning dialog is presented in which you must now type the word DELETE into an Edit box before PINN will continue. 
I hope this is a better way to prevent accidental overwriting of installed OSes.

closed 2022-03-26 by procount