procount/pinn #382: Kali on Raspberry Pi 4

kotpeepers, 2020-06-05:
I noticed that Kali is not listed in the OS for the Raspberry Pi 4 and wanted to see if there was a way to add it via Copy from USB or Custom OS, however the download of Kali is in .img format and when extracting the files to a folder there are no recovery images included with the Kali .img for it to be added into the /os folder.
Do you have any suggestions of how to get it added into the OS list or how to convert it so I have the correct files for the /OS custom install?

procount, 2020-06-05:
PINN (like NOOBS) does not install images, but tarballs.
As most OSes are supplied as images, they need to be converted to the correct format first.
You are welcome to have a go at converting the latest Kali version. You can find instructions in the [Readme ](https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version)or in the [wiki](https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN).
PINN includes older versions of Kali for other RPi models, so if you enable the `showall` option you can download the Kali meta files for another model as a basis for your new conversion.
Note that not all OSes are easy to convert because you may have to deal with removing UBoot or other such things to allow it to boot off a different partition. I think I had a few problems with Kali last time I tried to convert it, so I put it to one side. I would be willing to host your conversion if you manage it,

Dimensional, 2020-11-05:
I managed to create a working conversion, however I'll need to fix the conversion as I made a mistake when editing the cmdline file, which breaks the booting until it's fixed. Also will need to double check it to ensure it's configured the same like a fresh install, since some updates require changing some configurations.

procount, 2020-11-29:
Any more progress on this?

Dimensional, 2020-11-30:
I got the Kali 64bit version working, and I've extracted a 32bit version, but I haven't been able to test it. I can share the exported files  and you can test them, make modifications to the configurations, etc. One thing I had to disable in the configs was the want_maximum, as I was looking to create a specially made shared partition that would be default be mounted by all others systems to share files, or even as a unified /home directory, so that will need to be adjusted when releasing. Is there any place I can upload it to, or should I stick with placing it in a cloud storage and sharing it from there?

procount, 2020-11-30:
I don't have any specific place you can upload it to, so a link to any cloud storage space will be fine eg Google drive, OneDrive, sourceforge, GitHub etc.

Dimensional, 2020-11-30:
https://drive.protonmail.com/urls/1FSQDJ7QSR#sJ2LfJacxMa9 - Kali Full 32-bit
https://drive.protonmail.com/urls/M0QEVTJ1MR#Unjf8NhIwmWi - Kali Full 64-bit

The entire OS folder is stored in each Tar file. I couldn't create a link to an entire folder, so that's why they are in tar files.  If you can figure out why the OS icons don't render in Pinn, let me know. I've tried everything I could think of, which isn't much considering I'm not a graphics guy, but they should be rendering on the Pinn OS Select screen due to them having the same properties as the ones on the repo list.

procount, 2020-11-30:
Thanks, I'll take a look. 
For my reference, do you have a link to the original source images on the kali website, in case I need any additional info?

Dimensional, 2020-11-30:
They are right here: https://www.offensive-security.com/kali-linux-arm-images/

They are the **Kali Linux RaspberryPi 2, 3 and 4 (img.xz)** and **Kali Linux RaspberryPi 2 (v1.2), 3 and 4 (64-Bit) (img.xz)** images.

Dimensional, 2020-12-18:
I've uploaded Kali 2020.4 images to replace the 2020.3 version I was using earlier. I managed to test both of them and get them running with a few other distributions installed beside them. It appears that Kali on Pi by default uses ext3. When I tried to convert them to ext4, I forgot to change that part in the config.txt file, causing kernel panics. However they both will boot with the default ext3 filesystem. If you are able to make them boot with ext4 and still run properly, more power to you, but they are in a workable state and can possible be added to your repo. Since I don't have a lot of cloud storage space, I can't host files like these indefinitely.

The only issue I still haven't been able to resolve is the problem with the icon for the OS menus. They still won't render. If you have any idea how to get them working, please share them.

https://drive.google.com/drive/folders/1o09MzrhY1UCR6vMc7RXUropW5wdT1jQx?usp=sharing

procount, 2020-12-18:
Thanks, I shall download them and have a look.
Your icons probably just have the wrong name. They must have the same base name as the os_name.

Dimensional, 2020-12-18:
The same name as the OS in os.json, or same name as the folder name the files are all in? If it's the latter, they already are. If it's the former, then that wasn't something I noticed or thought about. It took me a while to figure out some of the stuff on the JSON wiki page, like download size, which I assumed was the download size of the root.tar.xz file.

procount, 2020-12-18:
Same name as the OS in os.json. (the enclosing folder name can be anything).
The wiki may be brief, but it's more than I had to start with 😄

Dimensional, 2020-12-18:
Weird issues I've been running into when reinstalling the OSs by rebuilding the table, Pinn refuses to mount the ext3 partition for Kali. I've checked and ext3 supports -O ^huge_file, but the debug log showed that when making the root partition, it only ran ``-O ^huge_file`` instead of ``mkfs.ext3 -O ^huge_file``. It wasn't doing that for anything else. Changing the settings for the cmdline.txt file and fstab to use ext4 instead works, but it's still unusual that it was doing that. I've already made changes to my instance and am in the process of uploading it to my storage, so the debug log is no more, but it's still unusual that when trying to make the file system pinn wasn't running mkfs.ext3 at all.

procount, 2020-12-18:
Just looking at the kernel build and PINN does not include ext3. That might explain it :wink:
I'll add it into the next version for you to see if that helps.

Dimensional, 2020-12-18:
That's weird because when accessing the PINN terminal, I see mkfs.ext3 exists.

procount, 2020-12-18:
Maybe the existing ext4 driver supports ext3 as well 🤷‍♂️

procount, 2020-12-18:
Indeed, ext3 is now handled by the ext4 driver.

Dimensional, 2020-12-19:
I don't have the fastest computer or bandwidth, so it took me a long while to edit both kali instances to boot using ext4 and then upload the changes, but it's done now. Since the only changes I made to these working files was editing cmdline.txt and fstab to use ext4 instead of ext3, they both should still work OOB.

lurch, 2020-12-19:
Yes, ext4 can read/write ext3 (and ext2) filesystems :slightly_smiling_face: 
But https://github.com/procount/pinn/blob/master/recovery/multiimagewritethread.cpp#L1327 doesn't have anything for handling `(fstype == "ext3")` :confused:  (it should probably also have an `else` clause to catch 'unknown' fstype?)

lurch, 2020-12-19:
@Dimensional Just out of curiosity, any idea why Kali are using ext3 instead of ext4?

Dimensional, 2020-12-19:
@lurch Looking over the build script for Kali on Arm, they intentionally picked ext3. No idea as to why they picked that over ext4, or even f2fs or btrfs, which from what I can find are both built into their kernel. https://gitlab.com/kalilinux/build-scripts/kali-arm

Dimensional, 2020-12-19:
The hardware these versions of Kali were made for are listed in the os file, but they won't run on the RPi1. These are the 32 and 64 bit versions for Pi 2 through 400.

lurch, 2020-12-19:
I did some poking around in that repo, and found https://gitlab.com/kalilinux/build-scripts/kali-arm/-/commit/3148a20fb87b7ca66f8bca38f9e04b8017b44669 which says "# 23 July 2020 - Build server produces broken ext4 filesystems.", so it sounds like the choice of ext3 is a problem with Kali's build server, rather than with Kali itself? (so it _should_ still be safe to run Kali from an ext4 partition).

Dimensional, 2020-12-19:
Probably. Feels like this issue has turned into some sort of conversation/forum. But if these are helping us figure out how to get Kali on PINN safely, I guess it'll work.

Dimensional, 2020-12-19:
So because PINN does the partition creation and disables expanding of partitions, it should be safe to use Kali on ext4 partition, just as long as we don't use kali's ext4 drivers to expand or create other partitions.

procount, 2020-12-19:
I've had a quick go at adding ext3 support into PINN. You can download it from here ->
https://sourceforge.net/projects/pinn/files/testing/pinn-3521.zip/download. It may take a few minutes for it to become available as it mirrors.

Please could you check it out with your Kali build?

Dimensional, 2020-12-19:
Not sure how soon I'll be able to test it myself. I've been having to use different SD cards for various stuff, which means I got nothing to spare purely for testing. The only other drive I have I can use to hold images is an NTFS formatted USB HDD that I use with my Windows system.

And I'll either have to edit the setup script to replace ext4 with ext3 in cmdline and fstab, which hasn't always worked for me, or edit them both in the tar files, which will take an hour each to re-compress.

lurch, 2020-12-19:
> it should be safe to use Kali on ext4 partition, just as long as we don't use kali's ext4 drivers to expand or create other partitions.

AFAICT from the link I found, there shouldn't be anything wrong with using Kali on the Raspberry Pi to repartition or resize ext4 partitions on the Raspberry Pi. I think the only problem is that the Kali project's AMD64 build server (which may or may not be running Kali linux itself?) isn't able to create ext4 partitions that Kali's ARM kernel is able to use properly? (I've got no idea what might be causing this problem)

> Still wish btrfs was an option and that some of these OSs supported it in the kernel instead of being a kmod.

I suspect btrfs probably _can_ be compiled into the kernel rather than being a separate kmod, but given how rarely (?) btrfs is used this would lead to a bigger kernel image which wouldn't benefit most people. Whereas ext4 is so widely used it _does_ make sense to compile it directly into the kernel rather than having it be a separate kmod.
So most distros only provide btrfs as a kmod, but if you're willing to build your own kernel you can probably build btrfs directly into the kernel?
(apologies if you already knew all that)

Dimensional, 2020-12-19:
> AFAICT from the link I found, there shouldn't be anything wrong with using Kali on the Raspberry Pi to repartition or resize ext4 partitions on the Raspberry Pi. I think the only problem is that the [...]
From my edits, I included a link where they explained that when kali attempts to resize its ext4 partition, which happens at first boot on Pi, it causes the journal to vanish and corrupts the entire FS. From what they said, it's not just that, but any time a resize is done. Now all that means if as long as it's not Kali that resizes it's own partition but instead PINN or another OS that does it, it shouldn't be a problem. They make it sound like the ext4 resizing process doesn't wait for boot to finish before continuing, unlike the process with ext3. However these are from the Kali 2020.2 version. I haven't attempted to test it on Kali 2020.4 if the bug still exists, and with the use of PINN the bug is circumvented.

https://gitlab.com/kalilinux/build-scripts/kali-arm/-/issues/189#note_361487615
> I suspect btrfs probably _can_ be compiled into the kernel rather than being a separate kmod, but given how rarely (?) btrfs is used this would lead to a bigger kernel image which wouldn't benefit [...]

I'm aware that btrfs in the kernel would increase it's size by a bit, but in actuality it would be beneficial. Btrfs supports subvolumes, snapshots, transparent file compression, and additional checksumming of data. I personally use it to create a shared partition that each OS can use together. This way instead of having to either have multiple copies of the same file, or having to mount each OS partition to access their files and risk breaking something, with btrfs I can create a subvolume that is mounted directly and restricts access to only that subvolume.

Dimensional, 2020-12-19:
@procount While I'm working on building my test environment for Kali with ext3, I recommend an edit to the wiki page about making custom releases for PINN. The only thing I recommend is adding into the xz commands that -T0 option. Running ``xz -9 -T0 -e`` will have the program run in multi-thread mode, thereby speeding up the compression process. By default it uses only a single CPU, and ``-T0`` will have it use every thread available. Took me 10 minutes to compress my Kali root.tar file instead of over an hour. These figures are from compressing using my desktop instead of Pi4. However obviously the cli argument could be any number as long as it's no more than the number of CPU threads on the system.

The feature apparently was added in xz 5.2.0, which was released in 2014, based on the age of this forum thread.

https://www.phoronix.com/forums/forum/software/general-linux-open-source/47329-xz-5-2-adds-new-multi-threaded-options

lurch, 2020-12-19:
> when kali attempts to resize its ext4 partition, which happens at first boot on Pi, it causes the journal to vanish and corrupts the entire FS.

Ahhh, fair enough. Maybe there's an ext4-related bug in whatever kernel-version Kali are using then? Apologies for giving you false information.

> I'm aware that btrfs in the kernel would increase it's size by a bit, but in actuality it would be beneficial, at least on PINN side if/when we are able to do customizable installations in the GUI.

Probably not going to see btfs supported in PINN itself until there are any PINN-installable OSes that use btrfs by default? And even if PINN allowed you to "customize" the installation of an OS to change the root partition's FS to btrfs, that won't work for any OSes that don't already have btrfs compiled directly into their kernel?

Dimensional, 2020-12-19:
It wasn't fake information. You have a valid source. They just had additional information that went into further detail, like what we're doing here. I'm almost done getting the test files compressed. I have the test version of PINN copied onto a spare SD I found, and am waiting for the Kali files to finish compressing before I start it up and get things moved.

procount, 2020-12-19:
I actually support btrfs for Xbian, but haven't sorted out backup for it yet.

Dimensional, 2020-12-19:
> I actually support btrfs for Xbian, but haven't sorted out backup for it yet.

I probably should create a different issue/enhancement thread for this, but do you mean using btrfs for backups? It requires the system to be installed inside a btrfs subvolume automatically. For instance if installing ubuntu 20.04 and selecting btrfs as the root partition, instead of just making all the normal folders, it'll create a ``@root`` subvolume and write all the data into there, then create a ``@home`` subvolume next to the root subvolume, and mount that inside it as the home folder. So to use btrfs to perform snapshots and backups, you'll need to utilize the subvolume feature from the very root of the installation.

A way to test is do what I did, install ubuntu 20.04 inside a virtual machine with it using btrfs, then what you do is reboot and run the live cd again, this time mounting the btrfs partition. Inside you'll see just two folders, ``@root`` and ``@home``, and when looking at ``@root/etc/fstab`` you will see both folders are specified inside with the mount options ``subvol=``

https://fedoramagazine.org/btrfs-snapshots-backup-incremental/

procount, 2020-12-19:
What I mean is, PINN can create a BTRFS partition to install Xbian onto (because it uses BTRFS). But my knowledge of BTRFS was pretty limited when I did this. It's not much better now, but I have been studying the similar ZFS cos I want to use it for my own backup solution. So I was wondering how to a backup a complete BTRFS partition and then restore it. Maybe creating an image file is the only way to do this in an efficient manner. Copying all subvolumes will end up requiring much more storage space due to its CoW behaviour.

procount, 2020-12-19:
Actually, it's only minimal support, since I install Xbian as a raw image file.

Dimensional, 2020-12-19:
The test PINN-3521 won't boot properly. It doesn't load the gui, copy files to memory, etc. Just a blank screen. I've tested to make sure it wasn't something I did wrong with my SD, but the latest release version still works.

procount, 2020-12-19:
OK. 😞 Sorry, thought it was just a minimal change. I'll test the next one first, but it will take a little longer. :wink:

Dimensional, 2020-12-20:
That's alright. We're just focused on Kali. I think we can safely add it to the Repo with ext4 used, since it won't be resizing the partition from within and thereby cause the corruption bug, and any resizing will be done from an external OS. Just download the files from my google drive link, they should be updated and using ext4 already. Let me know if there are any issues booting them.

procount, 2020-12-20:
After extracting pinn-3521.zip, can you try deleting the fixup*.dat files? They shouldn't be there in that version and might be causing the problem - just a long shot.
I already downloaded your files from Google drive yesterday. I'm installing them now after fixing them up a bit.
The nominal partition sizes seem rather large. They should usually be only 500MB bigger than the extracted file size.
You should avoid spaces in the os-name, but if you include them, they need to be replaced with underscores in any file that is derived from them, like the icon files.

Dimensional, 2020-12-20:
Sorry. I had them unusually large for my testing environment, wanting to keep extra space for other things afterwards that the installer couldn't handle.

lurch, 2020-12-20:
> I actually support btrfs...

My apologies once again for spreading false information! :rofl:

Dimensional, 2020-12-20:
> My apologies once again for spreading false information! 🤣

It's an educational experience for all of us. 👍

procount, 2020-12-20:
Oh ok. It's not that then.

procount, 2020-12-20:
Kali 32 bit installed ok, but PINN had lots of write errors when trying to install the 64-bit tar file.

Dimensional, 2020-12-20:
What sort of write errors? I followed the instructions for creating these and they worked for me. Do you regularly check the debug logs automatically, or was this from an error message?

procount, 2020-12-20:
In the debug log, it looked like every file being untarred had a write error, so at the end PINN deleted it. If it worked for you it was probably me, but it was odd because the 32 bit version installed ok. It could be I did something wrong as I haven't had time to try again or look more in depth, but I am juggling lots of balls in the air at the moment! :smile:

Dimensional, 2020-12-20:
The issue may be due to uncompressed tar size in the partitions file. It's smaller than 32 bit, but it probably should be bigger. The uncompressed tar file is reading as 7.68 GBs big on my drives, or 7866.88 MBs big. I'm changing the tar size from 5890 to 7890 to be safe. It's just weird that it didn't cause any problems.

procount, 2020-12-20:
I as installing both of them to the same 16GB SD card, so it could be it ran out of space if the nominal size was too small. The two would only just have fitted as it was.

procount, 2020-12-20:
I just installed the 64 bit version again on its own so it had more space. This time it installed fine. The actual size installed (reported by df) was 8570MB. So the previous time, it just ran out of space because the nominal size was too small.

procount, 2020-12-28:
Kali 2020.04 now added

closed 2020-12-28 by procount

Dimensional, 2020-12-29:
Word of caution, Kali 64-bit won't run on RPi2, but will run on Pi 2B v1.2, as was in the original json file. 

https://www.offensive-security.com/kali-linux-arm-images/ - 32-Bit version will run on RPi 2B and up, but 64-bit's oldest version is RPi 2B v1.2.

https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/revision-codes/README.md

procount, 2020-12-29:
Oops! 
You didn't provide an os_list.json file, so I had to create one and I didn't transcribe it properly.
It should be fixed now. Thanks. :smile:

BTW, it's not necessary to add "Pi 400" as an option since it is covered by "Pi 4", but I left it in anyway.