procount/pinn #383: USB MSD boot for Raspberry Pi 4

raspberrypisig, 2020-06-08:
From your comment here, https://github.com/raspberrypi/noobs/issues/588, I am interested in your progress on this or should I wait for NOOBS to update. Just interested in getting PINN itself to boot from USB/SSD, will you use autoboot.txt to boot the other OSes with updated start.elf/fixup.dat files.

procount, 2020-06-08:
I can post a beta version if you'd like to try it out? Only the basic mods at the moment, but as you will be upgrading OSes yourself it should be sufficient. 
Why do you use autoboot.txt?

procount, 2020-06-19:
Latest PINN v 3.3.4.2 will now allow PINN to boot from USB, but installed OSes may have to be modified by hand to support this boot mode.

closed 2020-06-19 by procount