procount/pinn #385: Support for USB mouse/keyboard switchers

dominic-03, 2020-06-15:
I just got a (rather old) USB device switcher, where you plug it in to 2 computers, then plug your mouse and keyboard in, and press the Scroll Lock key twice to switch between devices. In Windows 10, this works as intended, allowing control of the mouse as normal. However, in PINN the mouse is extremely erratic and practically unusable. Is this a problem that can be fixed?

(If the model no. of the switcher is needed, I can provide it.)

lurch, 2020-06-15:
Is this specific to PINN, or do you see the same behaviour in e.g. Raspberry Pi OS ?

dominic-03, 2020-06-15:
@lurch I haven't been able to test it in Raspberry Pi OS yet but I'll do that soon.

dominic-03, 2020-06-15:
Ok, just tested in Raspbian, it has the same problem. I'll reach out to the people on the forums to see what they can do.

closed 2020-06-15 by dominic-03