procount/pinn #388: Installation of existing OS fails on checksum mismatch. Instructions update request.

jotaemei, 2020-06-19:
Note: This is in regards to an OS (Picroft: https://github.com/MycroftAI/enclosure-picroft) that is not in the SourceForge repository.

Following the guide at https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN, in steps 1 and 2, there are no instructions to acquire and store the checksums for the 2 partitions.

When I failed to do so, the installation reported 2 failed checksums, allowed me to continue, and reported they were installed with errors. It then showed the OS as installed, but I was never given a menu option to boot to it, as it was apparently not detected.

I ran the checksums on its two compressed tar partitions, updated partitions.json with them, ran a quick reinstall from PINN, and everything was resolved.

I guess the instructions at https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN need to be updated to tell people to carry out this step, particularly if they're going to be installing new OSes that were not originally installed from the SourgeForge repo.

procount, 2020-06-19:
Thanks for reporting this. Indeed I forgot to update the wiki tutorial with the new checksum feature. 

I suppose you started from the latest Raspbian OS meta files that have checkums in them already?
Checksums are in fact optional, so another alternative is to just delete any checksum entries in os.json. os_list.json and partitions.json. 

I will updated the wiki accordingly.

procount, 2020-06-19:
@jotaemei - I've updated the wiki. If it looks ok, please close this issue.

closed 2020-06-22 by jotaemei