procount/pinn #390: USB-MSB boot requires new software

proctologic, 2020-06-20:
start4.elf fixup4.dat is old versions that not support usb-boot
\Pinn\os\RaspiOS_Full_armhf\boot.tar.xz
use the files from pinn/files/os/raspios64/boot64.tar.xz

gotouchgo, 2020-06-22:
Hi there, do you know how to make USB-MSB work? I create a small fat32 partition and copy all files from pinn-lite.zip on to USB SSD drive. It only show the rainbow square screen but can't boot up. This is on a Pi 4 B 8GB model. Thanks for any clue.

lurch, 2020-06-22:
Have you updated the bootloader on your Pi4? See https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

gotouchgo, 2020-06-22:
> 
> 
> Have you updated the bootloader on your Pi4? See https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

Yes I did. I followed https://www.tomshardware.com/how-to/boot-raspberry-pi-4-usb and use firmware dated 2020-05-15. It works with the official plain Pi OS image booting from USB SSD and installation.

lurch, 2020-06-22:
Probably better to use the [latest firmware](https://github.com/raspberrypi/rpi-eeprom/releases) dated 2020-06-15? Although I've no idea if that'll fix your problem or not! :wink: 

Also, I assume you're using the version of PINN that #383 refers to?

proctologic, 2020-06-22:
easiest way to get pinn to boot from SSD is use pinn-lite.img.zip https://sourceforge.net/projects/pinn/files/Etcher/
Use Etcher or Rasperry Pi Imager
 My problem is in the Pinn full version and Raspian OS 32bit. 64 Bit works fine.

lurch, 2020-06-22:
I believe that once you've edited `/etc/default/rpi-eeprom-update` to tell it to use the `stable` release-series, you only need to run `sudo rpi-eeprom-update -d` in the last step, and it'll automatically use the latest `stable` bootloader, no need to manually specify a bootloader filename.

Also, note that `rpi-eeprom-update` only copies the update-files to the `/boot/` partition, so you need to add this to the end of your list:
5. reboot the Pi with the SD card still inserted and this will then update the bootloader

gotouchgo, 2020-06-23:
Re: My problem is in the Pinn full version and Raspian OS 32bit. 64 Bit works fine.

Per the Tomshardware guide, thr latest *.dat and *.elf screw up USB boot and the files from 05-22 are needed. When I once tried with the latest files and official Pi OS 32-bit image, I also saw the error message like "software is too old, start4.elf is not correct, newer software is needed". So this could be the same issue you have seen. Just need to package the 05-22 files in the tarball.

gotouchgo, 2020-06-23:
My bootloader firmware is actually dated 06-15. It should be good as I can boot the plain Pi OS from USB SSD just fine. Just tried copy all files from pinn-lite.zip to my SSD boot partition. Still the same problem when booting, stuck at the rainbow square screen. Don't know what may cause the kernel not loading. I copied the backed up boot files back and it boots correctly.

~$ sudo vcgencmd bootloader_version
Jun 15 2020 14:36:19
version c302dea096cc79f102cec12aeeb51abf392bd781 (release)
timestamp 1592228179
:~$ sudo vcgencmd bootloader_config
[all]
...
BOOT_ORDER=0xf41

proctologic, 2020-06-23:
I know how to fix it. i did this to make devs aware of it and include it in \Pinn\os\RaspiOS_Full_armhf\boot.tar.xz

lurch, 2020-06-23:
There seems to be a bit of confusion here. In order to successfully boot from a USB-MSD you need both a) a bootloader in the EPPROM that's new enough to support USB booting and b) a start.elf & fixup.dat (or start4.elf & fixup4.dat) on the SD card that's new enough to support USB booting.
These are both released independently, so you shouldn't assume that the date of one is relevant to the date of the other.

When using PINN, then the start.elf needs to be "new enough" in both PINN itself and _also_ in each of the OSes that you're booting. AFAICT the start.elf included in the 32-bit (stable) release of Raspberry Pi OS isn't new enough to support USB booting, but the start.elf included in the 64-bit (beta) release of Raspberry Pi OS is new enough.
I'm sure that in a couple of months or so there'll probably be a new release of 32-bit Raspberry Pi OS (and maybe a new release of NOOBS?) that uses a new-enough start.elf to support USB-booting "out of the box".

Unfortunately, there seems to be some "cargo cult" knowledge floating around the internet which says something along the lines of "copy all 14 files from the boot partition of one OS to the boot partition of another OS" - this _may_ work in some cases, but this will also copy the kernel of the first OS to the second OS, and this might cause the second OS to fail to boot (as the kernel in the boot partition will no longer match the kernel-modules on the second OS's root partition).

USB-booting on the Pi4 is still considered a beta-level feature, with lots of development recently, so many of the 3rd-party guides or tutorials are out of date already.

lurch, 2020-06-23:
> I know how to fix it. i did this to make devs aware of it and include it in \Pinn\os\RaspiOS_Full_armhf\boot.tar.xz

That's the 32-bit version of Raspberry Pi OS, which as I mentioned above doesn't yet support USB booting out of the box. I think @procount is entirely justified in shipping this as-is without modifying it to add a beta-version of start4.elf to it.

procount, 2020-06-23:
There were some reports that those who had upgraded their eeprom to support the beta USB-boot feature on the Pi4 could no longer load NOOBS, PINN or Berryboot because their start4.elf or recover4.elf files were not new enough. This is the main reason that PINN now has upgraded firmware.

A consequence is that PINN can now also boot from USB, but it will be a while until all OSes are upgraded to support this feature. Only a few OSes support this already e.g. LibreELEC. For the remainder, you just have to have patience, or upgrade them yourself, but be sure you know what you need to upgrade. Once it is finally released (not just stable) I **_may_** add support to upgrade OSes automatically, but that is a while off yet as my time is a little limited atm.

lurch, 2020-06-23:
> There were some reports that those who had upgraded their eeprom to support the beta USB-boot feature on the Pi4 could no longer load NOOBS, PINN or Berryboot because their start4.elf or recover4.elf files were not new enough. This is the main reason that PINN now has upgraded firmware.

Oh, even when booting from SD-card? I've not done much experimenting with this myself yet, so I'd been assuming the "start4.elf not new enough" messages were only seen when people (who'd upgraded their bootloader) were trying to boot from USB?

procount, 2020-06-23:
Not sure, but I did it just in case this was also an issue.

gotouchgo, 2020-06-23:
The "start4.elf not new enough" is only when booting from USB. The thing is, 

1. bootloader dated 2020-04-16 is able to boot from USB. And it is released as "stable" dated 06-15.
From https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

> Version: pieeprom-2020-04-16.bin
> 
>     Boot mode 0x0 will retry the SD boot if the SD card detect pin indicates that the card has been inserted or replaced.
>     The default boot mode is now 0xf41 which means continuously try SD then USB mass storage.

2. start*.elf, fixup*.dat dated 05-22 is able to boot from USB. Per the guide on Tomshardware I used.

3. Rasp Pi OS dated 5-27 is shipped with *.elf, *.dat that doesn't boot from USB, instead it emit the "start4.elf not new enough" message. Just these files are needed on boot partition to enable an OS to boot from USB. kernel image should not be replaced.

I think other OS images probably will enable booting from USB on their own schedule. But PINN should not wait for other OS images to be ready. PINN should be made available from USB, then give warning that user should confirm the distro if then can boot from USB. Or maybe copy the *.elf, *.dat files to the boot partition of the OS, after user agrees through the UI.

lurch, 2020-06-23:
> The "start4.elf not new enough" is only when booting from USB

Thanks for confirming :+1: 

> Rasp Pi OS dated 5-27 is shipped with *.elf, *.dat that doesn't boot from USB

Correct. It's a bit convoluted, but it _is_ actually possible to work out which firmware-release (i.e. start4.elf etc.) is included in the 2020-05-27 release of Raspberry Pi OS (32-bit)...
1. Go to https://www.raspberrypi.org/downloads/ click on "Raspberry Pi OS" and then click on "Release notes"
2. http://downloads.raspberrypi.org/raspios_armhf/release_notes.txt says "Raspberry Pi firmware 21e1fe3477ffb708a5736ed61a924fd650031136"
3. Jump over to https://github.com/raspberrypi/firmware and get the corresponding commit
4. https://github.com/raspberrypi/firmware/commit/21e1fe3477ffb708a5736ed61a924fd650031136 was released on 12 May, which is obviously from before the first version of start4.elf that included USB booting

The reason for the "big gap" between 12th May and 27th May is because each Raspbian release goes through extensive testing to make sure it's stable, it doesn't just get pushed out with whatever beta-firmware happens to be the latest in https://github.com/raspberrypi/firmware

Bit of a lengthy explanation, but hopefully that explains the "discrepancy" you're seeing and emphasises my "you shouldn't assume that the date of one has any relevance to the date of the other" comment from earlier :wink: :smiley: 

> kernel image should not be replaced.

Correct. But I've seen at least one "USB Boot on Pi4!!!" youtube video which says copy all the files from one /boot partition to the other /boot partition :slightly_frowning_face: 

> PINN should be made available from USB

I've not tried it myself, but @procount has said that PINN can now boot from USB.

> Or maybe copy the *.elf, *.dat files to the boot partition of the OS, after user agrees through the UI.

See @procount 's earlier comment... "I **_may_** add support to upgrade OSes automatically, but that is a while off yet as my time is a little limited atm."

gotouchgo, 2020-06-23:
> 
>     PINN should be made available from USB
> 
> I've not tried it myself, but @procount has said that PINN can now boot from USB.
>
Maybe it is just me. @proctologic was able to boot it. I don't see any console message but only the rainbow square screen. Not sure if kernel is loaded but initrd ramdisk seems not loaded to me. Not sure if it is the *.elf issue. I can try the known good *.elf *.dat with PINN and see what happens.

jtremblant, 2020-06-24:
> > ```
> > PINN should be made available from USB
> > ```
> > 
> > 
> > I've not tried it myself, but @procount has said that PINN can now boot from USB.
> 
> Maybe it is just me. @proctologic was able to boot it. I don't see any console message but only the rainbow square screen. Not sure if kernel is loaded but initrd ramdisk seems not loaded to me. Not sure if it is the *.elf issue. I can try the known good *.elf *.dat with PINN and see what happens.
> 
> Ah, PINN is booting recovery image and don't use those .elf, .dat files directly. There is a tarball of firmware but I don't know how to update it. So I still can't boot PINN from USB.

Same situation here. I'm stuck with raspbian 32 bit on my Rpi 4 8gb because of the USB boot firmware mess. I'm really missing PINN Os and Ubuntu Server/mate 20.04 LTS on my Rpi. Patience is a virtue.

lurch, 2020-06-24:
> Ah, PINN is booting recovery image and don't use those .elf, .dat files directly

IIRC `recovery.elf` / `recover4.elf` are just renamed versions of `start.elf` / `start4.elf` :slightly_smiling_face: 

> because of the USB boot firmware mess

It's still a feature in the beta-test stage - "mess" is to be expected! :laughing: 

> Patience is a virtue.

Indeed :smile_cat:

gotouchgo, 2020-06-24:
eh, OK. and we never know what version of start|4.elf,dat is used. Maybe I can try replace them with the 05-22 one and see how it goes.

The USB boot thing is quite easy. Even the official image can be simply dd-ed to USB SSD, and after replacing with the 05-22 *.elf *.dat files, it boots just fine. Once PINN, BerryBoot wants to handle this it is not hard. But for now, the official image is the easiest way to go.

lurch, 2020-06-24:
> eh, OK. and we never know what version of start|4.elf,dat is used.

Yes we do :stuck_out_tongue_winking_eye: 
Inside the PINN zipfile there's a file called `BUILD_DATA`. If we open that file it says `rpi-firmware Git master @ cb2b95d73e9f0b1ebf05e03bb1959603d982feeb` and we can use the same method as I showed earlier to find that https://github.com/raspberrypi/firmware/commit/cb2b95d73e9f0b1ebf05e03bb1959603d982feeb was released on the 10th of June.
Just out of curioisity, why are you so insistent on using your "05-22" version all the time?

> Once PINN, BerryBoot wants to handle this it is not hard. But for now, the official image is the easiest way to go.

Indeed, welcome to the fun world of beta-testing :grin:

gotouchgo, 2020-06-24:
05-22 .elf .dat files are proved to be working by Tomshardware arcticle and myself. I can boot the official Pi image with it. Later ones are not good for USB boot. Yes it is regression while they made ither fixes. I have not tried if the 6-10 files in PINN works separately. It has been a month now. I will try the latest one to see if it works. It is easy to try, just copy to my exist /boot and reboot. Can be reverted easily.

Another thing in PINN is it uses the .elf but not the paired fixup .dat file. I am not technically savvy on Pi and just curious. This is different from boot files in other OS or recover image.

lurch, 2020-06-24:
>  Later ones are not good for USB boot. Yes it is regression while they made other fixes.

If there _was_ a regression, I'd be very surprised if it wasn't already fixed by the latest files in https://github.com/raspberrypi/firmware/tree/master/boot  - who knows, the latest files might improve compatibility with your USB drive? :slightly_smiling_face: 

> Another thing in PINN is it uses the .elf but not the paired fixup .dat file.

Yeah, I never fully understood that myself :confused:

gotouchgo, 2020-06-24:
Ok, good news. Just tried the firmware dat/elf files dated 0610 (used by PINN) and latest (0619). They are all good for USB boot and works fine. Then now I have no idea why PINN doesn't boot for me.

procount, 2020-11-29:
Can this be closed? Several verions of PINN have been released since this date and I'm seeing no USB boot issues.

lurch, 2020-11-29:
@procount I suspect a lot of the confusion in this thread stemmed from the fact that the beta Pi4 EEPROM and the beta 64-bit Raspberry Pi OS image supported USB boot, but the stable 32-bit Raspberry Pi OS image didn't support USB boot. (along with people getting over-excited about USB boot support?)

However since then the stable Pi4 EEPROM and the stable 32-bit Raspberry Pi OS image have both been updated with USB boot support, so AFAIK everything should now just work "out of the box" :)

Although if you wanted to get clever, I guess you could add an option to PINN's `os.json` files to signify whether the `start4.elf` used by a particular OS is new enough to allow that OS to USB-boot on a Pi4 ? :wink:

procount, 2020-11-29:
@Lurch - I suspect you are right, so I'll close it.

> Although if you wanted to get clever, I guess you could add an option to PINN's os.json files to signify whether the start4.elf used by a particular OS is new enough to allow that OS to USB-boot on a Pi4 ? 😉

I think it's complicated enough already! In fact I think it may be time to remove those fixes that automatically upgraded the firmware of various OSes if they were too old. I think enough water has passed under that bridge now. BTW - I've implemented the USB boot without partuuid now, and moved the build env to Ubuntu 20.04.

closed 2020-12-01 by proctologic