procount/pinn #393: Error downloading distribution list from the internet

KilliK69, 2020-06-27:
Hello.
I updated PINN to V3.3.4.2 and now  I am getting the error message of the title.
There is no problem with my network connection, since I used it to do the update without any problems.
I have an RPI3B+
here is a photo of the full error message:
https://i.imgur.com/ECsFmWZ.jpg

procount, 2020-06-27:
It looks like you are using Matt's webpage to control the size of your OS partitions, so this is where the problem is. Did you create that a while ago, because it includes Raspbian, which is now called raspios. Suggest you go to his webpage and redo your selections.

KilliK69, 2020-06-27:
yes, I used  his  page  to install a  few OS in a specific  partition sizes  some time  ago.  Actually you recommended it when  I asked about it in the  RPI  forum.
What exactly should I redo?  I dont  understand.

procount, 2020-06-27:
I don't think this is necessarily a problem of you upgrading PINN to 3.3.4.2, but due to the fact that Raspbain has changed it's name to Raspios.
When you went to Matt's webpage, you will have downloaded a new recovery.cmdline file that actually causes new meta data to be downloaded for the OSes that you have selected. But these names are hardcoded into the recovery.cmdline files and the OSes do not exist any more, so Matt's page can't provide the meta data.

So you just need to restore recovery.cmdline as follows:
1. Boot PINN, and press the shift key to enter the recovery menu.
2. Press the M key twice to go to the Maintenance Menu,
3. Highlight PINN and select Edit Config.
4. Click the Edit Options button (bottom left)
5. Select the source tab.
6. Delete everything in the alt_image_source= edit box.
7. in the repo_list= edit box, enter "http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json
8. Click OK, then OK again.
9. Now reboot PINN and you should be ok.

procount, 2020-08-24:
No response, so assuming this is fixed. Closing.

closed 2020-08-24 by procount

KilliK69, 2021-04-05:
I am sorry for the delayed response. yes, your guide fixed the problem. Thank you very much.