procount/pinn #394: Pinn lite--cannot boot volumio after installation in pi4

Desmond88, 2020-06-30:
Hi there,
I have downloaded the latest pinn lite and put it into sd for my pi4 according to instructions. Thereafter, I have installed raspbian, libreelec and volumio oses. I can boot rasbian and libreelec but cannot boot volumio. May I know to resolve it? I have tried several times, redownloading and reinstalling but still cannot boot volumio. Anybody can help? Thank you in advance.

procount, 2020-06-30:
ping @matthuisman

Desmond88, 2020-07-02:
Anybody can assist? Is volumio compatible with pi 4?

Desmond88, 2020-07-04:
My hardware is pi4 8gb and SanDisk 32 sdhc. I was told the kernal needs to change for 8gb for booting volumio. How to go about it?

procount, 2020-07-04:
You could copy it from the latest raspios 32 bit distro.
Are you sure they meant kernel and not firmware?
Try copying the firmware first - *.day and *.elf in boot partitions. Copying the kernel will depend on the architecture of the distro.

Desmond88, 2020-07-04:
I'm a beginner.  I dont know what goes wrong. One funny thing is after installting raspbian,  librelec and volumio from pinn lite. I can see the volumio folder in raspbian os.  Neverthless., it cannot boot it.  I tried to use a different sd card,  downloaded volumio os directly from volumio website but still can not boot. Oh god, please help me.

Desmond88, 2020-07-04:
I have posted the same question in volumio forum.  One user claimed that he has sucessfully booted the volumio os from obtained from directly from it website for his pi 4 2gb ram. But my is 8gb ram. Other user claimed that volumio is working with raspberry to change the kernal for 8gb ram.  It is coming soon.Please pardon my ignorance,  does different ram require different configuration?

matthuisman, 2020-07-08:
@Desmond88 

Raspberry PI 4 support was added in version 2.599
https://community.volumio.org/t/volumio-changelog/1446

current version in PINN is 2.692
I'll update to latest 2.779 and let you try that

Desmond88, 2020-07-10:
Thank you matthuisman. 
Can you give me a heads-up when the support is available?
Thank you so much.

Desmond88, 2020-07-10:
Hi matthuisman,
I did try to install volumio directly with the latest version 2.779  instate of using pinn. But it did not boot ie. The sreen is  blank ( nothing hapenning) after inserted the boot sd card into pi.

Desmond88, 2020-07-23:
Hi Matthuisman,

Is it possible to include the latest Moode 6.7.0 os  in Pinn Lite? It works beautifully for my Pi4 8gb? Also, LibreElec in Pinn Lite is outdated. It failed to reboot. Can you help to update it?

Thank you so much.

Desmond88, 2020-08-05:
Hi Matthuisman,

Volumio has released a new software version 2.806. I have tested it and it works for Pi4, 8gb ram.

Can you help to update this new version in Pinn Lite if not too much to ask?

Thank you in advance.

procount, 2020-11-29:
@matthuisman - I'm doing some tidy up. Is this something you can help with?