procount/pinn #395: Manjaro install image needs updating/partition label names

busywait, 2020-06-30:
Latest Manjaro ARM expects particular partition names, BOOT_MNJRO and ROOT_MNJRO.

After I installed Manjara XFCE from PINNS and updating the kernel (I moved to linux-rpi4-mainline) I hit the issue described here: [https://forum.manjaro.org/t/raspberry-pi-kernel-changes/146653](url)

I needed to boot in to another installed OS and change the partition names.

I *think* this could have been avoided if PINNS using a more recent Manjaro image.

procount, 2020-06-30:
Thanks for the heads up. I'll look into updating the version. 
It could perhaps be avoided by modifying the partition_setup.sh script as well to account for this.
Also, you can use the recovery terminal in PINN to change the partition names if you don't have another OS installed.

busywait, 2020-06-30:
> It could perhaps be avoided by modifying the partition_setup.sh script as well to account for this.
The change seems sensible anyway, it's easier to distinguish from the other OS boot and root now :)

> Also, you can use the recovery terminal in PINN to change the partition names if you don't have another OS installed.
I should have looked harder - I did go in to the maintenance UI, but didn't spot the place to rename partitions. (In fact, I assumed that it was deliberately not an option in case an OS required a particular name, as in this case).

procount, 2020-06-30:
The only problem with them going with fixed labels, is that it makes installing 2 copies of Manjaro on the same SD card a bit problematic, like in Ubuntu. But maybe it can be accounted for in the script.

Re: renaming partitions, it's not a feature of PINN's Gui. You have to drop to a command terminal and do it manually.

busywait, 2020-07-03:
Is the PINN install configuration and partition_setup.sh for Manjaro in the PINN source tree? Or, where should I look to see it?

busywait, 2020-07-03:
I posted on to the Manjaro ARM forum about this https://forum.manjaro.org/t/raspberry-pi-kernel-changes/146653/20?u=busywait to find out if there is a reason the Manjaro devs chose not to set root using UUID.

procount, 2020-07-03:
See the appropriate Manjaro folders in https://sourceforge.net/projects/pinn/files/os/

procount, 2020-07-03:
You can't have 2 partitions with the same name. If you install 2 manjaros side by side, PINN will give the 2nd one different partition names. DOn't rename the partitions, but change the references. in fstab and cmdline. Then they will work ok (until the next kernel upgrade). You can try running the fixup script from the PINN gui after a kernel upgrade (on a fully backed up & tested or discardable image), which may sort it for you.

busywait, 2020-07-04:
Thank you. I played with identically named partitions to see what would happen - I know that PINN will always ensure sensible unique names.

With my current PINN install re-running partition_setup.sh from the "Fix Partitions" UI in PINN fixes any actual problem booting after a kernel upgrade, but it uses /dev/mmcXXXX device names which Manjaro are deliberately avoiding. 

There are some "fixes" for the current PINN Manjaro configurations that would be helpful to keep the image more consistent with the base Manjaro image:

It seems to be a simple change to fix the partition names for Manjaro in the two PINN partition.json files
./os/manjaro-rpi4-XXX/partitions.json
so that the label attributes are correct:
   "label" : "BOOT_MNJRO",   
   "label" : "ROOT_MNJRO",

In ./os/manjaro-rpi4-XXX/partition_setup.sh don't replace these with device names, instead use labels:
sed /tmp/1/cmdline.txt -i -e "s|root=[^ ]*|root=${id2}|"
sed /tmp/2/etc/fstab -i -e "s|^[^#].* / |${id2}  / |"
sed /tmp/2/etc/fstab -i -e "s|^[^#].* /boot |${id1}  /boot |"

With that, a single PINN install of Manjaro would replicate the base Manjaro disk image looks like, and completely future-proof the install for the fraction of people that only install a single Manjaro version per disk/card via PINN.

Users with 2 Majaro installs on the same card will need to manually patch up the changes that will be made by future linux-rpi4 (kernel) package upgrades. Two options: 
- Replace /boot/cmdline.txt with /boot/cmdline.pacsave
or
- Run PINN > Maintanence > Fix (partition_setup)

The final "easy" and helpful change is to upgrade the PINN tarballs from 20.04 to 20.06, which would save users from finding a conflict in the first package upgrade of a 20.04 install.

busywait, 2020-07-07:
There are lines in the Manjaro partition_setup.sh files related to resizing the root partition which still reference raspi-config. Seems inappropriate? There is no raspi-config in the Manjaro image.

Manjaro handles partition expansion via a script that is deleted after the first boot:
/usr/share/manjaro-arm-oem-install/manjaro-arm-oem-install

The lines related to resizing are 
    msg "Resizing partition..."
    resize-fs 1> /dev/null 2>&1
The script is called from /root/.bash_profile after auto-login of the root account during the first boot. 

The script is part of a pacman package, and the source is hosted at https://gitlab.manjaro.org/manjaro-arm/packages/community/manjaro-arm-oem-install/-/tree/master

Ae3NerdGod, 2020-07-25:
> Users with more than one Majaro install on the same boot device will need to manually patch up the changes that will be made by future linux-rpi4 (kernel) package upgrades. Two options:
> 
> * Replace /boot/cmdline.txt with /boot/cmdline.pacsave after any linux-rpi4 (kernel) package upgrade
>   or
> * Run PINN > Maintanence > Fix (partition_setup)
> 
> (The "first" Manjaro install in the partition table will not have a problem.)
> 
> The final "easy" and helpful change is to upgrade the PINN tarballs from 20.04 to 20.06, which would save users from finding a conflict in the first package upgrade of a 20.04 install (because of a little oversight in the Manjaro repo).

good thing i read this, saved me a troubleshoot.

my partition setup makes a .hook file in /usr/share/libalpm/hooks/60-linux-rpi-pinn.hook so if cmdline.txt or fstab gets changed by any package it fixes things based on the partition names when the OS was installed
i tested it using the mainline package you installed, worked fine.
It also removes the resize command from the script you mentioned earlier so pinn can take care of it.
take a look:
http://bowtie-ent.com/pinng/os/Manjaro-ARM-kde-plasma-rpi4-20-06/manjaro-setup.sh

heres the actual code:
```
if [ ! -f /tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook ]; then
echo "[Trigger]" >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "Type = File" >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "Operation = Install" >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "Operation = Upgrade" >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "Operation = Remove" >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "Target = boot/cmdline.txt" >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "Target = etc/fstab" >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "[Action]" >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "Description = running PINN post install partition setup..." >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "When = PostTransaction" >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "Exec = /usr/bin/sed /boot/cmdline.txt -i -e \"s|root=[^ ]*|root=${id2}|\" " >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "Exec = /usr/bin/sed /etc/fstab -i -e \"s|^[^#].* / |${id2}  / |\" " >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
echo "Exec = /usr/bin/sed /etc/fstab -i -e \"s|^[^#].* /boot |${id1}  /boot |\" " >>/tmp/2/usr/share/libalpm/hooks/60-linux-rpi-pinn.hook
fi
```

it seemed the best way was to make the whole file using partition-setup.sh, to avoid downloading the hook file from elsewhere, and avoid making a separate file on each partition containing the pinn partition labels