procount/pinn #396: Add twister OS

procount, 2020-06-30:
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&start=1025#p1683085

Ae3NerdGod, 2020-07-30:
Twister OS tested working fine ( with the tarball exploit )

i put 4gb ram only because i assume /tmp is tmpfs loaded in ram... if needed i can make it download to the actual partition the os will be installed in which would avoid that but be slower most likely. let me know if i need  to change 

https://github.com/Ae3NerdGod/PINNg/commit/de883fdd6cb8ff2eb4e88c84749b81b18bf302e0

http://raw.githubusercontent.com/Ae3NerdGod/PINNg/master/os_list_public.json

Ae3NerdGod, 2020-07-30:
got it to download and extract at the same time
https://github.com/procount/pinn/issues/403#issuecomment-666346602

Ae3NerdGod, 2020-08-03:
ok, migrated to sourceforge now that i realized its the webclient that has a 1gb upload not the site itself. whoops. twisted no longer uses the exploit, and os_list.json is in the same place at github, not moving it. the custom oses i make will be in another json

mbernardi1961, 2020-09-23:
Can I add a request for Twister to be added as one of the defaults available in the vanillia PINN configuration. Probly more useful than 3 versions of Raspberry Pi OS 32 bit ;-)

Ae3NerdGod, 2020-10-01:
No idea if procount plans to add my repo or not, but it has twisteros1.5(I'm pretty sure) and it works really well, easily updated via it's own updater on the website. 
You can add my repo along with the vanilla repo just in case you're unaware, by adding `repo=https://raw.githubusercontent.com/Ae3NerdGod/PINNg/master/os_list_public.json ` to cmdline.txt

I'll update it soon, I've been working on some huge arcade images that take a long time to copy and compress and such. Need more harddrives lol

procount, 2020-11-02:
Added twister

closed 2020-11-02 by procount