procount/pinn #397: Support F2FS (kernel FS driver and F2FS tools)

busywait, 2020-07-02:
I would like to be able to back up an OS, reformat its partition to F2FS, and restore the OS into the F2FS partition.

procount, 2020-07-02:
You could probably do this manually, by backing up, formatting as f2fs, then manually restoring the root tar.xz file. Your kernel would obviously have to support f2fs and there maybe some config changes to do as well.

PINN does not support f2fs atm, so if you did this you would no longer be able to backup the os and maybe some other features would not work.

I would need to add f2fs support to pinn for it to work fully.

busywait, 2020-07-02:
I'd forgotten about needing to edit the OS /etc/fstab or for the change in partition type, and I've also just learned that Manjaro/Arch doesn't put an entry for the root filesystem in to /etc/fstab by default at all. Instead the rootfs and options are only those passed on the kernel command line: [https://bbs.archlinux.org/viewtopic.php?id=242116](url). 

In my Manjaro install the /boot/cmdline.txt includes
root=LABEL=ROOT_MNJRO rw rootwait 

In this case I'm expecting that the filesystem type is autodetected, and no change is needed. I'll try it and see.

Is this feature suggestion appropriate for PINN, or does it take it away from being an OS installer and selector?

procount, 2020-07-02:
I'm not sure about the conversion but certainly I may need to provide support to install or backup f2fs partitions.

Ircama, 2024-12-29:
When running PINN on the first generation Pis including Zero/W (armhf), F2FS is not available in the Kernel of the rescue shell, but can be loaded as a Kernel loadable module.

The following procedure creates *crc32_generic.ko* and *f2fs.ko* by [rebuilding PINN](https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-rebuild-pinn) with Ubuntu 22.04.5 LTS or Ubuntu 24.04.1 LTS (e.g., running with WSL).

```bash
sudo apt-get update
sudo apt-get -y upgrade

sudo apt-get install -y python-is-python3 build-essential rsync texinfo libncurses-dev whois unzip bc libssl-dev libgmp-dev make git gcc g++ lhasa flex bison gettext ncurses-dev autoconf libmpfr-dev libmpc-dev wget cpio

wget https://www.libarchive.org/downloads/libarchive-3.3.1.tar.gz
tar xzf libarchive-3.3.1.tar.gz
rm libarchive-3.3.1.tar.gz
cd libarchive-3.3.1
./configure
make
sudo make install
cd ..

git clone https://github.com/procount/pinn.git
cd pinn
vi ./buildroot/package/m4/m4.mk  # set M4_VERSION = 1.4.19
rm ./buildroot/package/m4/*.patch
./make_arch_linux-menuconfig.sh armv6  # take long time and exit with spill_indirect_levels++ error
```

On spill_indirect_levels++ error:

```bash
vi ./buildroot/output/build/host-gcc-initial-4.9.2/gcc/reload1.c  # spill_indirect_levels++ modified to spill_indirect_levels = 1
./make_arch_linux-menuconfig.sh armv6  # take long time and exit again with spill_indirect_levels++ error
```

Again on spill_indirect_levels++ error:

```bash
vi ./buildroot/output/build/host-gcc-final-4.9.2/gcc/reload1.c  # spill_indirect_levels++ modified to spill_indirect_levels = 1
./make_arch_linux-menuconfig.sh armv6  # run menuconfig
```

Add F2FS suppport as a kernel loadable module:

- File systems > F2FS filesystem support: select M; select all subitems with an asterisk, but LZ4 compression
- Exit > Exit > Yes

```bash
./BUILDME.sh  # take long time
```

Copy *crc32_generic.ko* and *f2fs.ko* to the RECOVERY filesystem from the following directories:

- ./buildroot/output/target/lib/modules/5.10.103-rescue/kernel/fs/f2fs/f2fs.ko
- ./buildroot/output/target/lib/modules/5.10.103-rescue/kernel/crypto/crc32_generic.ko

or:

- \\wsl.localhost\Ubuntu-22.04\home\USER\pinn\buildroot\output\target\lib\modules\5.10.103-rescue\kernel\fs\f2fs
- \\wsl.localhost\Ubuntu-22.04\home\USER\pinn\buildroot\output\target\lib\modules\5.10.103-rescue\kernel\crypto

These two files are included in the [f2fs.zip archive](https://github.com/user-attachments/files/18269552/f2fs.zip), which can be downloaded.

We assume that the SD has an ext4 root filesystem in the /dev/mmcblk0p7 partition and then mounts a /data F2FS filesystem in the /dev/sda1 partition.

The following commands show how to mount /data F2FS filesystem:

```bash
ssh root@192.168.... (password: raspberry)
```

After login to the rescue shell:

```bash
insmod /mnt/crc32_generic.ko
insmod /mnt/f2fs.ko

umount /mnt
mount /dev/mmcblk0p7 /mnt
mount /dev/sda1 /mnt/data

chroot /mnt bash
mount -t proc proc /proc
PS1='${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
...
exit
umount /mnt/proc
umount /mnt/data
umount /mnt
```

In order to perform the filesystem check on the two filesystems from the rescue shell:

```bash
# Do a fsck of the root partition
e2fsck -f /dev/mmcblk0p7

# Do a fsck of the /data partition 
mount /dev/mmcblk0p7 /mnt
/mnt/usr/lib/arm-linux-gnueabihf/ld-2.31.so --library-path /mnt/usr/lib/arm-linux-gnueabihf /mnt/sbin/fsck.f2fs --dry-run /dev/sda1
umount /mnt
```