procount/pinn #398: PINN can't backup Raspberry Pi OS

realOgenki, 2020-07-03:
When I tried to backup my OS with PINN a splash windows show me this message:

[image]

procount, 2020-07-03:
OOPS, small problem since they renamed their OS.
Please reboot PINN **TWICE** whilst connected to the internet, and then it should be ok.

lurch, 2020-07-03:
LOL, I didn't realise PINN suffered from the "turn it off and on again" meme too :rofl:

procount, 2020-07-03:
:smiley: You can't get an easier support solution than that, though, can you!

realOgenki, 2020-07-03:
> OOPS, small problem since they renamed their OS.
> Please reboot PINN **TWICE** whilst connected to the internet, and then it should be ok.

"Microsoft solution" works, but I found another issue:

[image]

Where is log file and how I can solve it?

procount, 2020-07-03:
see https://github.com/procount/pinn/wiki/Troubleshooting for info.

realOgenki, 2020-07-03:
> see https://github.com/procount/pinn/wiki/Troubleshooting for info.

Thanks for your advice and help but I connect to PINN through VNC and I don't know the way to see and use the recovery shell.
Any idea?
Regards

procount, 2020-07-03:
Go to maintenance, select PINN, edit config, Edit Options and set SSH and reboot.
Then you can log in with ssh (root/raspberry) and do the same things as if you were in the recovery shell.
But the reboot will use your logs, so you'll have to try backing up again to repeat the error.

realOgenki, 2020-07-03:
> Go to maintenance, select PINN, edit config, Edit Options and set SSH and reboot.
> Then you can log in with ssh (root/raspberry) and do the same things as if you were in the recovery shell.
> But the reboot will use your logs, so you'll have to try backing up again to repeat the error.

Great! I had not seen ssh option on edit option before. I have did it with a direct connection and I show you the last lines:

```
"sh -c "cd /tmp/src; tar -c . | pigz > /tmp/media/sda1/os/Raspberry_Pi_OS_(32-bit)#20200703-160533/boot.tar.gz"" 
sh: syntax error: unexpected "("
"Error writing Raspberry Pi OS (32-bit): Disk full?" 
```

My USB is not full, maybe it could be a issue with the parethesis...

procount, 2020-07-03:
Yes, looks like an issue with parentheses. I thought I had fixed that, but looks like I missed one.
Bear with me, it might take a little while for me to fix it.

realOgenki, 2020-07-03:
> Yes, looks like an issue with parentheses. I thought I had fixed that, but looks like I missed one.
> Bear with me, it might take a little while for me to fix it.

Don't worry, you are the developer and I have a lot of patience ;-)
Thank you.

realOgenki, 2020-07-16:
Any news about this issue? I hope you didn't forget it.. ;-)

procount, 2020-07-16:
Sorry, I'm struggling with resources at the moment (time & laptop). 
I hope to have it done soon.

realOgenki, 2020-08-14:
Hi, procount.
Recently I tried again to backup my Raspberry Pi OS because I can't an easy way to resize my SD partitions, but PINN show me the same issue. If I can edit OS name when I backup the system, I could delete the parenthesis... Some progress?
Regards.

procount, 2020-08-14:
I'm really sorry I've not been able to look at this for you yet. I have some really pressing work to do at the moment and some issues to resolve, but I will get back to this as soon as those are out of the way.

procount, 2020-08-24:
Fixed in v3.3.4.3 Just released.

closed 2020-08-24 by procount

realOgenki, 2020-08-27:
> Fixed in v3.3.4.3 Just released.

Checked! Thank you very much, procount.
I have a question: could I restore the backup on a partition of different size than original?

procount, 2020-08-27:
Yes, provided it is big enough to store all the content. Plus you need some additional working spare disk capacity (for temp files, apt upgrade etc.) otherwise you may find you haven't got enough disk space to boot it. The boot partition will be the same size, but the main root partition can vary

realOgenki, 2020-08-27:
Thank you again, procount.
Could you tell me something more about this additional working spare disk capacity?

procount, 2020-08-27:
Not really. It depends on your OS and what you will do with it.
All I'm saying is that if you use a general purpose OS like Raspbian /RasPiOS and your backup is 8GB in size, don't put it on a partition that is only 8.1GB, otherwise you will soon run out of disk space and it may stop working.

realOgenki, 2020-08-27:
I understand, procount. Thank you for your help and comments. My best wishes for this fantastic project.

Retro-Devils, 2022-06-17:
So im having a similar issue on newest PINN. The backup button wont even let me select it at all. Very confusing lol. 

Hoping to get some  help here. Tried reboot twice lol

cRoCx, 2023-06-17:
> > OOPS, small problem since they renamed their OS.
> > Please reboot PINN **TWICE** whilst connected to the internet, and then it should be ok.
> 
> "Microsoft solution" works, but I found another issue:
> 
> [image]
> 
> Where is log file and how I can solve it?

Sorry to open this up again, but I have just encountered the same error message (see screenshot) and after some debugging I think I found the cause:
1. Debug log says it can't write more data -> Looked up free space on backup device -> Still >100 GB available for a backup of only 17 GB. But ... it could only write 4 GB -> This is the file size limit on FAT32.
2. PINN does recognise other file systems which allow for larger files and recognises them i.e. at the start screen, but it just hides them when you go to the other menu where you do backups. The list emties out there. Happened to me with ext4 and exFAT.
Conclusion: You can only backup to FAT32 with the 4GB file size limitation. All other file systems seem unavailable to the backup functionality.

cRoCx, 2023-06-21:
Oh yes, I think I forgot about that. My bad. :-( After the FAT32 backup finished with errors (>4GB), I think I started fiddling around, trying to find a solution for this, formatting and reformatting SD cards with other file system types, totally forgetting about about this `/os` folder, which must be added manually each time. ... and I also overlooked the sentence with this hint in the readme.

Could you add the appropriate formatting to the word "/os" in the backup section, like this.

Before: The backup function allows an installed OS to be backed up to a USB drive. The USB drive must first be formatted as FAT32, ext4 or ntfs and have a /os folder on it to store the backups in, otherwise it will not be recognised.

After: The backup function allows an installed OS to be backed up to a USB drive. The USB drive must first be formatted as FAT32, ext4 or ntfs and have a `/os` folder on it to store the backups in, otherwise it will not be recognised.

In other sections of the readme the `/os` stands out like this, but not in the _backup_ section. I could also create a pull request for this, if you want.

Sorry that I wasted a little of your time, since it was clearly my mistake somehow.

procount, 2023-06-21:
No worries, I updated the backup section.