procount/pinn #400: Buildroot 2020

AaronDewes, 2020-07-03:
After the old buildroot PR was I deleted, I wanted to continue it and create another PR, but it would use QT5. Because I'm no C/C++ developer, I can't port PINN itself to QT5. I did some research and found this:

- QWS was deleted in QT5, QPA should be used instead
- A waylandcompositor would be needed for this. AFAIK weston doesn't support switching the keyboard layout while it's running, so a custom compositor should be used instead. This can be a separate app or added to PINN with qwaylandcompositor.

I already created a QT5 fork of Arora with some small optimizations.

I'll submit a buildroot only PR if you want, and give you push access (everyone else can open a PR to my repo).

@procount Are you interested in this?

procount, 2020-07-03:
Were you previously known as RaspberryPiFan?

AaronDewes, 2020-07-03:
No, but I was in his discord server about a different project, so I knew him.

procount, 2020-07-03:
There are a number of areas in PINN that could be brought more up to date, such as the build environment, buildroot version and qt framework. But if you want a PR on any of these to be accepted, they need to be done in an incremental manner so I can understand the changes (as I will need to maintain them), following a described roadmap, and each PR needs to be fully tested and working.
Trying to advance any one of them could require extensive changes requiring a long while to do. Meanwhile this cannot hold up the main continuous development of PINN. So it probably requires plenty of merging of the main branch into your development branch. Converting from QT4 to QT5 will be a bit hard if you're not a C/C++ developer.
What I'd really like is to have the build environment advanced to a more recent distro versison so it can be built on Ubuntu 20 or the x86 version of Debian with Raspberry Pi Desktop. PR#249 was a good attempt, but for some reason that I can't remember at the moment, I had to revert it. It would be good if this could be continued as I think it can be done standalone. I think changing qt and buildroot will be inter-related so that would be a BIG change and would almost require a complete rewrite.

AaronDewes, 2020-07-04:
I'll try to switch to buildroot 2018.11, which should at least fix some issues with buildroot, because that's the latest version which supports QT4. If that fully works, I would create a separate branch in my fork, which contains a later version of buildroot, so interested C++ developers can use that to have the latest QT features. I'll submit a PR for buildroot 2018.11 later.

procount, 2020-08-24:
As I responded elsewhere, switching to a newer buildroot first with the existing Qt4 might be the best solution seeing as Qt5 has dropped QWS and it now needs much more resources (X11 or Wayland compositor etc.)

ShadowEO, 2020-09-11:
PR #249 was mine, I'm a novice when it comes to buildroot, so even though I thought I fixed it up on my system, there were reports of ca-certs not installing properly or at all. I gave it a good try, but it appears I ultimately failed. ^_^;;