procount/pinn #402: Error creating partition table

mll0, 2020-07-18:
(Quite noobish regarding Linux/RPI here)
So I ask PIMM to install a hefty number of OSes on an external 1TB SSD, and whatever I do, I always get the dreaded "Error creating partition table".
[image]
[image]
[image]
[image]
At first I thought the PIMM SD card's size was the culprit (128 MB only), then I switched to 2 MB and finally to the minimum recommended size (8 MB), to no avail.
I tried to take a loog to the debug file. I guess the line `/sbin/sfdisk: Warning: given size (16046080) exceeds max allowable size (15419392)` is where things start to go wrong, but I don't see where it comes form since my SSD is pretty much above 15419392 bytes.

Here are the dmesg dump and the debug file:
[dmesg (2).txt](https://github.com/procount/pinn/files/4942415/dmesg.2.txt)
[debug (2).txt](https://github.com/procount/pinn/files/4942416/debug.2.txt)

procount, 2020-07-18:
I wonder if @maxnet has any idea?
A 1TB drive should be fine as MBR goes up to 2TB. 
29 partitions should be ok as it is less than the maximum of 63.
Is it because one of the partitions has been made greater than its fs can support?

maxnet, 2020-07-19:
>I wonder if @maxnet has any idea?
>A 1TB drive should be fine as MBR goes up to 2TB.

Well, judging from the debug output it does not have a problem partitioning the hard drive, but when creating the 28 FAT32 partitions on the SD card...

Think you should be able to use USB boot instead of using the SD card.

maxnet, 2020-07-19:
Oh, and sizes are not bytes but sectors (512 bytes).
Apparently one of the OSes specified it wants a large FAT32 partition.

mll0, 2020-07-19:
Thanks for your quick reply guys.

> A 1TB drive should be fine as MBR goes up to 2TB.

Actually, what I didn't say is that I made many different attempts, sometimes with only a couple of OSes, and it always failed, many times with "Error creating partition table", sometimes with other flavours of errors. I actually had given up on PIMM (spent hours of failed attemps), but since you replied so helpfully, I may retry and fetch the debug file.

> (And make sure the Pi 4 can boot from USB first, by flashing it with the right eeprom boot firmware/settings).

Sounds both tempting and scary. I wanted to try that, and I stalled when I saw this was in beta. Do you think that's a rather stable beta? 

> Apparently one of the OSes specified it wants a large FAT32 partition.

Could it be the reason of my various failures, if the given OS is always in y short-list?
Would your suggestion of booting from the HD prevent this anyway?

mll0, 2020-07-19:
(While I'm here, a quick OT question: is there any plan to include [piCorePlayer](https://www.picoreplayer.org/) in the list of available OSes? I saw one can customise it, but I fear this is out of my skills.)

procount, 2020-07-19:
@maxnet - thanks for replying with your thoughts.

In no particular order...

piCorePlayer is available in PINN, just not for the Pi4 as yet. I will need to look if that has been upgraded to support the Pi4 and will update it in due course.

Whilst PINN (note it is with 'N's not 'M's) does now support USB-Boot with Pi4, not all of the OSes it installs have been upgraded to support this mode, and PINN does not provide any services to upgrade them itself as yet. So it's probably best to keep booting off the SD card for now, unless you know for sure that the OSes you are installing support it. (I may have marked them as non-usb bootabel anyway)

The 2nd OS seems to want to install a 3GB FAT32 boot partition. That seems a bit large. I must improve the debugging to include the name of the OSes and the order they are installed. If the list at the start is indicative, it could be LibreELEC..? but that sounds improbable.

I'm a bit stuck at debugging this at the moment, and it may be a while before I can free up a 1TB drive to test this out. but I will endeavour to look into it in more detail. Meanwhile, if you have time, I would suggest just trying to install one OS at a time, preferably starting with the Raspios versions, to see if it is a particular OS that causes the failure.

mll0, 2020-07-19:
Thanks procount.
I tried to install only 2 OSes : Raspberry_Pi_OS_(32-bit) and DietPi. I got `OSes were installed with errors` . Attached log seems to show many write failures, so maybe the SSD is faulty. I just did a surface test, and it went smoothly, however it might be that under the stress of the install it stats bein unresponsive or something like that.
[log.txt](https://github.com/procount/pinn/files/4943441/log.txt)

Now I'll go and try installing them one at a time.

mll0, 2020-07-19:
> Now I'll go and try installing them one at a time.

Done that, and it didn't succeed: `/sbin/sfdisk: Error closing /dev/sda: Input/output error`, as per [log 1 OS (Rpi full).txt](https://github.com/procount/pinn/files/4943473/log.1.OS.Rpi.full.txt) (btw, an install on PI_NN_'s ;) SD card did succeed). I guess something's wrong in my SSD or its enclosure. I might try again with a spare 

> piCorePlayer is available in PINN, just not for the Pi4 as yet. I will need to look if that has been upgraded to support the Pi4 and will update it in due course.

Great news! FYI, It very well handled my 4B.

procount, 2020-07-19:
It certainly looks like an SSD problem now as it is failing to install the OS, whereas initially it looked more like a partition problem, maybe with a specific OS.

mll0, 2020-07-19:
I could lay my hands on a 500 GB mechanical HD. I could install Raspberry_Pi_OS_(32-bit) in a breeze. Now I restarted PINN's recovery, and I'm installing a slew of other distros on it. Will keep you updated if it fails, but it's going smoothly so far.
Sorry for taking your time dealing with a faulty SSD. :/

closed 2020-07-19 by mll0

procount, 2020-07-19:
No problem. There is a sticky post on the forum about SSD drives. Some need some quirks to disable uas mode to get them to work properly. I suggest you check it out before discarding it as faulty.

mll0, 2020-07-19:
[Found it](https://www.raspberrypi.org/forums/viewtopic.php?t=245931). Will do. Thanks.

lurch, 2020-07-19:
I once had a USB SSD enclosure (a few years ago) that only worked on my Linux laptop (i.e. non-Raspberry-Pi) when connected via a USB2 cable, and would fail with random errors when connected via a USB3 cable (this was in the days before USB-C). So plugging your SSD into the Pi4's USB2 ports instead of it's USB3 ports might be something else to try, if the UAS quirks don't help? :shrug: 

> The 2nd OS seems to want to install a 3GB FAT32 boot partition. That seems a bit large.

I guess if one of the OSes has `want_maximised: true` for a fat32 partition, and you're installing onto a large HDD, then that might end up trying to creating a bigger partition than fat32 allows?

procount, 2020-07-19:
>I guess if one of the OSes has `want_maximised: true` for a fat32 partition, and you're installing onto a large HDD, then that might end up trying to creating a bigger partition than fat32 allows?

That"s the only thing I can think of. I haven't checked any OSes, but maybe one that runs off a ramdisk, therefore just requiring a boot partition would be one to look for.

mll0, 2020-07-19:
What's weird is that on my 500 GB HD, I asked PINN to install nearly as many OSes as on my 1 TB SSD, and I didn't ged the error described on my OP.

maxnet, 2020-07-19:
> I guess if one of the OSes has `want_maximised: true` for a fat32 partition, and you're installing onto a large HDD, then that might end up trying to creating a bigger partition than fat32 allows?

`sfdisk` only knows partition tables, and has no clue about file systems, and what maximum size they support.
So that is not the problem.

Probably just a bug in which when `want_maximised: true` is set on a FAT32 partition, it will size the FAT32 partition proportional to the disk size of the USB root device, instead of the boot SD card.

mll0, 2020-07-19:
> No problem. There is a sticky post on the forum about SSD drives. Some need some quirks to disable uas mode to get them to work properly. I suggest you check it out before discarding it as faulty.

So I've done that, and launched the install of many OSes. Yet I still get `Error creating partition table`: 
[log many OSes+quirks.txt](https://github.com/procount/pinn/files/4944161/log.many.OSes%2Bquirks.txt).
Now I seem to remember that SSD (a Crucial MX500) caused me troubles a while ago. I ran Crucial's full diagnosis on it, and it did not detect any problem. Weird, but still an SSD problem I think.