procount/pinn #403: not having partition customizations in custom repo causes install to "fail" (reboots into OS fine)

Ae3NerdGod, 2020-07-22:
so, i have been working on my own repo (you can look at it [here](http://bowtie-ent.com/pinng) ) and when i try to install an OS i made out of an image; after it downloads and extracts both partitions, runs partition_setup.sh, according to the log it then looks for customizations on the server (at least thats what /tmp/debug seems to indicate)  when it does that it returns "404 file not found" for the imaginary .tar its looking for, then it does it again for the tar.xz of the nonexistent partition customizations, then stops the installs ( no os after a custom one will install and "fail" )  and i get a dialogue box in PINN saying "unsupported archive format FILE EXTENSION". which makes no sense as it definitely is .xz.  if i close that, pinn manager windows flashes, then i get the same error again, when i close that i get the walpaper and nothing else, i have to unplug the pi.

afterwards, the pi boots, and the menu shows any official OS i install, then the one custom one i selected. it boots, it installed fine, but it detects a fail from xz and stops the install according to /tmp/debug

if needed ill provide the log, but its easily reproducible: 

to reproduce:
use my repo and try to install manjaro KDE. PS it has dummy .tar.xz files now, and it still "fails"
manjaro XFCE has no dummy archives and fails the same way

very frustrating, seeing as all the raspi OS from my repo install fine.

again my repo is at http://bowtie-ent.com/pinng/os_liost.json

would love some help...

procount, 2020-07-22:
It looks like you are amassing an impressive set of OSes.
So Raspios is a copy of the original and I assume all others are your own custom creations, but none of your custom OSes install properly? That suggests there is something wrong with your OS conversion. 
I will have a go at installing them to see if I can discover what is wrong.

Ae3NerdGod, 2020-07-22:
The only custom one that installs properly is Raspi OS cinnamon, and now that you mention it I've only tried the two manjaro's, other than that, I'll give Ubuntu and Kali a try and come back with logs in a couple hours, sorry for not having them already lol

Ae3NerdGod, 2020-07-22:
Also to be clear, moth manjaro's INSTALL and BOOT 

It's the install process that's throwing out an error that seems to be false as far as I can tell. 

Would using tar instead of bsdtar be a possible culprit? I assumed that wasn't the problem Because it actually does extract and boot properly, but I can test it with bsdtar.

Also when I get back home I can put my conversion script in the repo if necessary

Ae3NerdGod, 2020-07-23:
to further clarify, i AM NOT trying to use the partition customization feature, the extra tars in the manjaro KDE folder were dummies in an attempt to troubleshoot. trying some other distros now, will link logs shortly

Ae3NerdGod, 2020-07-23:
i did a few other distros, log time

script here: http://bowtie-ent.com/pinng/osmaker.sh

manjaro kde with dummy customization: extracted , booted but install gave same error as described.
http://bowtie-ent.com/pinng/debug-w-dummy-archives

ubuntu 18.04.4 server: ( did not boot, spelling error in json. retrying)
http://bowtie-ent.com/pinng/debug-ubuntu-server-18.04.4-arm64

kali linux: ( did not boot, extracted fine, most likely me re-using raspi os partition_setup.sh )
http://bowtie-ent.com/pinng/debug-w-dummy-archives

PS i usually include a raspi OS with the install im testing so i can try and diagnose on the pi, i just discovered if i install them alone, none of them trigger the boot menu and therefore im, unable to boot them except raspi-os cinnamon (which the tar was made the same way as before except i used tar -J for xzip in my script at first and switched to gzip in troubleshooting the error from manjaro

Ae3NerdGod, 2020-07-23:
update:
tried with xfce with bsdtar
heres the log: 
http://bowtie-ent.com/pinng/debug_manj_xfce_bsdtar

and i figured out why they werent showing up on the boot menu, i forgot the bootable=true tag in the json in most of themim currently not focused on them booting, just installing properly.

procount, 2020-07-23:
I installed your Manjaro XFCE distro and saw the issue. As you said, it doesn't prevent them from being installed, it's just a notification that is being shown as an error. Very odd. I have an older manjaro distro that installs without the error messages. Did you compare your meta-files with that one? As you are creating many distros, a useful tip is to validate any JSON files you create with jsonlint.com before deploying.

I will struggle to debug this further at the moment, but I definitely will as soon as I am able to. Fortunately I think it is just a minor annoyance. 
If you are not customising any partitions, you shouldn't need to create any dummy customisation files. The associated error messages are just notifications that they cannot be found. I probably need to adjust them.

Ae3NerdGod, 2020-07-23:
glad to see its not just me, only have one pi to test atm. VALIDATION! lol

and well the main issue is, that bug causes any OS in the "queue" thats supposed to be installed AFTER it, NOT to install, so personally id qualify it as a medium bug/annoyance at least. take your time fixing it though, no rush.
thanks for the tip on jsonlint.com, ill use that.

my next troubleshooting plan was to rename the tars and drop them in a raspi OS folder that i know works, to confirm if its metadata related which it seems to be, i just cant figure out why :/ ive had the feeling its something small and obvious since the first error.

will report back with any update in the mean time.

Ae3NerdGod, 2020-07-24:
MAJOR UPDATE
TLDR:
---ITS THE VERSION NUMBERS---

http://bowtie-ent.com/pinng/testing/
http://bowtie-ent.com/pinng/os_list_testing.json

ok so, under that repo there are 2 categories: bowtie-working and bowtie-testing
under working youll see a copy of raspi lite and NOT raspi lite. ( and manjaro versiontest-- we'll get to that)

i started by making a copy of the raspios lite folder and renamed it [not_raspios_lite_armhf](http://bowtie-ent.com/pinng/testing/not_raspios_lite_armh)f/  and put the manjaro TAR files in there, renamed them to raspi OS partition names
i then changed the icon, added "NOT" to the osname and description and copied the repo entry with the same changes

NO ERROR 
" AHA! so IT IS MY METADATA " i say in my head (spoiler alert: nope)

so copied the [not_raspios_lite_armhf](http://bowtie-ent.com/pinng/testing/not_raspios_lite_armh)f/  folder and renamed it [Manjaro-ARM-xfce-rpi4-20.06](http://bowtie-ent.com/pinng/testing/Manjaro-ARM-xfce-rpi4-20.06/) and CAREFULLY only edited the icon, osname, description,  in the repo and os.json. ( i left the partition names the same)

ERROR ..... AGAIN 
so i look at the difference between the only thing i really changed, the folder name. whats different? hyphens? xz trying to read stdin? dots? bsdtar thinks its creating not extracting? 

so i make two copies of [not_raspios_lite_armhf](http://bowtie-ent.com/pinng/testing/not_raspios_lite_armh)f/ with hyphens and dots: [not-raspios-lite-armhf](http://bowtie-ent.com/pinng/testing/not-raspios-lite-armh)f/ and [not.raspios.lite.armhf](http://bowtie-ent.com/pinng/testing/not.raspios.lite.armh)f/ with copied repo entries

THEY. BOTH. WORK.
WHATAMIMISSINGWTF!
so then i take a break. i go eat, i come back. 
and i say F it.

i, with intense disdain and no hope whatsoever, copied the NOT WORKING manjaro entry,
to [Manjaro-ARM-xfce-rpi4-versiontest]( http://bowtie-ent.com/pinng/testing/Manjaro-ARM-xfce-rpi4-versiontest/)
i begrudgingly edit OS jason and replace the version number.
i mumble under my breath very hateful things as i make yet another repo entry
i power-cycle the pi, with no adjectives or emotion because this thing was expensive (4b 8gb model)
i select the same two OS i have been, working raspi lite and the current test, and i click install, and i go play some amtruck with friends in a discord.

<some time passes>
i pull up into a gas station and look at my other input while its fueling

..
what

WHAT

i must not have selected the right os, 
i try again. 

IT. FREAKING. WORKS.  http://bowtie-ent.com/pinng/testing/Manjaro-ARM-xfce-rpi4-versiontest

youll find it under the BowTie-Working tab 👍 of the testing repo

i dont get it. so aparantly xz? is reading ~.06/.tar.xz as the extension BUT NOT ~.armhf.tar.xz? or perhaps PINN code isnt parsing it correctly? MAKES ABSOLUTELY NO SENSE TO ME, idk but i have found a working solution. hope this helps 

sorry im so talkative lol

Ae3NerdGod, 2020-07-24:
PS im going to leave 
http://bowtie-ent.com/pinng/testing/
http://bowtie-ent.com/pinng/os_list_testing.json

up for a bit but ill probably end up deleting it in a week or month or so for housekeeping, sorry for my horrible upload bandwidth, blame spectrum for giving me half of what i pay for

lurch, 2020-07-24:
P.S. I dunno what difference it makes (and this is probably unrelated to your problem here), but I _think_ the `nominal_size` in os_list.json is supposed to be the total of the partition sizes, i.e. you'd want to use 10496 rather than 10240 (as you have a 256 MB boot and 10240 MB root).

procount, 2020-07-24:
It sounds like PINN assumes there should be no dots in the filename apart from the extensions.  I can't remember any other OS having dots in the root part of the filename.

You might want to check the supported models. Is manjaro suitable for all Pi models from the Pizero to the Pi4?
What about USB boot/root support?
Also ensure the partition names are of the correct length. i think 11 chars is the max for FAT, although I think PINN will adjust them as necessary, so they may not be totally what you expect.

Ae3NerdGod, 2020-07-24:
@lurch thank you, you're right I wasn't talking the boot partitions int account, or more than 2 partitions
Into account with that. 

And I left those models in for metadata troubleshooting and because the os generator script has no way of knowing what models they support, all of that will be checked after generation and edited manually, I just wanted to get a universal template going for future images, make the hardest part easier.

PS replacing the dots in the filename with hyphens using sed is a working solution for this issue atm.

Ae3NerdGod, 2020-07-26:
> It sounds like PINN assumes there should be no dots in the filename apart from the extensions. I can't remember any other OS having dots in the root part of the filename.
>
> You might want to check [...]

@procount it appears it is not related to JUST filenames: could be both or just metadata:

http://bowtie-ent.com/pinng/os_list_ub.json

the first entry has a period in the metadata but NOT in the path/foldername and still "fails"

Ae3NerdGod, 2020-07-26:
the + was causing problems too, although its a different error messege: it says "os completed with errors, see debug log for details" and there doesnt seem to be an error at all in the log

on the plus side, thi has finally made me learn how sed works. very useful

[image]<div><a href="https://imgflip.com/memegenerator">from Imgflip Meme Generator</a></div>

procount, 2020-07-28:
I've got a small window to do some debugging on this, so please keep your testing OSes available for a while, so I can use them for verification. Thanks.

procount, 2020-07-28:
It appears some of the PNG files are missing so I can't test them. Trying os_list_ub.json instead.

procount, 2020-07-28:
OK I tracked it down. The answer is just not to put any additional dots in your os_name. Or '+'s.
Also, please format your release-dates as YYYY-MM-DD, as these are compared to decide if a new OS version is available.

Ae3NerdGod, 2020-07-29:
Awesome. 

Also I have fixed the dates for finished imgs, that was just for me to keep track during testing. Btw, I have two finished vanilla Manjaro 20.06 imgs (correct dates, kernels, versions etc. Shasums will come soon.) currently on my public repo. Tested working with kernel updates and all if you want to include them with pinn:

http://raw.githubusercontent.com/Ae3NerdGod/PINNg/master/os_list_public.json

 Kali Linux imgs are next, but the roots are 2+gb so hosting them is the main issue. If you have somewhere pinn-compatible to host them, they'll be in the public repo probably after a day or so of testing. then I'm gonna try and tackle the Das U-Boot

Ae3NerdGod, 2020-07-29:
PS id recommend putting the + and . thing as a warning in the part of the readme about making custom OS's if possible

procount, 2020-07-29:
I have some old Kali images already, but last time I tried to convert them I had a few problems. You may find my post on a Kali forum somewhere.

I can't really see the point of U-boot on a RPi so I normally just remove it. If you modify it to work, you may have to do it for all 3 architecture variants.

Ae3NerdGod, 2020-07-30:
kali wont boot for me, pinn or not. i believe its eeprom related in my case, and im happy with my eeprom version thankyouverymuch

### IN OTHER EXCITING NEWS,  

**the "its not a bug its a feature" reward goes to me this year.** i turned the lack of sanitation in the tarballs section into something useful:
```
            "tarballs": [ 
                "http://bowtie-ent.com/pinng/os/TwisterOSv1_4_1/boot.tar.xz", 
                "http://bowtie-ent.com/pinng/os/TwisterOSv1_4_1/rootfs_join | sh && cat /tmp/join/rootfs.tar.xz" 
            ] 
```
which turns into 
```
Executing: "sh -o pipefail -c "wget  --retry-connrefused  --read-timeout=120 --tries=inf --no-verbose -O- http://bowtie-ent.com/pinng/os/TwisterOSv1_4_1/rootfs_join | sh && cat /tmp/join/rootfs.tar.xz | tee /tmp/fifo | xz -dc | bsdtar -xf - -C /mnt2  | sha512sum /tmp/fifo > /tmp/sha1.out.txt""
```

and rootfs_join:
```
#!/bin/sh
mkdir -p /tmp/join

wget  --retry-connrefused  --read-timeout=120 --tries=inf --no-verbose -O/tmp/join/rootfs.tar.xz http://bowtie-ent.com/pinng/os/TwisterOSv1_4_1/rootfs_part1

wget  --retry-connrefused  --read-timeout=120 --tries=inf --no-verbose -O- http://bowtie-ent.com/pinng/os/TwisterOSv1_4_1/rootfs_part2 >> /tmp/join/rootfs.tar.xz

```
now i can just split the rootfs into 2gb using dd
AND THE PROGRESS BAR WORKS
_I IZ A HACKER NOW_ 

im gonna experiment with echo and stdout to see if i can download and extract the same time but this is definitely proof of concept

closed 2020-07-30 by Ae3NerdGod

lurch, 2020-07-30:
> i turned the lack of sanitation in the tarballs section into something useful:

OMG, that's hideous :vomiting_face:  Congratulations :rofl: 
:nauseated_face:

Ae3NerdGod, 2020-07-30:
@lurch why thank you sir. im very proud

even got it to work with stdout so it downloads and extracts at the same time:
```
 "tarballs": [ 
    "https://github.com/Ae3NerdGod/PINNg/releases/download/twist-os-1.4.1/boot.tar.xz", 
    "http://bowtie-ent.com/pinng/os/TwisterOSv1_4_1/rootfs_join_v2 > /tmp/rootfs.tar.xz && sh /tmp/rootfs.tar.xz" 
  ] 
```
and rootfs_join_v2:

```
#!/bin/sh

wget  --retry-connrefused -q --read-timeout=120 --tries=inf --no-verbose -O- http://github.com/Ae3NerdGod/PINNg/releases/download/twist-os-1.4.1/rootfs_part1 |cat && wget  --retry-connrefused -q --read-timeout=120 --tries=inf --no-verbose -O- http://github.com/Ae3NerdGod/PINNg/releases/download/twist-os-1.4.1/rootfs_part2 |cat 

```

procount, 2020-07-30:
Will be AFK for a while. Will check on return.

lurch, 2020-07-30:
If I was in @procount 's shoes (which I'm not!) I'd use this as an excuse to add better JSON validation :wink:  Reminds me very much of remote-code-exploits.

@Ae3NerdGod Is there actually any practical reason for doing this, or are you just pushing the bounds to see what's (currently) possible?

procount, 2020-07-30:
@lurch - already thinking on those lines! If you have any mitigation ideas, please share. And probably needs a backport into NOOBS too. Time for some googling.

Ae3NerdGod, 2020-07-30:
@lurch GitHub releases are limited to 2gb a file, sourceforge limited to 1gb, couldn't compress twisted OS or kali down small enough to host them.

Also, the json is valid, I wrote the line in jsonlint.com.

It's the using of the variable in the SH line that needs sanitation I think, but I hope you don't fix it

Ae3NerdGod, 2020-07-30:
@procount if you're going to fix it, could you add split archive support as well when you do? Imgs are only getting bigger

procount, 2020-07-30:
I have a 2.4gb file hosted on sourceforge, so I don't see any 1gb limit.

Ae3NerdGod, 2020-07-30:
@procount but I'd argue for keeping it, as a counter to the remote code execution argument; Is that not what pinn does with partition-setup.sh already?

If someone wanted to make a malicious repo they already could.. I say leave it the way it is fwiw.

I know it's ugly, but I still love it the same.

lurch, 2020-07-30:
@procount 
> If you have any mitigation ideas, please share

It looks like properly-quoting the wget URL _might_ be all that's needed? On my Linux laptop I get:
```
$ wget "http://bowtie-ent.com/pinng/os/TwisterOSv1_4_1/rootfs_join | sh && cat /tmp/join/rootfs.tar.xz"
--2020-07-30 23:26:24--  http://bowtie-ent.com/pinng/os/TwisterOSv1_4_1/rootfs_join%20%7C%20sh%20&&%20cat%20/tmp/join/rootfs.tar.xz
Resolving bowtie-ent.com (bowtie-ent.com)... 66.168.208.138
Connecting to bowtie-ent.com (bowtie-ent.com)|66.168.208.138|:80... connected.
HTTP request sent, awaiting response... 403 Forbidden
2020-07-30 23:26:25 ERROR 403: Forbidden.
```
(but obviously you'll need to test that the `wget` in PINN works the same way)

> And probably needs a backport into NOOBS too.

If you submit a PR, the NOOBS maintainer will hopefully merge it before the next NOOBS version.

---
@Ae3NerdGod 
> as a counter to the remote code execution argument; Is that not what pinn does with partition-setup.sh already?

Haha, that's true :smiley: 

> I know it's ugly, but I still love it the same.

The other problem with this "hack" is that it'll be very difficult for people to create an "offline install" of PINN with this OS included.

> Imgs are only getting bigger

When NOOBS (the parent project of PINN) was first released, the intention was that the project creating & hosting the IMG version of the distro, would also create & host the NOOBS version of the distro. So IMHO the "IMGs being too big" is a problem for the distro-creators to solve, not for PINN to solve.
(as it turns out, easy-to-use IMG flashing tools like Etcher and Raspberry Pi Imager have made NOOBS much less important than it used to be, so not many distro-creators also create NOOBS versions of their distros)

procount, 2020-07-30:
Admittedly partition_setup.sh can already be hashed to guard against modification, however RCE is not something I'd like to encourage.

Ae3NerdGod, 2020-07-30:
@lurch 

You'd end up doubling the quotes in the json, and I could just pipe after the first of the double quote I think

Ae3NerdGod, 2020-07-30:
@procount malicious partition_setup.sh can be based as well.

The repo itself would have to be messed with either way, so as long as that's securely on a server, and no one with malicious intent could modify it, then it's up to whether or not you trust the repo (same as it is now)

Although I definitely understand where you're coming from, RCE is bad practice, but it's not making it any less secure in this situation. Nothing you do will stop someone from making a malicious repo. Besides the existing data ALWAYS gets wiped before either tarball is downloaded ( or executed)

I can't think of a way to use this other than my implementation.

Ae3NerdGod, 2020-07-30:
The offline image argument doesn't hold weight to me. The repo isn't involved (which is where the rce takes place) and you can just not split the archive, or join it when copying to the SD card

procount, 2020-07-30:
> You'd end up doubling the quotes in the json, and I could just pipe after the first of the double quote I think

No, we mean to properly quote in the sourcecode, not the json file.

> I can't think of a way to use this other than my implementation.

But I'm sure someone else would. :wink:

> The repo itself would have to be messed with either way, so as long as that's securely on a server, and no one with malicious intent could modify it, then it's up to whether or not you trust the repo (same as it is now)

My saving grace.

Ae3NerdGod, 2020-07-30:
> 
> No, we mean to properly quote in the sourcecode, not the json file.
> 

You could still break it with a pipe couldnt you?

> 
> But I'm sure someone else would. :wink:
> 
They'd STILL have to setup their own repo, and anything they could do would be easier done in partition setup.

The ONLY think I'm hacking is a progress bar lol

> > The repo itself would have to be messed with either way, so as long as that's securely on a server, and no one with malicious intent could modify it, then it's up to whether or not you trust the repo (same as it is now)
> 
> My saving grace.
> 
> 

 Not sure what you mean by that lol

Ae3NerdGod, 2020-07-31:
if you quote it in the source code i could just do something like this:

 "tarballs": [ 
    "https://github.com/Ae3NerdGod/PINNg/releases/download/twist-os-1.4.1/boot.tar.xz", 
    "http://bowtie-ent.com/pinng/os/TwisterOSv1_4_1/rootfs_join_v2 > /tmp/rootfs.tar.xz | echo this will end the quote \" && sh /tmp/rootfs.tar.xz" 
  ]

Ae3NerdGod, 2020-07-31:
github wont show the backslash but imagine one after "v2" and before " >

Ae3NerdGod, 2020-07-31:
i think the only way to mitigate this would be to let a script handle the downloading rather than that super clean yet somewhat exploitable one liner you got doing it right now.

Ae3NerdGod, 2020-07-31:
i mean, lets be honest, **nothing about the pi is secure**. there is direct access to the serial console, the schematics to everything are online, it would be hilariously easy to read the ram while running, and im pretty sure i could do it from GPIO if i tried hard enough, there is no partition encryption, hell, a script could easily replace PINNs kernel with a malicious one, or the repo. Or, add and automatically install an OS at next boot, from user-space in ANY operating system that allows user-space mounting. the only os thats even possible to somewhat secure the boot process in id think is ubuntu, which cant be booted that way with pinn currently, not to mention USB boot. OH, you can literally CHANGE THE WIFI from the boot partition for middle man attacks. im pretty sure i can go on, but my intent here is not to be annoying or right, i just wanna make a full case FOR keeping this hack as a feature (that i happen to have discovered)

which brings me to my next point. the name of PINN. is itself not a hack you kept because it works so well? are you not proud of it. if PINN is truely not NOOBs, then let them be the lame ones and fix it, this could introduce the next worlds leading security expert whose currently 13 into the world of exploits or some shit like that.

I SAY TO THEE; DOES THIS HACK NOT REPRESENT THE VERY NATURE OF THE GNU SPIRIT!?

you buying any of this? lol

lurch, 2020-07-31:
> which brings me to my next point. the name of PINN. is itself not a hack you kept because it works so well and was (legally) more or less safe from exploits? are you not proud of it..

Not sure what you're implying with that comment? @procount was experimenting with NOOBS add-ons and his own custom fork of [NOOBS](https://github.com/raspberrypi/noobs) back in early 2015 (I was the NOOBS maintainer at the time), and I suggested tongue-in-cheek that procount could name his fork "PINN" - Pinn Is Not Noobs (taking inspiration from the name WINE), and I guess he must have liked the suggestion :wink: 
Over time, PINN has diverged more and more from NOOBS and now contains many "advanced features" that won't ever be incorporated back into NOOBS (because NOOBS is deliberately very simple / easy-to-use).

At around the same time as procount was starting work on PINN, [Pi-Kitchen](https://pihw.wordpress.com/guides/pi-kitchen/) was also being developed, which is [now dormant](https://pihw.wordpress.com/2017/08/15/the-way-forward-for-pi-kitchen/). 
[PiBakery](https://www.pibakery.org/) was also a similar project that came along later, but that too is [now dormant](https://github.com/davidferguson/pibakery/issues/180).

IMHO procount deserves **FULL** credit for keeping [PINN](https://github.com/procount/pinn) up-to-date and working (and tested) with _all_ models of Raspberry Pi (which I know from experience can be a very time-consuming job). And also for all the work he's done in converting so many 3rd-party IMG files into PINN-compatible versions. :heart:  :clap:  :clap:

Ae3NerdGod, 2020-07-31:
Oh I wasnt implying anything, I already forked pinn, his software do what ever he wants. I was just talking out my ass half defending it half joking because it's a really nice solution imo.

PS actually GNU was the first to use that name scheme, it stands for GNUs Not Unix. Linus Torvalds described the name as a as a "hack" so he could remake Unix and be legally safe because gnu isn't Unix, it's in the name. That's all I was implying, if anything, that the name, software itself, and the exploit for the joining of spilt tarballs are great, and there's no need to fix a hack on the sole reason that it's a hack if it solves a problem

Ae3NerdGod, 2020-07-31:
Also yes, @procount  bravo on keeping this train going. Berryboot was terrible, noobs was lacking, and I knew there was something from back when I got my first rpi2 that I forgot about when I got my rpi4, when I finally rediscovered PINN I was excited as hell to make a new repo

Ae3NerdGod, 2020-08-01:
1 stallman! yes, i get them confused, bad with names, but he said that in a documentary i watched. i was just using it to be needlessly annoying anyway

2 i did not know that about the author of berryboot, and i just dont like it. the concept is great, but in practice its clunky and annoying (just in my opinion), noobs/pinn is far more usable.

3 let me be very clear; i am in no way upset at procount. none whatsoever. i just want to be able to host my OSs on a faster server than my own, this hack allows that, and so would google drive url resolving, split archives or something like that. i agree that im abusing PINN, i agree that in normal practice for non-niche software this would be a terrible idea. my only point is that FOR NOW its a better solution to larger OS's than 1) not doing anything, claiming its the distros responsibility or 2) letting PINN (and i guess noobs) die out. for a lot of people, it seems to me, not supporting newer ubuntu (which is a different problem entirely) or kali is a dealbreaker, looking at the SF. 

all that being said, im aware this is most likely a losing argument on my side, and prepared to let it go, i just wanted to get my full opinion out there fwiw, which is most likely worth not much at all 

sorry for any confusion on my part i like to talk out of my ass :) 

</embarrASSing myself on he internet>

lurch, 2020-08-01:
> i just want to be able to host my OSs on a faster server than my own, this hack allows that

I know this is going to sound harsh, but IMHO your lack of a fast server is more of your problem than it is of PINN's problem? :confused:  Have you looked into why you're only able to host 1GB files on Sourceforge, when @procount claims he has no such limitations himself?
In case you haven't already realised, for local testing & development you could set up a HTTP server (e.g. Apache) on your local network, and point `os.json` / `partitions.json` at the IP-address of that, rather than constantly uploading and downloading files from the internet :smiley: 

> not supporting newer ubuntu (which is a different problem entirely) or kali (which has a huge rootfs) is a dealbreaker, looking at the SF reviews and forum.

Both of those problems (and also your problem of a lack of a fast server?) would instantly be solved if those OSes themselves provided NOOBS / PINN compatible tarballs :slightly_frowning_face: Which is why it's a shame that NOOBS and PINN haven't become more popular...

Ae3NerdGod, 2020-08-01:
I didn't question it, when I went to upload Manjaro I hadn't used xz -9 to speed up testing and it said I couldn't upload it because it was over a gb. I redid it with -9 and it was still over, so I went to GitHub.

(btw, yes I have a local server, http://bowtie-ent.com/pinng is local for me ) 

I want to make a public one for when I'm on the road and a couple friends, he seemed interested in using Manjaro  vanilla and others if I made them. 

That responsibility shifting crap...
 dude, honestly, that's a bunch of immature bullshit, I'm sorry. But I tried to say it nicely. You're just being stuck up "fine I guess you just don't like my services" *harrumph*

You want your product to succeed, you put the work into it. 

Hell you got a guy RIGHT IN FRONT OF YOU who is entirely over enthusiastic about it and youre just like "bro chill, they're killing it, whatever not my problem if they wanna kill it"

Ae3NerdGod, 2020-08-02:
@procount  on the subject of mitigation, took a look at the code; im not fluent at all in C++, but could you not just check the `tarball` variable for pipes and ampersands?  and error out if theyre detected?
soemthing like 

```
string stopitgetsomehelp ("|");

if (tarball.find(stopitgetsomehelp) != string::npos) {
//.. error out somehow
} 
```
or something written by someone who knows c++ far more than i

Ae3NerdGod, 2020-08-03:
i figured out sourceforge, i didnt see where it says i have to use FTP to upload when i tried the first time. 

i am still advocating for split archives. when they get to 4gb they wont fit on SD. future proof.

lurch, 2020-08-03:
Perhaps https://github.com/procount/pinn/blob/master/changes.md#how-to-customise-an-os-install is a cleaner way of splitting a large tarfile up into multiple chunks? (with the difference being that each 'chunk' would need to be a complete standalone tarfile, rather than being a binary split of a parent tarfile). Although how to decide which files go in the "base" tarfile and which files go in the "extras" tarfile is a different problem entirely!

> when they get to 4gb they wont fit on SD.

Back when NOOBS was first released, 4GB was the size of a typical SD card :laughing:

Ae3NerdGod, 2020-08-03:
You say it's cleaner and describe it's messiness lol

Ae3NerdGod, 2020-08-03:
The custom OS's I was gonna share with my friends were arcade images, so they can get pretty big. (Does pinn support encrypted archives? Another reason to keep the hack in: you could unencrypt it using a pipe

lurch, 2020-08-03:
https://github.com/procount/pinn/search?q=encrypted&unscoped_q=encrypted&type=Issues

Ae3NerdGod, 2020-08-09:
@lurch seems like procount is still AFK, would you possibly know what build environment i need? tired in ub20.04 and 18.04, 20.04 gave a error related to the newer version of gcc they have, and 18.04 gives an ncurses build error? i see that you guys are working on updating the build process for newer versions, just wondering what os/environment currently works.

for reference heres the ncurses error:

```
make[2]: Entering directory '/mnt/g/dev/build/pinn/buildroot/output/build/host-ncurses-5.9/ncurses'
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/lib_gen.c -o ../obj_s/lib_gen.o
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./base/lib_insnstr.c -o ../obj_s/lib_insnstr.o
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./base/lib_instr.c -o ../obj_s/lib_instr.o
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./base/lib_isendwin.c -o ../obj_s/lib_isendwin.o
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./base/lib_leaveok.c -o ../obj_s/lib_leaveok.o
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./base/lib_mouse.c -o ../obj_s/lib_mouse.o
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./base/lib_move.c -o ../obj_s/lib_move.o
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./tty/lib_mvcur.c -o ../obj_s/lib_mvcur.o
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./base/lib_mvwin.c -o ../obj_s/lib_mvwin.o
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./base/lib_newterm.c -o ../obj_s/lib_newterm.o
/mnt/g/dev/build/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include/ncurses -O2 -I/mnt/g/dev/build/pinn/buildroot/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./base/lib_newwin.c -o ../obj_s/lib_newwin.o
_30957.c:835:21: error: expected ‘)’ before ‘int’
Makefile:795: recipe for target '../obj_s/lib_gen.o' failed
make[2]: *** [../obj_s/lib_gen.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/mnt/g/dev/build/pinn/buildroot/output/build/host-ncurses-5.9/ncurses'
Makefile:307: recipe for target '../lib/libncurses.so' failed
make[1]: *** [../lib/libncurses.so] Error 2
make[1]: Leaving directory '/mnt/g/dev/build/pinn/buildroot/output/build/host-ncurses-5.9/progs'
package/pkg-generic.mk:182: recipe for target '/mnt/g/dev/build/pinn/buildroot/output/build/host-ncurses-5.9/.stamp_built' failed
make: *** [/mnt/g/dev/build/pinn/buildroot/output/build/host-ncurses-5.9/.stamp_built] Error 2
```

Ae3NerdGod, 2020-08-09:
PS when i attempt to find the file that needs a parenthesis, it does not exist and the entire "lib" folder is empty

procount, 2020-08-09:
I use Ubuntu 14.04. I believe 16.04 will also work, but I haven't tried it for other reasons.

Ae3NerdGod, 2020-08-10:
16.04 also gave me the ncurses error, but ok I'll try good old Trusty

Ae3NerdGod, 2020-08-11:
Trusty proved it's name once again. Build completed fine.