procount/pinn #404: [Feature request] Google drive tarball download url resolving

Ae3NerdGod, 2020-07-24:
i want to make a public repo, but my upload speed is terrible. 
currently i have plans to make pinn download stubs for the tars, and then have partition-setup.sh try and resolve/extract google sharing urls, if i can get this to work, i wouldnt have a problem just doing that, but itd be nicer with the inbuilt progress bar working and such. 

just throwing the idea out there. if i make a script that works ill put the resolve logic here or link the script.

lurch, 2020-07-24:
Or you could do what @procount does, and [host the tarballs on sourceforge](https://github.com/procount/pinn-os/blob/master/os/os_list_v3.json) ?

procount, 2020-07-24:
I was just about to suggest that :smile: . or alternatively you could use github (in the releases)

procount, 2020-07-24:
Are these plain vanilla OS conversions, or have you customised any of them for your own purposes?
If they are plain vanilla versions and you want to share them with the general public, I could offer to host some of them on my sourceforge site, once they are tested.

Ae3NerdGod, 2020-07-24:
They are vanilla, and I don't plan on customising them any more than I have to to make them work on pinn properly. Does sourceforge cost money? I kinda assumed it did. The releases thing is a good idea too. I'll look into those

procount, 2020-07-24:
No it doesn't cost for open source stuff.

Ae3NerdGod, 2020-07-24:
actually the only custom one i wanted to make public was raspi OS Cinnamon, which is literally raspi os lite with tasksel install cinnamon-desktop ran, updated with apt, and then imaged

i will look into making a sourceforge but if youd like to host the updated manjaros and ubuntus if/when i get them to work with pinn ive got no issues, however ill only be making pi4 compatible stuffs for the foreseeable future

imgs i consider finished will be put in http://bowtie-ent.com/pinng/os_list_working.json and stored at http://bowtie-ent.com/pinng/working until i get a sourceforge set up

lurch, 2020-07-24:
> actually the only custom one i wanted to make public was raspi OS Cinnamon, which is literally raspi os lite with tasksel install cinnamon-desktop ran, updated with apt, and then imaged

You might want to look at https://github.com/RPi-Distro/pi-gen which might make your life easier in the long-term :wink: 

@XECDesign Are there any copyright / trademark / branding / naming issues to consider for people that want to distribute their own re-spins of Raspberry Pi OS ?