procount/pinn #405: Cannot install raspberry pi os(32bit) from /os Folder  with solution to resolv for workaround

georg3003, 2020-07-24:
Hello,
first of all, thank you for this great software. And sorry for my imperfect english.

Cannot install Raspberry Pi OS(32bit) (normal/full or lite) in no version of 32bits from /os folder in my case from usb-stick.
Error Message is: OS(es) installed with errors. See debug logs for details
[image]

I looked around, and found this:
`
3+0 records out
Processing OS: "Raspberry Pi OS (32-bit)" 
"/sbin/findfs LABEL=boot" 
 "/sbin/findfs: unable to resolve 'LABEL=boot'
" 
 
Executing: "/sbin/mkfs.fat -n boot -F 32 /dev/mmcblk0p6" 
Executing: "sh -o pipefail -c "cat /tmp/media/sda1/os/Raspberry_Pi_OS_(32)/boot.tar.xz | tee
 /tmp/fifo | xz -dc | bsdtar -xf - -C /mnt2  --no-same-owner  | sha256sum /tmp/fifo > /tmp/s
ha1.out.txt"" 
**"sh: syntax error: unexpected "("**
" 
mount: wrong fs type, bad option, bad superblock on /dev/mmcblk0p7,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
Error opening file: "/settings/installed_os.json" 
Number of bootables =  0 `
--------------------------------------------------### 
**_### My workaround solution is very simple:_**
I changed the name of the Foldername in the /os Folder.  

I changed it from (32bit) simply to 32bit , and it worked.

I deleted simply the "()" brackets or paranthesis from the Foldername.

Thanks

procount, 2020-07-24:
see also #398

georg3003, 2020-07-25:
I am glad to contribute a little bit. Nothing big. Only this small workaround.

georg3003, 2020-07-25:
Sorry I closed this without intention.

Ae3NerdGod, 2020-08-09:
replace the parenthesis with underscores or omit them,

 the `"/sbin/findfs: unable to resolve 'LABEL=boot'` error is not an error, if it were to find the label, that means you would have more than one partition named "boot" for your installation, when the program gets that error it knows it doesnt need to change the label of that partition

procount, 2020-08-24:
Should be fixed in 3.3.4.3

closed 2020-08-24 by procount