procount/pinn #407: Boot delay keeps messing up

pandalatte, 2020-08-07:
Hi everybody!

I have PINN installed on Pi3 B, with Raspbian and LibreELEC.
The boot menu (OS selection) delay is set for 10 seconds.

But without any reason it has began booting without any delay.
A wallpaper splash and then immediately it starts booting the default OS.
It happened once a few weeks ago, I just rolled back from my backup.
But today it happened again.

Where can I find the file that holds the delay value? 
Will it work if I fix it and set the Read-Only flag?

Thanks!

procount, 2020-08-09:
Did you tick one of the Sticky default boxes alongside one of your installed OSes? Press SHIFT when PINN boots, to get to the recovery screen, press ESC to get to the boot menu, and untick any selected OS.

procount, 2020-08-24:
Is this issue now fixed?

closed 2020-10-14 by procount